WorldFamousElectronics/PulseSensor_Amped_Arduino #1: Node.js implementation

Senade, 2015-05-05:
Hey,

I have written javascript code using the the same algorithm so that the pulse sensor can be used with node.js. However I am getting inaccurate values for BPM. What might the issue be?

 ```javascript
var rate = [0, 0, 0, 0, 0, 0, 0, 0, 0, 0],    
      sampleCounter = 0, 
      lastBeatTime = 0,  
      P = 512,               
      T = 512,                  
      thresh = 525,          
      amp = 100,              
      firstBeat = true,       
      secondBeat = false,
      IBI = 600,
      Pulse = false,
      BPM,
      Signal,
      QS = false;

var five = require('johnny-five'),
      board, sensor;
board = new five.Board();

board.on('ready', function() {
  	sensor = new five.Sensor({
            	pin: "A0",
              	freq: 2
        	});        

            sensor.scale([0,1024]).on('read', function() {
                  Signal = this.scaled;
                  calculate_bpm();

	            if(QS === true) {
            		console.log('BPM : ', BPM); 
            		QS = false;
		} 
            });                    
});

function calculate_bpm() {

            sampleCounter += 2;                           
            N = sampleCounter - lastBeatTime;       
          
            if(Signal < thresh && N > (IBI/5)*3) {       
                  if (Signal < T) {                        
                        T = Signal;                  
                  }
            }

            if(Signal > thresh && Signal > P) {          
                  P = Signal;                             
            }                                       

        	if (N > 250) {   

              if ((Signal > thresh) && (Pulse === false) && (N > (IBI/5)*3)) {        
                  Pulse = true;                               
                  IBI = sampleCounter - lastBeatTime;         
                  lastBeatTime = sampleCounter;               

                  if(secondBeat) {                        
                      secondBeat = false;                  
                      for(var i=0; i<=9; i++){             
                          rate[i] = IBI;                      
                      }
                  }

                  if(firstBeat) {                         
                      firstBeat = false;               
                      secondBeat = true;      
                      return;                              
                  }   

                  var runningTotal = 0;                 

                  for(var i=0; i<=8; i++) {              
                      rate[i] = rate[i+1];                  
                      runningTotal += rate[i];         
                  }

                  rate[9] = IBI;                          
                  runningTotal += rate[9];        
                  runningTotal /= 10;               
                  BPM = 60000/runningTotal; 
                  QS = true;                            
              }                       
          }

          if (Signal < thresh && Pulse === true){   
              Pulse = false;                         
              amp = P - T;                           
              thresh = amp/2 + T;               
              P = thresh;                            
              T = thresh;
          }

          if (N > 2500) {                           
              thresh = 512;                        
              P = 512;                               
              T = 512;                               
              lastBeatTime = sampleCounter;         
              firstBeat = true;                      
              secondBeat = false;               
          }
} 
```

samuraipapa, 2015-07-14:
I'd like to try to replicate your hardware/software set-up.  It's faster/better to address your issue that way. 

Could you post a photo and more detail on hardware/software environment you are experiencing an issue on.
   
Thanks.

samuraipapa, 2015-07-14:
Great looking javascript btw.

Senade, 2015-07-15:
Thank you! This sort of feature is better implemented at the Microcontroller itself. There is an effort to change the firmata protocol.

https://github.com/rwaldron/johnny-five/issues/788

samuraipapa, 2015-07-15:
It's debatable where it is best to put the pulse detection logic.  There are trade-offs on all sides. Thanks for the johnny-five link!  Cool stuff.

closed 2015-07-15 by samuraipapa