WorldFamousElectronics/PulseSensor_Amped_Arduino #3: mac app output gibberish - no signal

sabdolla, 2015-07-14:
Hi,
I just got an Arduino Uno and the pulse sensor. I downloaded the mac app too and followed all instructions. For some reason, the "serial monitor read-out" output of the pulse sensor mac app does not show anything, just some gibberish letters when I try 1) "static boolean serialVisual = true;"
[image]
 or , 2) "static boolean serialVisual = false;".

samuraipapa, 2015-07-14:
Gibberish letters means the serial communication is working.  It's mostly likely a speed setting on serial.  

Can you upload second screen shot of gibberish when you change the serialVisual "true/false" setting on Arduino.

sabdolla, 2015-07-14:
Sure, this is "true":
[image]
and this is "false":
[image]

samuraipapa, 2015-07-14:
If you open Arduino monitor in Arduino what do you see, with true and false changed and uploaded to Board ?

Serial Monitor in Arduino is Opened/Closed with the "magnifying glass icon" in the control panel. 

Please post shot-screen grab of Arduino's Serial Monitor.  What should be the last clue here. 

Also is the "purple cable" in the AO (Analog 0) pin on Arduino?

samuraipapa, 2015-07-14:
Also, when you open the Mac App, did you "refresh ports" and select the "..tty.USB..." Option?  

From your screen shot it looks like a serial port maybe not actually be selected.  Double check that selection.

sabdolla, 2015-07-15:
Hi,
I think I know where the problem might be: the port. Opening the Arduino monitor, I get the following message "Board at COM1 is not available". The Arduino website says that for a Mac, the following ports should be available:
![screen shot 2015-07-15 at 7 00 21 amjul 15 2015 (https://cloud.githubusercontent.com/assets/13322502/8697022/8d28eec6-2abf-11e5-81ae-550e335db414.png)
However, my Macbook Air only has these limited ports:
[image]
I tried to go on different websites and blogs to see if someone has this port problem, and some people do, but I haven't tried a proposed solution that works. What most suggest is installing the FTDI driver, however, Arduino Uno boards apparently don't have an FTDI hardware to need this driver.
Any suggestions?

Thanks,

samuraipapa, 2015-07-15:
You have the right Port. 
In your screen cap of Arduino select Prot>/dev/ucbmodem1411 (Arduino Uno). 
[image]

That should work for you.   The info online is confusing your I believe b/c it for boards before the Uno. 

1)  Please select above Port. 
2)  Open  File>Examples>Basics>Blink  
[image]

3)  Upload that to your board to see if it Uploads correctly. 
4)  You can change:

```javascript
void loop() {
  digitalWrite(13, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);              // wait for a second
  digitalWrite(13, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);              // wait for a second
}
```

```javascript
to 
void loop() {
  digitalWrite(13, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(100);              // wait for a second
  digitalWrite(13, LOW);    // turn the LED off by making the voltage LOW
  delay(100);              // wait for a second
}
```

and you should see the Pin 13's LED blink at a different speed.  
[image]

sabdolla, 2015-07-15:
Wow! Thank you so much, it worked.

I now have a question about the output, the serial monitor read-out window continuously outputs things (e.g. s502, s503, etc.), even when my finger is not in the pulse. 

Should I only pay attention to the top bar that ranges from 50 to 140? Are the units BMP?

Also, I am wondering if I can run the visualization code for mac or is it specific to windows?

Thank you so much,
Sara

samuraipapa, 2015-07-15:
Sara, 
I'm going to close this issue's item, OK.  The "Serial Gibberish" item is settled....

Please repost your second question with the GitHub Issues tab of the software in question.  
Screen-grabs & snapshots of your Arduino/PulseSensor are worth a thousand words.  :fireworks: 
Let's address those other questions like that.

Thanks!

closed 2015-07-15 by samuraipapa