WorldFamousElectronics/PulseSensor_Amped_Arduino #4: Pulse sensor output question

sabdolla, 2015-07-15:
Hi,
I have a question about the output, the serial monitor read-out window continuously outputs things (e.g. s502, s503, etc.), even when my finger is not in the pulse.

Should I only pay attention to the top bar that ranges from 50 to 140? Are the units BMP?

Also, I am wondering if I can run the visualization code for mac or is it specific to windows?

Thank you,

Sara

samuraipapa, 2015-07-20:
[image]

Good Question.   

The Mac App is suppose to look like above screen grab. 
The **"Serial Monitor Read-Out" window** just runs a "live log" of everything Arduino is saying via Serial communication.   So, you can expect to see an rapid roll of numbers that are "Pre-fixed with a Letter".
this 

The Letter identifies what that number means. 
"S" Means Signal
"B" means BPM Number
"Q" is a Heart-rate Variability Number

For example:
"S511" or "S###" is the raw Signal number for the Pulse Sensor.  It's simply the intesty of light the optical sensor literally sees.   

1) Try Moving your finger on and off the sensor.  When you cover and uncover the sensor, you'll see that you can kind of control the Signal the "S" + "Some Number Between ~0-1000".  "S515" for example. 

**The Serial Monitor Read-Out" window** is simple a raw and live feed from Arduino to the Mac App.  The Visual Level Indicators, turn this live numbers feed into a "Digital Level".
[image]

samuraipapa, 2015-07-20:
> *Should I only pay attention to the top bar that ranges from 50 to 140? Are the units BMP?

sabdolla, 2015-07-20:
Yes, thank you so much. Can I also see the actual signal by running the visualization code?

samuraipapa, 2015-07-20:
You mean by setting serial visualization = true?

samuraipapa, 2015-07-22:
[image]
Yes, there is an ASCI Visualizer in the *Arduino Software* Itself.  
This will give you the below output on:  Mac, Windows, Linux

sabdolla, 2015-07-22:
Thanks, I set it to false and it worked, very nice. What I had in mind is running the processing visualization app (http://pulsesensor.com/pages/processing-visualization). Can this app run on mac?
Thanks again,
Sara

samuraipapa, 2015-07-22:
Yes.  The Processing App runs on Mac, Win, Linux.

closed 2015-08-03 by samuraipapa

xueting123456, 2017-10-26:
Hi, hello. I saw your information on GitHub. Do you have the code to calculate the peak time difference through 2 pulse sensors?