WorldFamousElectronics/PulseSensor_Amped_Arduino #12: Error in Compilation

Mujaddadi, 2015-10-22:
Hi,
     As one more person mentioned in another issue of the code, I am having the same error. I am using Arduino IDE to compile for Intel Galileo 

Interrupt.ino: In function 'void interruptSetup()':
Interrupt.ino:17:3: error: 'TCCR2A' was not declared in this scope
Interrupt.ino:18:3: error: 'TCCR2B' was not declared in this scope
Interrupt.ino:19:3: error: 'OCR2A' was not declared in this scope
Interrupt.ino:20:3: error: 'TIMSK2' was not declared in this scope
Interrupt.ino:21:7: error: 'sei' was not declared in this scope
Interrupt.ino: At global scope:
Interrupt.ino:27:4: error: expected constructor, destructor, or type conversion before '(' token
Error compiling.

Anyone has any idea why we are getting this error?

jpanz, 2015-12-18:
Same Issue. Any help on this?

Mujaddadi, 2015-12-18:
Yeah I found that since arduino is Atmel based and Galileo is Quark based so they are not fully compatible. That's why we are getting interrupts error. I found out that for pulse sensor they have c++ code in Intel upm library http://iotdk.intel.com/docs/master/upm/dir_3d88723ba241360d69dd8d7099a56335.html but it not selectable in eclipse sensor view so I created an issue in upm GitHub page  https://github.com/intel-iot-devkit/upm/issues/300 but apparently the pulse sensor developer are busy in some more important issues then fixing bugs. There is also support for this sensor in Intel IOT eclipse for Java but for that first we will need to install Java in the Linux image which Intel provides, since it do not come pre installed with it ironically. So I am trying to install Java in the linux image at the moment.  

I am trying to install Java because I have some other requirements but if you want to use arduino IDE, I found one http://gowrisomanath.github.io/gsPulseHeartRate/. Also they have C++ library too but I haven't tried it yet

stefanoscarani, 2016-01-09:
Hi, I've the same problem, but with an ArduinoUno (last version) board.

sketch_jan09c.ino: In function 'void setup()':
sketch_jan09c.ino:37:18: error: 'interruptSetup' was not declared in this scope
sketch_jan09c.ino: In function 'void loop()':
sketch_jan09c.ino:47:18: error: 'serialOutput' was not declared in this scope
sketch_jan09c.ino:55:37: error: 'serialOutputWhenBeatHappens' was not declared in this scope

I'm an old Max users, but very new in Arduino.

yury-g, 2026-08-16:
This repository has been retired as a read-only historical archive. Its code and discussion remain preserved, but fixes and support no longer happen here.

For current Arduino installation, examples, supported boards, and source, use [PulseSensorPlayground](https://github.com/WorldFamousElectronics/PulseSensorPlayground). If a problem is reproducible with the current Playground release, open a new issue there with the board and core versions, library version, wiring and supply voltage, raw waveform, and a minimal sketch.

Closing this historical issue as not planned in this retired repository.

closed 2026-08-16 by yury-g