WorldFamousElectronics/PulseSensor_Amped_Arduino #13: Detecting BPM without touching the sensor?

george-i, 2015-10-28:
This is my code on GitHub: https://github.com/george-i/Adafruit-Flora---PulseSensorAmped

I've adapted the code to work with Flora, however, on the serial it prints various values without even touching the sensor.
After I touch the sensor, it looks like the bpm is adjusting, but then suddenly changes.

Any idea what's going on?

biomurph, 2015-10-28:
There isn't very much noise reduction in the Pulse Sensor Amped code. So it is reacting to small noise fluctuations in the ambient light when it's not touching your finger. To fix this, you will need to add some noise cancelling code. There is a varialbe, called 'amp' in the interrupt code. That variable is updated every time the Pulse Sensor finds a beat. You can use a threshold on that value to limit the size of the 'pulse' signal that you want the sensor to react to. 
This is something that we're working on...

george-i, 2015-10-28:
I understand. 

On the other hand, when I touch the sensor the value is very high.
I have a strap heart rate monitor and I figured it out that I have to divide to 2, to have similar value. 
I don't know why...

biomurph, 2015-10-28:
what development board are you using with the Pulse Sensor? You may need to
change some of the code to get the bpm right.

~~~
Making the world safe for robots

On Wed, Oct 28, 2015 at 2:58 PM, George Ilie <notifications@github.com>
wrote:

> I understand.
>
> On the other hand, when I touch the sensor the value is very high.
> I have a strap heart rate monitor and I figured it out that I have to
> divide to 2, to have similar value.
> I don't know why...
>
> —
> Reply to this email directly or view it on GitHub
> <https://github.com/WorldFamousElectronics/PulseSensor_Amped_Arduino/issues/13#issuecomment-151955522>
> .
>

george-i, 2015-10-29:
Adafruit Flora, based on ATmega32u4.
I did the changes (detailed here: http://forum.arduino.cc/index.php?topic=355957.0).

biomurph, 2015-10-29:
You don't need to use the analogReference(EXTERNAL); unless you are using
two different power supplies. If the whole system is running at 3.3V, then
don't include that line of code.
What clock speed is the Flora running at?

J

~~~
Making the world safe for robots

On Thu, Oct 29, 2015 at 1:11 AM, George Ilie <notifications@github.com>
wrote:

> Adafruit Flora, based on ATmega32u4.
> I did the changes (detailed here:
> http://forum.arduino.cc/index.php?topic=355957.0).
>
> —
> Reply to this email directly or view it on GitHub
> <https://github.com/WorldFamousElectronics/PulseSensor_Amped_Arduino/issues/13#issuecomment-152077074>
> .
>

yury-g, 2026-08-16:
This repository has been retired as a read-only historical archive. Its code and discussion remain preserved, but fixes and support no longer happen here.

For current Arduino installation, examples, supported boards, and source, use [PulseSensorPlayground](https://github.com/WorldFamousElectronics/PulseSensorPlayground). If a problem is reproducible with the current Playground release, open a new issue there with the board and core versions, library version, wiring and supply voltage, raw waveform, and a minimal sketch.

Closing this historical issue as not planned in this retired repository.

closed 2026-08-16 by yury-g