WorldFamousElectronics/PulseSensor_Amped_Arduino #16: Arduino MKR1000

beckmx, 2016-03-02:
Hello guys, I was testing the library, I noticed there is a compatibility issue with the timer interrupt 2, I was wondering if the timer can be replaced for something more general for other arduinos.

Greetings

biomurph, 2016-03-08:
What is the target arduino that you are using? Check the Interrupt Notes tab for details about how to change the timer and ISR vector.

beckmx, 2016-03-08:
Hello, I could make it work, actually is ok for all arduinos. What I did
was to replace the ISR for a counter,

void loop(){
loopTmr();
  delay(2);
  reps+=1;
  if(reps==10){
     //execute serial output
}
}

The signal displays very good and smoothly, it could be considered as an
alternative for other arduinos as well.

Greetings!

CarlosE473, 2016-03-24:
Could you please show the full replacement of the code? Or what is "loopTmr();"

beckmx, 2016-03-24:
Look at the dependencies,there is a method called like that just look at
the inside and fix according to my notes

rewant24, 2017-06-12:
@beckmx Hi, I looked at the dependencies and still couldn't figure it out. I was wondering if is it possible for you to kindly share your file? (newbie here, my apologies) 
I am using a mkr1000 as well and getting all the issues as everyone else with the interrupt timer. I have looked online (a lot of other people with the same problem couldn't resolve this issue either), and played around with the code, but nothing has worked for me so far. I just want to print the BPM values in serial monitor instead of the raw analog/ signal values!  Is there an easy workaround for this problem?

biomurph, 2017-06-12:
@beckmx As long as this 'software timer' solution works for you, that's great. But when you use 
delay(2);
to do the timing, the system can be easily broken if you add in other delay or other operations.
In my 'software timer', I use a couple of variables to monitor the passage of micros(). This is a more flexible solution.  You can find the example code at the end of this document.
https://github.com/WorldFamousElectronics/PulseSensor_Amped_Arduino/blob/master/PulseSensorAmped_Arduino_1.5.0/Timer_Interrupt_Notes.ino

yury-g, 2026-08-16:
This repository has been retired as a read-only historical archive. Its code and discussion remain preserved, but fixes and support no longer happen here.

For current Arduino installation, examples, supported boards, and source, use [PulseSensorPlayground](https://github.com/WorldFamousElectronics/PulseSensorPlayground). If a problem is reproducible with the current Playground release, open a new issue there with the board and core versions, library version, wiring and supply voltage, raw waveform, and a minimal sketch.

Closing this historical issue as not planned in this retired repository.

closed 2026-08-16 by yury-g