WorldFamousElectronics/PulseSensor_Amped_Arduino #17: Lots of Random Characters are being outputted

JAD1995, 2016-03-04:
I am trying to run this program and it is never going into the serialoutputwhenbeethappens. 
Also it is outputting some sensible data from the serialoutput but after about 6 lines it just outputs a never ending crazy long string of random characters.

biomurph, 2016-03-08:
Can you show me what the Serial output looks like?
Also, what is your target hardware?

anandhost, 2016-05-06:
[image]
Hey I'm getting this output, what's solution?

yury-g, 2026-08-16:
This repository has been retired as a read-only historical archive. Its code and discussion remain preserved, but fixes and support no longer happen here.

For current Arduino installation, examples, supported boards, and source, use [PulseSensorPlayground](https://github.com/WorldFamousElectronics/PulseSensorPlayground). If a problem is reproducible with the current Playground release, open a new issue there with the board and core versions, library version, wiring and supply voltage, raw waveform, and a minimal sketch.

Closing this historical issue as not planned in this retired repository.

closed 2026-08-16 by yury-g