WorldFamousElectronics/PulseSensor_Amped_Arduino #18: compiling issue

mdagcilar, 2016-03-04:
We cannot seem to compile the code onto our Micro Blend Arduino board 
( http://redbearlab.com/blendmicro/ )

We have correctly wired the board to the pulse sensor and the board compiles other files perfectly fine.

This is the stack trace:
Exception in thread "Thread-214" java.lang.StackOverflowError
	at java.util.regex.Pattern$GroupHead.match(Pattern.java:4168)
	at java.util.regex.Pattern$Loop.match(Pattern.java:4295)
	at java.util.regex.Pattern$GroupTail.match(Pattern.java:4227)
	at java.util.regex.Pattern$Curly.match0(Pattern.java:3789)
	at java.util.regex.Pattern$Curly.match(Pattern.java:3744)
	at java.util.regex.Pattern$Neg.match(Pattern.java:4609)
	at java.util.regex.Pattern$BmpCharProperty.match(Pattern.java:3366)
	at java.util.regex.Pattern$GroupHead.match(Pattern.java:4168)
	at java.util.regex.Pattern$Loop.match(Pattern.java:4295)
	at java.util.regex.Pattern$GroupTail.match(Pattern.java:4227)
	at java.util.regex.Pattern$Curly.match0(Pattern.java:3789)
	at java.util.regex.Pattern$Curly.match(Pattern.java:3744)
	at java.util.regex.Pattern$Neg.match(Pattern.java:4609)
	at java.util.regex.Pattern$BmpCharProperty.match(Pattern.java:3366)
	at java.util.regex.Pattern$GroupHead.match(Pattern.java:4168)
	at java.util.regex.Pattern$Loop.match(Pattern.java:4295)
	at java.util.regex.Pattern$GroupTail.match(Pattern.java:4227)
	at java.util.regex.Pattern$Curly.match0(Pattern.java:3789)
	at java.util.regex.Pattern$Curly.match(Pattern.java:3744)
	at java.util.regex.Pattern$Neg.match(Pattern.java:4609)
	at java.util.regex.Pattern$BmpCharProperty.match(Pattern.java:3366)
	at java.util.regex.Pattern$GroupHead.match(Pattern.java:4168)
	at java.util.regex.Pattern$Loop.match(Pattern.java:4295)
	at java.util.regex.Pattern$GroupTail.match(Pattern.java:4227)
	at java.util.regex.Pattern$Curly.match0(Pattern.java:3789)
	at java.util.regex.Pattern$Curly.match(Pattern.java:3744)
	at java.util.regex.Pattern$Neg.match(Pattern.java:4609)
	at java.util.regex.Pattern$BmpCharProperty.match(Pattern.java:3366)
	at java.util.regex.Pattern$GroupHead.match(Pattern.java:4168)
	at java.util.regex.Pattern$Loop.match(Pattern.java:4295)
	at java.util.regex.Pattern$GroupTail.match(Pattern.java:4227)
	at java.util.regex.Pattern$Curly.match0(Pattern.java:3789)
	at java.util.regex.Pattern$Curly.match(Pattern.java:3744)
	at java.util.regex.Pattern$Neg.match(Pattern.java:4609)
	at java.util.regex.Pattern$BmpCharProperty.match(Pattern.java:3366)
	at java.util.regex.Pattern$GroupHead.match(Pattern.java:4168)
	at java.util.regex.Pattern$Loop.match(Pattern.java:4295)
	at java.util.regex.Pattern$GroupTail.match(Pattern.java:4227)
	at java.util.regex.Pattern$Curly.match0(Pattern.java:3789)
	at java.util.regex.Pattern$Curly.match(Pattern.java:3744)
	at java.util.regex.Pattern$Neg.match(Pattern.java:4609)
	at java.util.regex.Pattern$BmpCharProperty.match(Pattern.java:3366)
	at java.util.regex.Pattern$GroupHead.match(Pattern.java:4168)
	at java.util.regex.Pattern$Loop.match(Pattern.java:4295)
	at java.util.regex.Pattern$GroupTail.match(Pattern.java:4227)
	at java.util.regex.Pattern$Curly.match0(Pattern.java:3789)
	at java.util.regex.Pattern$Curly.match(Pattern.java:3744)
	at java.util.regex.Pattern$Neg.match(Pattern.java:4609)
	at java.util.regex.Pattern$BmpCharProperty.match(Pattern.java:3366)
	at java.util.regex.Pattern$GroupHead.match(Pattern.java:4168)
	at java.util.regex.Pattern$Loop.match(Pattern.java:4295)
	at java.util.regex.Pattern$GroupTail.match(Pattern.java:4227)
	at java.util.regex.Pattern$Curly.match0(Pattern.java:3789)
	at java.util.regex.Pattern$Curly.match(Pattern.java:3744)
	at java.util.regex.Pattern$Neg.match(Pattern.java:4609)
	at java.util.regex.Pattern$BmpCharProperty.match(Pattern.java:3366)
	at java.util.regex.Pattern$GroupHead.match(Pattern.java:4168)
	at java.util.regex.Pattern$Loop.match(Pattern.java:4295)
	at java.util.regex.Pattern$GroupTail.match(Pattern.java:4227)
	at java.util.regex.Pattern$Curly.match0(Pattern.java:3789)
	at java.util.regex.Pattern$Curly.match(Pattern.java:3744)
	at java.util.regex.Pattern$Neg.match(Pattern.java:4609)
	at java.util.regex.Pattern$BmpCharProperty.match(Pattern.java:3366)
[... 936 lines cut ...]
	at java.util.regex.Pattern$Neg.match(Pattern.java:4609)
	at java.util.regex.Pattern$BmpCharProperty.match(Pattern.java:3366)
	at java.util.regex.Pattern$GroupHead.match(Pattern.java:4168)
	at java.util.regex.Pattern$Loop.match(Pattern.java:4295)
	at java.util.regex.Pattern$GroupTail.match(Pattern.java:4227)
	at java.util.regex.Pattern$Curly.match0(Pattern.java:3789)
	at java.util.regex.Pattern$Curly.match(Pattern.java:3744)
	at java.util.regex.Pattern$Neg.match(Pattern.java:4609)
	at java.util.regex.Pattern$BmpCharProperty.match(Pattern.java:3366)
	at java.util.regex.Pattern$GroupHead.match(Pattern.java:4168)
	at java.util.regex.Pattern$Loop.match(Pattern.java:4295)
	at java.util.regex.Pattern$GroupTail.match(Pattern.java:4227)
	at java.util.regex.Pattern$Curly.match0(Pattern.java:3789)
	at java.util.regex.Pattern$Curly.match(Pattern.java:3744)
	at java.util.regex.Pattern$Neg.match(Pattern.java:4609)
	at java.util.regex.Pattern$BmpCharProperty.match(Pattern.java:3366)
	at java.util.regex.Pattern$GroupHead.match(Pattern.java:4168)
	at java.util.regex.Pattern$Loop.match(Pattern.java:4295)
	at java.util.regex.Pattern$GroupTail.match(Pattern.java:4227)
	at java.util.regex.Pattern$Curly.match0(Pattern.java:3789)
	at java.util.regex.Pattern$Curly.match(Pattern.java:3744)
	at java.util.regex.Pattern$Neg.match(Pattern.java:4609)
	at java.util.regex.Pattern$BmpCharProperty.match(Pattern.java:3366)
	at java.util.regex.Pattern$GroupHead.match(Pattern.java:4168)
	at java.util.regex.Pattern$Loop.match(Pattern.java:4295)

biomurph, 2016-03-08:
Hi,

The module that you link to is a ATmega32u4. Please make sure that you are selecting the correct interrupt setup and vector for that target. 
The list of errors you send is unhelpfully long. Try commenting out some of the more critical code to see if you can sneak up on the problem.

mdagcilar, 2016-03-08:
we fixed this, the setup tutorial on the blend micro page is outdated. We followed a new one. 

Issue closed, not related to this repo.

closed 2016-03-08 by mdagcilar