WorldFamousElectronics/PulseSensor_Amped_Arduino #19: Random digits.

hydrogenium74, 2016-04-01:
[image]
this is how it looks, just using the code on this forum and plugges my sensor in the arduino and uploaded the code om my Arduino anyone know why this happnes some help would be nice.

ps Thanks in advance

hydrogenium74, 2016-04-01:
Fixed it the port my arduino was using wasnt right so just had to change it in the code.

closed 2016-04-01 by hydrogenium74

kratitripathi, 2016-04-19:
Where exactly did you make the change in the code? Facing the same issue.

biomurph, 2016-04-20:
@kratitripathi it looks from the image that they have the baud rate set for 9600. It should be 115200.
You can change the baud rate settings in the lower right of the terminal window. There is a drop-down menu there.