WorldFamousElectronics/PulseSensor_Amped_Arduino #20: using Pulse Sensor with RFDuino/Simblee

shadow85926, 2016-04-06:
Will not work after replacing timing code to work with an ARM Cortex M0 by replacing timer with delay().

[Timer_Interrupt_Notes.txt](https://github.com/WorldFamousElectronics/PulseSensor_Amped_Arduino/files/206933/Timer_Interrupt_Notes.txt)
[AllSerialHandling.txt](https://github.com/WorldFamousElectronics/PulseSensor_Amped_Arduino/files/206936/AllSerialHandling.txt)
[Interrupt.txt](https://github.com/WorldFamousElectronics/PulseSensor_Amped_Arduino/files/206934/Interrupt.txt)
[PulseSensorAmped_Arduino_1dot4.txt](https://github.com/WorldFamousElectronics/PulseSensor_Amped_Arduino/files/206935/PulseSensorAmped_Arduino_1dot4.txt)

biomurph, 2016-04-06:
In your Interrupt.ino file you are trying to do a Serial.print every sample (500Hz!) 
Instead, try using a modulo on the sample counter or something to dial that down. 
Maybe that will help?

shadow85926, 2016-04-06:
Still doesn't work only sending ever 20ms or 40ms

biomurph, 2016-04-06:
You are using the SerialOutput() function? Could that be doing something?

shadow85926, 2016-04-06:
wave form for Serial.print every 50 ms

<blockquote class="imgur-embed-pub" lang="en" data-id="a/o2eOo"><a href="//imgur.com/a/o2eOo">View post on imgur.com</a></blockquote><script async src="//s.imgur.com/min/embed.js" charset="utf-8"></script>

shadow85926, 2016-04-06:
i moved it to after the function that looks for heart rate, but it still doesn't work

biomurph, 2016-04-06:
I'll have time to set this up on my own simblee tomorrow and we'll figure it out

shadow85926, 2016-04-06:
Awesome thanks!

biomurph, 2016-04-07:
Ok, it looks like I've found your problem. At least with your latest version on the git.
You have a delay(20); in there from my original code. That was causing the software to not find any heartbeats. I took that out, and also made a couple of adjustments and have it working on Simblee with our Processing visualizer! Yay!

I forked your code, and made changes here
https://github.com/biomurph/SimbleeArduinoCode
It uses a millis() timer to send serial data, and a micros() timer to sample the sensor.

Cheers

shadow85926, 2016-04-08:
Thank you so much!

yury-g, 2026-08-16:
This repository has been retired as a read-only historical archive. Its code and discussion remain preserved, but fixes and support no longer happen here.

For current Arduino installation, examples, supported boards, and source, use [PulseSensorPlayground](https://github.com/WorldFamousElectronics/PulseSensorPlayground). If a problem is reproducible with the current Playground release, open a new issue there with the board and core versions, library version, wiring and supply voltage, raw waveform, and a minimal sketch.

Closing this historical issue as not planned in this retired repository.

closed 2026-08-16 by yury-g