WorldFamousElectronics/PulseSensor_Amped_Arduino #21: incorrect  Signal

5y, 2016-05-19:
Hi,
trying to get the right signal from the Sensor but i got follow Singal, How to Fix it ?:
[image]

biomurph, 2016-05-20:
That does look rather odd. Did you make any changes to the code in any way?
There does seem to be some connection between the computer and arduino.
Let us know what hardware you are using, and give us more details about your system setup to help troubleshoot.

5y, 2016-05-20:
@biomurph  No I used:
1- Arduino UNO 
2- Linux OS
3- PulseSensorAmped_Arduino for get data from Arduino
4- PulseSensor_Amped_Processing_Visualizer for Show Data 
I downloaded exact the same code in the github and run.
The Another odd things is my sensor wire color is different than Other Pulse sensor : 
[image]

biomurph, 2016-05-20:
Thanks for the info. You are using a 'knock-off' off our Pulse Sensor. That is not our tech, and there are many imitators of our open source hardware. Please tell us where you got it from, so other folks will know not to purchase from them. 

Unfortunately, we cannot help you with the problem, as we have no idea what is going on with the hardware you have.

5y, 2016-05-20:
Thanks @biomurph I buy it from a electric local shop and it's looks like your sensor.

biomurph, 2016-05-20:
What shop? Where are you?

5y, 2016-05-21:
I Live in Iran and need your sensor for a research program, I can't buy your sensor from your official website Directly so after looking around about your sensor one local shop in my city (Mashhad) give to me a very similar, after trying very hard for fix the problem I decided send an issue about sensor and until that I i thought my sensor is original!

Please posting a warning in your website about the Fake pulse sensor for other Person.
Thanks.

biomurph, 2016-05-21:
Well, that is something... We have alot of copiers in China, which is the likely source of your counterfeit part. I have tried to purchase some of the copies, but that one does not look familiar. It may be possible to try and make it work, but we are very far away. I'd like to figure out if there is a way we can help you. If you email me joelatpulsesensordotcom we can pick up the conversation there.

5y, 2016-05-21:
@biomurph Thank I'll send you email.

closed 2016-05-21 by 5y