WorldFamousElectronics/PulseSensor_Amped_Arduino #23: Dual Sensor Processing

PCesarano, 2016-06-15:
So, I am trying to use 2 sensors at the same time for an application...one pulse sensor on your ear, one on your finger.  I am curious if anyone knows how to produce 2 different serial outputs from the two sensor inputs...one for each...and display them in the visualizer. 

I imagine the Mega (which is what I have) will permit this, but I'm new and thought that this "easy" task would be...it has not been.

Any thoughts?

biomurph, 2016-06-15:
Hi PCesarano,

I have actually done this recently and just put my code up on github here
https://github.com/biomurph/PulseSensorAmped_2_Sensors
Please let me know if this works for you. It runs on an UNO

PCesarano, 2016-06-15:
Thanks Joel!!  I'll give it a try.  Much appreciated.

Patrick

PCesarano, 2016-07-01:
Joel,

I have a random question...I'm trying to find the area under the curve of a
given pulse, and am also trying to detect the time difference between
pulses.  Your 2-sensor code works perfectly and I really appreciate your
help on that.  Do you have any insights on essentially integrating each
pulse, and also detecting timing between pulses?  I can write the code, but
I'm hopeful it's already out there somewhere.

Thanks.

Patrick

xueting123456, 2017-10-26:
Hi, hello. I saw your information on GitHub. Do you have the code to calculate the peak time difference through 2 pulse sensors?

joelSensor, 2017-10-26:
I don't have the code handy, but it's not too hard to do. 
When a beat is found on one sensor, you can take the time and measure it to the beat of the second sensor and then vice versa. You'll want to use the shortest time as the difference

xueting123456, 2017-10-26:
can you help me write this code based on your code Pluse senors_2。this time PTT is used to estimute the bloode pressure.

joelSensor, 2017-10-26:
Oh, I see. You are placing the sensors on the same person in different places to derive PTT. Got it.
Well, you should know which sensor is closer to the heart, and so you can measure the time between that sensor's beat and the other sensor's beat. We sample the data at 2mS intervals, so that would be the resolution of your PTT.
I can't help you with the code, however. That would be a lot of work and I could only do it for a fee.

xueting123456, 2017-10-26:
My idea is this, two sensors are placed in front and back positions of the fingers, respectively, to measure two pulse waves. When the first pulse wave peak is detected, the recording time is T1=micros (); then the peak value of the second waveform is detected; the recording time is T2 =micros (); PTT=T1-T2

xueting123456, 2017-10-26:
I can pay for you

joelSensor, 2017-10-26:
please contact me joel-at-pulsesensor-dot-com to discuss your project further and see if it's anything that I can help with.

xueting123456, 2017-10-27:
Jole,i send a emil to you,thanks you

xueting123456, 2017-10-27:
hi,joel,do you recive my email?

xueting123456, 2017-10-27:
有人可以帮忙吗？

xueting123456, 2017-10-28:
joel,Can you help me? It seems simple, but I don't know how to make it happen. Looking at the examples of your 2 sensors, I think it can be implemented as long as it is modified, but I don't know how to modify it.

yury-g, 2026-08-16:
This repository has been retired as a read-only historical archive. Its code and discussion remain preserved, but fixes and support no longer happen here.

For current Arduino installation, examples, supported boards, and source, use [PulseSensorPlayground](https://github.com/WorldFamousElectronics/PulseSensorPlayground). If a problem is reproducible with the current Playground release, open a new issue there with the board and core versions, library version, wiring and supply voltage, raw waveform, and a minimal sketch.

Closing this historical issue as not planned in this retired repository.

closed 2026-08-16 by yury-g