WorldFamousElectronics/PulseSensor_Amped_Arduino #25: my pulse detect 200 more bpm

lukaskris, 2016-07-13:
can you help me why my pulse detect more value, i didnt touch sensor, and if i touch the bpm isnt stable, im confuse for that, im use 4.7k resistor and 10k resistor and 10mf 32v kapasitor for reduce the noise, it cant help anything -_-

biomurph, 2016-07-13:
Hello lukaskris,

I need more information to help you. Can you please send a photo of your setup?
What OS are you using? What arduino board are you using? what code are you using?

lukaskris, 2016-07-13:
i use arduino uno, im use windows 10, i use code pulse sensor amped with update HRV
this is my screenshoot visualize in processing
[image]

biomurph, 2016-07-13:
Hi,

It does look like something has changed. What did you do besides add the blue '0 HRV'?
Also, did you do anything to the Arduino code? It looks like from the picture that your data stream is not updating regularly.

lukaskris, 2016-07-13:
#include <LiquidCrystal_I2C.h>
/*  Pulse Sensor Amped 1.4    by Joel Murphy and Yury Gitman   http://www.pulsesensor.com
----------------------  Notes ----------------------  ---------------------- 
This code:
1) Blinks an LED to User's Live Heartbeat   PIN 13
2) Fades an LED to User's Live HeartBeat
3) Determines BPM
4) Prints All of the Above to Serial
Read Me:
https://github.com/WorldFamousElectronics/PulseSensor_Amped_Arduino/blob/master/README.md   
Updated with HRV display by Elliot Mebane, Roguish, Inc. Feb 2016
HRV requires 1 minute of data before it's accurate. 
 ----------------------       ----------------------  ----------------------
*/  
LiquidCrystal_I2C lcd(0x27, 16, 2);
//  Variables
int pulsePin = 0;                 // Pulse Sensor purple wire connected to analog pin 0
int blinkPin = 13;                // pin to blink led at each beat
int fadePin = 5;                  // pin to do fancy classy fading blink at each beat
int fadeRate = 0;                 // used to fade LED on with PWM on fadePin

// Volatile Variables, used in the interrupt service routine!
volatile int BPM;                   // int that holds raw Analog in 0. updated every 2mS
volatile int Signal;                // holds the incoming raw data
volatile int IBI = 600;             // int that holds the time interval between beats! Must be seeded! 
volatile boolean Pulse = false;     // "True" when User's live heartbeat is detected. "False" when not a "live beat". 
volatile boolean QS = false;        // becomes true when Arduoino finds a beat.

volatile int RRDelta = 0;
volatile double RMSSD = 0;
volatile double LN20RMSSD = 0;
volatile double SDNN = 0;
volatile double SDSD = 0;
volatile double NN50 = 0;
volatile double PNN50 = 0;

// Regards Serial OutPut  -- Set This Up to your needs
static boolean serialVisual = false;   // Set to 'false' by Default.  Re-set to 'true' to see Arduino Serial Monitor ASCII Visual Pulse 

void setup(){
  pinMode(blinkPin,OUTPUT);         // pin that will blink to your heartbeat!
  pinMode(fadePin,OUTPUT);          // pin that will fade to your heartbeat!
  Serial.begin(115200);             // we agree to talk fast!
  interruptSetup();                 // sets up to read Pulse Sensor signal every 2mS 
  lcd.begin();
   // IF YOU ARE POWERING The Pulse Sensor AT VOLTAGE LESS THAN THE BOARD VOLTAGE, 
   // UN-COMMENT THE NEXT LINE AND APPLY THAT VOLTAGE TO THE A-REF PIN
//   analogReference(EXTERNAL);   
}

//  Where the Magic Happens
void loop(){
  
    serialOutput() ;       
    
  if (QS == true){     // A Heartbeat Was Found
                       // BPM and IBI have been Determined
                       // Quantified Self "QS" true when arduino finds a heartbeat
        fadeRate = 255;         // Makes the LED Fade Effect Happen
                                // Set 'fadeRate' Variable to 255 to fade LED with pulse
        serialOutputWhenBeatHappens();   // A Beat Happened, Output that to serial.     
        lcd.clear();
        lcd.print(BPM);
        QS = false;                      // reset the Quantified Self flag for next time    
  }
     
  ledFadeToBeat();                      // Makes the LED Fade Effect Happen 
  delay(500);                             //  take a break
}

void ledFadeToBeat(){
    fadeRate -= 15;                         //  set LED fade value
    fadeRate = constrain(fadeRate,0,255);   //  keep LED fade value from going into negative numbers!
    analogWrite(fadePin,fadeRate);          //  fade LED
  }

lukaskris, 2016-07-13:
i just download and uploading into my arduino
i didnt know what the problem
1 problem sensor always output 200 more 
and 2 the output when i touch the sensor if my ESP8266 is on is bad, better if the ESP8266 off

lukaskris, 2016-07-13:
thanks it work well,
why the delay set 20? and 500 make big problem?
yes i use 4rd module, pulse+esp8266+sd+lcd
i want create heart tracker with android,
this HRV just stay 0??

biomurph, 2016-07-13:
I would need to see what your processing sketch is doing to know why HRV is 0.

don't use any delay in the loop longer than 20, or else the loop will not see the heartbeats.

lukaskris, 2016-07-13:
[image]

lukaskris, 2016-07-13:
[image]
[image]

lukaskris, 2016-07-13:
if i use RTC DS1302 for get the real time, can i use millis() for get delay to output in my lcd screen?

biomurph, 2016-07-13:
You can use the Uno board millis() to time your LCD writes if you need to. You don't really need an RTC for that.

If you have your computer plugged into AC power, you can unplug it  and see a much cleaner signal...

lukaskris, 2016-07-13:
i use my laptop, and my laptop still connected with the adaptor, but my pulse hrv still 0 point,
so if i want real time, where i must get it? from esp8266? or where?
thanks

yury-g, 2026-08-16:
This repository has been retired as a read-only historical archive. Its code and discussion remain preserved, but fixes and support no longer happen here.

For current Arduino installation, examples, supported boards, and source, use [PulseSensorPlayground](https://github.com/WorldFamousElectronics/PulseSensorPlayground). If a problem is reproducible with the current Playground release, open a new issue there with the board and core versions, library version, wiring and supply voltage, raw waveform, and a minimal sketch.

Closing this historical issue as not planned in this retired repository.

closed 2026-08-16 by yury-g