WorldFamousElectronics/PulseSensor_Amped_Arduino #26: Getting serial monitor output wrong

sachin0987, 2016-07-18:
I am getting output in about 100+ bpm.

code is using for this:

// Variables
int pulsePin = 0; // Pulse Sensor purple wire connected to analog pin 0
int blinkPin = 13; // pin to blink led at each beat
int fadePin = 5; // pin to do fancy classy fading blink at each beat
int fadeRate = 0; // used to fade LED on with PWM on fadePin

// Volatile Variables, used in the interrupt service routine!
volatile int BPM; // int that holds raw Analog in 0. updated every 2mS
volatile int Signal; // holds the incoming raw data
volatile int IBI = 600; // int that holds the time interval between beats! Must be seeded! 
volatile boolean Pulse = false; // "True" when User's live heartbeat is detected. "False" when not a "live beat". 
volatile boolean QS = false; // becomes true when Arduoino finds a beat.

volatile int RRDelta = 0;
volatile double RMSSD = 0;
volatile double LN20RMSSD = 0;
volatile double SDNN = 0;
volatile double SDSD = 0;
volatile double NN50 = 0;
volatile double PNN50 = 0;

// Regards Serial OutPut -- Set This Up to your needs
static boolean serialVisual = true; // Set to 'false' by Default. Re-set to 'true' to see Arduino Serial Monitor ASCII Visual Pulse

void setup(){
pinMode(blinkPin,OUTPUT); // pin that will blink to your heartbeat!
pinMode(fadePin,OUTPUT); // pin that will fade to your heartbeat!
Serial.begin(115200); // we agree to talk fast!
interruptSetup(); // sets up to read Pulse Sensor signal every 2mS 
// IF YOU ARE POWERING The Pulse Sensor AT VOLTAGE LESS THAN THE BOARD VOLTAGE, 
// UN-COMMENT THE NEXT LINE AND APPLY THAT VOLTAGE TO THE A-REF PIN
// analogReference(EXTERNAL);

}

// Where the Magic Happens
void loop(){

serialOutput() ;       
if (QS == true){ // A Heartbeat Was Found
// BPM and IBI have been Determined
// Quantified Self "QS" true when arduino finds a heartbeat
fadeRate = 255; // Makes the LED Fade Effect Happen
// Set 'fadeRate' Variable to 255 to fade LED with pulse
serialOutputWhenBeatHappens(); // A Beat Happened, Output that to serial.

QS = false; // reset the Quantified Self flag for next time

}

ledFadeToBeat(); // Makes the LED Fade Effect Happen 
delay(20); // take a break
}

void ledFadeToBeat(){
fadeRate -= 15; // set LED fade value
fadeRate = constrain(fadeRate,0,255); // keep LED fade value from going into negative numbers!
analogWrite(fadePin,fadeRate); // fade LED
}
[image]

biomurph, 2016-07-19:
you are adding a bunch of variables without saying what they do or showing the code that they  are used in. please be more explicit.

what do these do?
volatile int RRDelta = 0;
volatile double RMSSD = 0;
volatile double LN20RMSSD = 0;
volatile double SDNN = 0;
volatile double SDSD = 0;
volatile double NN50 = 0;
volatile double PNN50 = 0;

yury-g, 2026-08-16:
This repository has been retired as a read-only historical archive. Its code and discussion remain preserved, but fixes and support no longer happen here.

For current Arduino installation, examples, supported boards, and source, use [PulseSensorPlayground](https://github.com/WorldFamousElectronics/PulseSensorPlayground). If a problem is reproducible with the current Playground release, open a new issue there with the board and core versions, library version, wiring and supply voltage, raw waveform, and a minimal sketch.

Closing this historical issue as not planned in this retired repository.

closed 2026-08-16 by yury-g