WorldFamousElectronics/PulseSensor_Amped_Arduino #27: my pulse isnt stable 200 bpm more

lukaskris, 2016-07-30:
Ask, why my pulse up and down
isnt stable again
my code

#include <LiquidCrystal_I2C.h>
/*  Pulse Sensor Amped 1.4    by Joel Murphy and Yury Gitman   http://www.pulsesensor.com
----------------------  Notes ----------------------  ---------------------- 
This code:
1) Blinks an LED to User's Live Heartbeat   PIN 13
2) Fades an LED to User's Live HeartBeat
3) Determines BPM
4) Prints All of the Above to Serial
Read Me:
https://github.com/WorldFamousElectronics/PulseSensor_Amped_Arduino/blob/master/README.md   
Updated with HRV display by Elliot Mebane, Roguish, Inc. Feb 2016
HRV requires 1 minute of data before it's accurate. 
 ----------------------       ----------------------  ----------------------
*/  
LiquidCrystal_I2C lcd(0x27, 16, 2);
//  Variables
int pulsePin = 0;                 // Pulse Sensor purple wire connected to analog pin 0
int blinkPin = 13;                // pin to blink led at each beat
int fadePin = 5;                  // pin to do fancy classy fading blink at each beat
int fadeRate = 0;                 // used to fade LED on with PWM on fadePin

// Volatile Variables, used in the interrupt service routine!
volatile int BPM;                   // int that holds raw Analog in 0. updated every 2mS
volatile int Signal;                // holds the incoming raw data
volatile int IBI = 600;             // int that holds the time interval between beats! Must be seeded! 
volatile boolean Pulse = false;     // "True" when User's live heartbeat is detected. "False" when not a "live beat". 
volatile boolean QS = false;        // becomes true when Arduoino finds a beat.

volatile int RRDelta = 0;
volatile double RMSSD = 0;
volatile double LN20RMSSD = 0;
volatile double SDNN = 0;
volatile double SDSD = 0;
volatile double NN50 = 0;
volatile double PNN50 = 0;

// Regards Serial OutPut  -- Set This Up to your needs
static boolean serialVisual = false;   // Set to 'false' by Default.  Re-set to 'true' to see Arduino Serial Monitor ASCII Visual Pulse 

void setup(){
  pinMode(blinkPin,OUTPUT);         // pin that will blink to your heartbeat!
  pinMode(fadePin,OUTPUT);          // pin that will fade to your heartbeat!
  Serial.begin(115200);             // we agree to talk fast!
  interruptSetup();                 // sets up to read Pulse Sensor signal every 2mS 
  lcd.begin();
   // IF YOU ARE POWERING The Pulse Sensor AT VOLTAGE LESS THAN THE BOARD VOLTAGE, 
   // UN-COMMENT THE NEXT LINE AND APPLY THAT VOLTAGE TO THE A-REF PIN
//   analogReference(EXTERNAL);   
}

//  Where the Magic Happens
void loop(){
  
    serialOutput() ;       
    
  if (QS == true){     // A Heartbeat Was Found
                       // BPM and IBI have been Determined
                       // Quantified Self "QS" true when arduino finds a heartbeat
        fadeRate = 255;         // Makes the LED Fade Effect Happen
                                // Set 'fadeRate' Variable to 255 to fade LED with pulse
        serialOutputWhenBeatHappens();   // A Beat Happened, Output that to serial.     
        lcd.clear();
        lcd.print(BPM);
        QS = false;                      // reset the Quantified Self flag for next time    
  }
     
  ledFadeToBeat();                      // Makes the LED Fade Effect Happen 
  delay(20);                             //  take a break
}

void ledFadeToBeat(){
    fadeRate -= 15;                         //  set LED fade value
    fadeRate = constrain(fadeRate,0,255);   //  keep LED fade value from going into negative numbers!
    analogWrite(fadePin,fadeRate);          //  fade LED
  }

i use pulse + esp8266 in arduino uno

lukaskris, 2016-07-30:
My ESP connect with my hotspot mobile, and my pulse like this :(
how to fix this im confuse
two week ago, my esp just on, not join with my android hotspot

[image]

lukaskris, 2016-07-30:
in 3.3v, i think my pulse work well, before i use 5v

biomurph, 2016-07-30:
First, are you powering your arduino with 5V or 3.3V?
Are you powering your Pulse Sensor with 5V or 3.3V?

It looks like the signal is offset due to a mismatch between the Pulse Sensor supply and the Analog Reference...

Second, there is alot of weird noise in your signal...

lukaskris, 2016-07-30:
[image]

this is using 3.3v, before use 5v
im problem with voltage :v

biomurph, 2016-07-30:
are you adding the line 
analogReference(EXTERNAL);
in your arduino setup()?

This pulse wave looks better than the last... Can you try on different parts of your body? What part of your body are you using?

lukaskris, 2016-07-30:
this is source code pulse sesnor amped v 1.4

#include <LiquidCrystal_I2C.h>
/*  Pulse Sensor Amped 1.4    by Joel Murphy and Yury Gitman   http://www.pulsesensor.com
----------------------  Notes ----------------------  ---------------------- 
This code:
1) Blinks an LED to User's Live Heartbeat   PIN 13
2) Fades an LED to User's Live HeartBeat
3) Determines BPM
4) Prints All of the Above to Serial
Read Me:
https://github.com/WorldFamousElectronics/PulseSensor_Amped_Arduino/blob/master/README.md   
Updated with HRV display by Elliot Mebane, Roguish, Inc. Feb 2016
HRV requires 1 minute of data before it's accurate. 
 ----------------------       ----------------------  ----------------------
*/  
LiquidCrystal_I2C lcd(0x27, 16, 2);
//  Variables
int pulsePin = 0;                 // Pulse Sensor purple wire connected to analog pin 0
int blinkPin = 13;                // pin to blink led at each beat
int fadePin = 5;                  // pin to do fancy classy fading blink at each beat
int fadeRate = 0;                 // used to fade LED on with PWM on fadePin

// Volatile Variables, used in the interrupt service routine!
volatile int BPM;                   // int that holds raw Analog in 0. updated every 2mS
volatile int Signal;                // holds the incoming raw data
volatile int IBI = 600;             // int that holds the time interval between beats! Must be seeded! 
volatile boolean Pulse = false;     // "True" when User's live heartbeat is detected. "False" when not a "live beat". 
volatile boolean QS = false;        // becomes true when Arduoino finds a beat.

volatile int RRDelta = 0;
volatile double RMSSD = 0;
volatile double LN20RMSSD = 0;
volatile double SDNN = 0;
volatile double SDSD = 0;
volatile double NN50 = 0;
volatile double PNN50 = 0;

// Regards Serial OutPut  -- Set This Up to your needs
static boolean serialVisual = false;   // Set to 'false' by Default.  Re-set to 'true' to see Arduino Serial Monitor ASCII Visual Pulse 

void setup(){
  pinMode(blinkPin,OUTPUT);         // pin that will blink to your heartbeat!
  pinMode(fadePin,OUTPUT);          // pin that will fade to your heartbeat!
  Serial.begin(115200);             // we agree to talk fast!
  interruptSetup();                 // sets up to read Pulse Sensor signal every 2mS 
  lcd.begin();
   // IF YOU ARE POWERING The Pulse Sensor AT VOLTAGE LESS THAN THE BOARD VOLTAGE, 
   // UN-COMMENT THE NEXT LINE AND APPLY THAT VOLTAGE TO THE A-REF PIN
//   analogReference(EXTERNAL);   
}

//  Where the Magic Happens
void loop(){
  
    serialOutput() ;       
    
  if (QS == true){     // A Heartbeat Was Found
                       // BPM and IBI have been Determined
                       // Quantified Self "QS" true when arduino finds a heartbeat
        fadeRate = 255;         // Makes the LED Fade Effect Happen
                                // Set 'fadeRate' Variable to 255 to fade LED with pulse
        serialOutputWhenBeatHappens();   // A Beat Happened, Output that to serial.     
        lcd.clear();
        lcd.print(BPM);
        QS = false;                      // reset the Quantified Self flag for next time    
  }
     
  ledFadeToBeat();                      // Makes the LED Fade Effect Happen 
  delay(20);                             //  take a break
}

void ledFadeToBeat(){
    fadeRate -= 15;                         //  set LED fade value
    fadeRate = constrain(fadeRate,0,255);   //  keep LED fade value from going into negative numbers!
    analogWrite(fadePin,fadeRate);          //  fade LED
  }

i didnt change anything
just change on 5v to 3.3v 
in 5v so crazy my beat
i not touch but still beat

lukaskris, 2016-07-30:
this is on 5v
[image]

biomurph, 2016-07-30:
There's a couple of things that could be going on.
If you are using a laptop and have the power cord connected, that could be adding a bunch of noise to the USB power, which you could see in the signal at 5V. the 3.3V power is abit cleaner, because the board is regulating it. 
If you want to power your Pulse Sensor at 3.3V, and keep your board powered by the 5V USB, you must uncomment the line
analogReference(EXTERNAL);
and also apply the 3.3V to the pin labeled AREF (above pin13)
That way, your analog to digital converter will convert at the correct scale.

lukaskris, 2016-07-30:
i uncomment analogreference, on 3.3v didnt work, 
[image]

if i use 3.3v, my esp on 3,3 too, my pulse same like im using 5v

biomurph, 2016-07-30:
Try taking your LCD. First, just unplug it and see if it makes a difference.
Also, make sure you are setting up the analogReference correctly. check here https://www.arduino.cc/en/Reference/AnalogReference
also this is not a bad tutorial
http://tronixstuff.com/2013/12/12/arduino-tutorials-chapter-22-aref-pin/

carnal0wnage, 2016-11-03:
just to add a comment since i was having a similar problem. with no finger on the sensor its currently doing this for me, but when lightly putting my finger on the sensor its reading properly at 5v.

yury-g, 2026-08-16:
This repository has been retired as a read-only historical archive. Its code and discussion remain preserved, but fixes and support no longer happen here.

For current Arduino installation, examples, supported boards, and source, use [PulseSensorPlayground](https://github.com/WorldFamousElectronics/PulseSensorPlayground). If a problem is reproducible with the current Playground release, open a new issue there with the board and core versions, library version, wiring and supply voltage, raw waveform, and a minimal sketch.

Closing this historical issue as not planned in this retired repository.

closed 2026-08-16 by yury-g