WorldFamousElectronics/PulseSensor_Amped_Arduino #28: Pulse sensor didn't work correctly

baogia00123, 2016-08-07:
My pulse sensor didn't work correctly (problem can be seen in my picture)
[image]
I have deleted a tab name Timer_Interrupt_Notes because if i keep this tab, Arduino IDE won't run
All code is the same with your example code.
I hope you can help me.

biomurph, 2016-08-08:
Hi,

I need more information. Please run our Visualization software to have a look at the pulse wave signal. 
Are you holding the sensor on your fingertip? What OS are you using?

deepakKC123, 2016-11-10:
i have the same issue , it is displaying as s3 preceeded by something , wat to do..... i am using windows 10 os ...is dat the problem of my os ?

baogia00123, 2016-11-10:
It isn't error, you just need to change serialVisual to true. After that,
you will see your heart beat

deepakKC123, 2016-11-10:
how to get change it to serial visual ,please help me this thing is part of my final year engineering project ..am not much familiar in arduino codes...

chinmoyrick, 2016-11-11:
In this case i also suffering same problem ..what ever i read the sketch.
i think .. analog data not receive properly. or time inter-apt function is not proper way ..so if incoming data not match with hart bit value witch included in sketch . i test it with arduino uno mega micro  board all result are same.
 i don't have any way out how to know analog data receive ..

yury-g, 2026-08-16:
This repository has been retired as a read-only historical archive. Its code and discussion remain preserved, but fixes and support no longer happen here.

For current Arduino installation, examples, supported boards, and source, use [PulseSensorPlayground](https://github.com/WorldFamousElectronics/PulseSensorPlayground). If a problem is reproducible with the current Playground release, open a new issue there with the board and core versions, library version, wiring and supply voltage, raw waveform, and a minimal sketch.

Closing this historical issue as not planned in this retired repository.

closed 2026-08-16 by yury-g