WorldFamousElectronics/PulseSensor_Amped_Arduino #29: Code would not load - errors

project-science, 2016-11-03:
Hello,

I've tried to run the code as is.  There are several errors I've seen trying to run it.  (I'm using an Arduino UNO).

Most of the errors seem to be that variables and pins were not defined.

Variables not 'declared in this scope':

Signal
IBI
Pulse
BPM
QS

Pins which needs to be declared:

blinkPin
pulsePin

I tried to create the variables, and added a void setup() for the pinModes, and attached is the error which I got.

Perhaps I'm not doing this correctly, but looking for some help.

Thanks.

`/*
 * Pulse Sensor code, from:
 * https://github.com/WorldFamousElectronics/PulseSensor_Amped_Arduino/commit/88c1548d77f43464d3719a3a83df067fdf7ebea8
 * 
 */

// My additions
int Signal = 0;
int IBI = 0;
int Pulse = 0;
int BPM = 0;
int pulsePin = A5;
int blinkPin = 13;
boolean QS;

// My additions
void setup()
{
  pinMode(pulsePin, INPUT);
  pinMode(blinkPin, OUTPUT);
}

volatile int rate[10];                    // array to hold last ten IBI values
volatile unsigned long sampleCounter = 0;          // used to determine pulse timing
volatile unsigned long lastBeatTime = 0;           // used to find IBI
volatile int P =512;                      // used to find peak in pulse wave, seeded
volatile int T = 512;                     // used to find trough in pulse wave, seeded
volatile int thresh = 525;                // used to find instant moment of heart beat, seeded
volatile int amp = 100;                   // used to hold amplitude of pulse waveform, seeded
volatile boolean firstBeat = true;        // used to seed rate array so we startup with reasonable BPM
volatile boolean secondBeat = false;      // used to seed rate array so we startup with reasonable BPM

void interruptSetup(){     
  // Initializes Timer2 to throw an interrupt every 2mS.
  TCCR2A = 0x02;     // DISABLE PWM ON DIGITAL PINS 3 AND 11, AND GO INTO CTC MODE
  TCCR2B = 0x06;     // DON'T FORCE COMPARE, 256 PRESCALER 
  OCR2A = 0X7C;      // SET THE TOP OF THE COUNT TO 124 FOR 500Hz SAMPLE RATE
  TIMSK2 = 0x02;     // ENABLE INTERRUPT ON MATCH BETWEEN TIMER2 AND OCR2A
  sei();             // MAKE SURE GLOBAL INTERRUPTS ARE ENABLED      
} 

// THIS IS THE TIMER 2 INTERRUPT SERVICE ROUTINE. 
// Timer 2 makes sure that we take a reading every 2 miliseconds
ISR(TIMER2_COMPA_vect){                         // triggered when Timer2 counts to 124
  cli();                                      // disable interrupts while we do this
  Signal = analogRead(pulsePin);              // read the Pulse Sensor 
  sampleCounter += 2;                         // keep track of the time in mS with this variable
  int N = sampleCounter - lastBeatTime;       // monitor the time since the last beat to avoid noise

    //  find the peak and trough of the pulse wave
  if(Signal < thresh && N > (IBI/5)*3){       // avoid dichrotic noise by waiting 3/5 of last IBI
    if (Signal < T){                        // T is the trough
      T = Signal;                         // keep track of lowest point in pulse wave 
    }
  }

  if(Signal > thresh && Signal > P){          // thresh condition helps avoid noise
    P = Signal;                             // P is the peak
  }                                        // keep track of highest point in pulse wave

  //  NOW IT'S TIME TO LOOK FOR THE HEART BEAT
  // signal surges up in value every time there is a pulse
  if (N > 250){                                   // avoid high frequency noise
    if ( (Signal > thresh) && (Pulse == false) && (N > (IBI/5)*3) ){        
      Pulse = true;                               // set the Pulse flag when we think there is a pulse
      digitalWrite(blinkPin,HIGH);                // turn on pin 13 LED
      IBI = sampleCounter - lastBeatTime;         // measure time between beats in mS
      lastBeatTime = sampleCounter;               // keep track of time for next pulse

      if(secondBeat){                        // if this is the second beat, if secondBeat == TRUE
        secondBeat = false;                  // clear secondBeat flag
        for(int i=0; i<=9; i++){             // seed the running total to get a realisitic BPM at startup
          rate[i] = IBI;                      
        }
      }

      if(firstBeat){                         // if it's the first time we found a beat, if firstBeat == TRUE
        firstBeat = false;                   // clear firstBeat flag
        secondBeat = true;                   // set the second beat flag
        sei();                               // enable interrupts again
        return;                              // IBI value is unreliable so discard it
      }   

      // keep a running total of the last 10 IBI values
      word runningTotal = 0;                  // clear the runningTotal variable    

      for(int i=0; i<=8; i++){                // shift data in the rate array
        rate[i] = rate[i+1];                  // and drop the oldest IBI value 
        runningTotal += rate[i];              // add up the 9 oldest IBI values
      }

      rate[9] = IBI;                          // add the latest IBI to the rate array
      runningTotal += rate[9];                // add the latest IBI to runningTotal
      runningTotal /= 10;                     // average the last 10 IBI values 
      BPM = 60000/runningTotal;               // how many beats can fit into a minute? that's BPM!
      QS = true;                              // set Quantified Self flag 
      // QS FLAG IS NOT CLEARED INSIDE THIS ISR
    }                       
  }

  if (Signal < thresh && Pulse == true){   // when the values are going down, the beat is over
    digitalWrite(blinkPin,LOW);            // turn off pin 13 LED
    Pulse = false;                         // reset the Pulse flag so we can do it again
    amp = P - T;                           // get amplitude of the pulse wave
    thresh = amp/2 + T;                    // set thresh at 50% of the amplitude
    P = thresh;                            // reset these for next time
    T = thresh;
  }

  if (N > 2500){                           // if 2.5 seconds go by without a beat
    thresh = 512;                          // set thresh default
    P = 512;                               // set P default
    T = 512;                               // set T default
    lastBeatTime = sampleCounter;          // bring the lastBeatTime up to date        
    firstBeat = true;                      // set these to avoid noise
    secondBeat = false;                    // when we get the heartbeat back
  }

  sei();                                   // enable interrupts when youre done!
}// end isr`

biomurph, 2016-12-23:
Are you still having trouble?
It looks like you may not be running the entire sketch. You need to have all of the tabs
Pulse_Sensor_Amped_Arduino_1dot4.ino
Interrupts.ino
AllSerialHandling.ino

ambleskramble, 2017-02-24:
Hi there

I'm getting similar problems. I have all three tabs done, but when I upload it gives me:

Arduino: 1.6.3 (Mac OS X), Board: "Arduino Uno"

/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10603 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard /var/folders/mk/hk2wx6k90_74xx2l2v62nnmc0000gn/T/build4000576784421979804.tmp/sketch_feb24c.cpp -o /var/folders/mk/hk2wx6k90_74xx2l2v62nnmc0000gn/T/build4000576784421979804.tmp/sketch_feb24c.cpp.o 
sketch_feb24c.ino: In function 'void serialOutput()':
sketch_feb24c.ino:9:10: error: 'outputType' was not declared in this scope
sketch_feb24c.ino:10:10: error: 'PROCESSING_VISUALIZER' was not declared in this scope
sketch_feb24c.ino:11:29: error: 'Signal' was not declared in this scope
sketch_feb24c.ino:13:10: error: 'SERIAL_PLOTTER' was not declared in this scope
sketch_feb24c.ino:14:20: error: 'BPM' was not declared in this scope
sketch_feb24c.ino:16:20: error: 'IBI' was not declared in this scope
sketch_feb24c.ino: In function 'void serialOutputWhenBeatHappens()':
sketch_feb24c.ino:28:10: error: 'outputType' was not declared in this scope
sketch_feb24c.ino:29:10: error: 'PROCESSING_VISUALIZER' was not declared in this scope
sketch_feb24c.ino:30:28: error: 'BPM' was not declared in this scope
sketch_feb24c.ino:31:28: error: 'IBI' was not declared in this scope
interupt.ino: In function 'void __vector_7()':
interupt.ino:29:3: error: 'Signal' was not declared in this scope
interupt.ino:29:23: error: 'pulsePin' was not declared in this scope
interupt.ino:34:30: error: 'IBI' was not declared in this scope
interupt.ino:47:32: error: 'Pulse' was not declared in this scope
interupt.ino:47:57: error: 'IBI' was not declared in this scope
interupt.ino:49:20: error: 'blinkPin' was not declared in this scope
interupt.ino:79:7: error: 'BPM' was not declared in this scope
interupt.ino:80:7: error: 'QS' was not declared in this scope
interupt.ino:85:26: error: 'Pulse' was not declared in this scope
interupt.ino:86:18: error: 'blinkPin' was not declared in this scope
Error compiling.

Seems like a lot of errors happening with this code.

joelSensor, 2017-02-24:
It looks like you need to update your Arduino. They are at v1.8.1 now.
Can you share the code you're using? It does seem like you are missing some important elements of the code

ambleskramble, 2017-02-24:
Thanks :-)

So tab 1 is:

// Variables
int pulsePin = 0; // Pulse Sensor purple wire connected to analog pin 0
int blinkPin = 13; // pin to blink led at each beat
int fadePin = 5; // pin to do fancy classy fading blink at each beat
int fadeRate = 0; // used to fade LED on with PWM on fadePin

// Volatile Variables, used in the interrupt service routine!
volatile int BPM; // int that holds raw Analog in 0. updated every 2mS
volatile int Signal; // holds the incoming raw data
volatile int IBI = 600; // int that holds the time interval between beats! Must be seeded!
volatile boolean Pulse = false; // "True" when User’s live heartbeat is detected. "False" when not a "live beat".
volatile boolean QS = false; // becomes true when Arduoino finds a beat.

// Regards Serial OutPut — Set This Up to your needs
static boolean serialVisual = true; // Set to ‘false’ by Default. Re-set to ‘true’ to see Arduino Serial Monitor ASCII Visual Pulse

void setup(){
pinMode(blinkPin,OUTPUT); // pin that will blink to your heartbeat!
pinMode(fadePin,OUTPUT); // pin that will fade to your heartbeat!
Serial.begin(115200); // we agree to talk fast!
interruptSetup(); // sets up to read Pulse Sensor signal every 2mS
// UN-COMMENT THE NEXT LINE IF YOU ARE POWERING The Pulse Sensor AT LOW VOLTAGE,
// AND APPLY THAT VOLTAGE TO THE A-REF PIN
// analogReference(EXTERNAL);
}

// Where the Magic Happens
void loop(){
serialOutput() ;

if (QS == true){ // A Heartbeat Was Found
// BPM and IBI have been Determined
// Quantified Self "QS" true when arduino finds a heartbeat
digitalWrite(blinkPin,HIGH); // Blink LED, we got a beat.
fadeRate = 255; // Makes the LED Fade Effect Happen
// Set ‘fadeRate’ Variable to 255 to fade LED with pulse
serialOutputWhenBeatHappens(); // A Beat Happened, Output that to serial.
QS = false; // reset the Quantified Self flag for next time
}
else {

digitalWrite(blinkPin,LOW); // There is not beat, turn off pin 13 LED
}

ledFadeToBeat(); // Makes the LED Fade Effect Happen
delay(20); // take a break
}

TAB 2: 

//////////
/////////  All Serial Handling Code,
/////////  It's Changeable with the 'outputType' variable
/////////  It's declared at start of code.
/////////

void serialOutput(){   // Decide How To Output Serial.
  switch(outputType){
    case PROCESSING_VISUALIZER:
      sendDataToSerial('S', Signal);     // goes to sendDataToSerial function
      break;
    case SERIAL_PLOTTER:  // open the Arduino Serial Plotter to visualize these data
      Serial.print(BPM);
      Serial.print(",");
      Serial.print(IBI);
      Serial.print(",");
[... 102 lines cut ...]
      QS = true;                              // set Quantified Self flag
      // QS FLAG IS NOT CLEARED INSIDE THIS ISR
    }
  }

  if (Signal < thresh && Pulse == true){   // when the values are going down, the beat is over
    digitalWrite(blinkPin,LOW);            // turn off pin 13 LED
    Pulse = false;                         // reset the Pulse flag so we can do it again
    amp = P - T;                           // get amplitude of the pulse wave
    thresh = amp/2 + T;                    // set thresh at 50% of the amplitude
    P = thresh;                            // reset these for next time
    T = thresh;
  }

  if (N > 2500){                           // if 2.5 seconds go by without a beat
    thresh = 530;                          // set thresh default
    P = 512;                               // set P default
    T = 512;                               // set T default
    lastBeatTime = sampleCounter;          // bring the lastBeatTime up to date
    firstBeat = true;                      // set these to avoid noise
    secondBeat = false;                    // when we get the heartbeat back
  }

  sei();                                   // enable interrupts when youre done!
}// end isr

ambleskramble, 2017-02-24:
Idea lly I just need something which will send the raw data to max. 
I'm trying to use Maxuino but so far nothing is coming through. 

This is the error message when I use the 3 above now:

/Applications/Arduino 2.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino 2.app/Contents/Java/hardware -tools /Applications/Arduino 2.app/Contents/Java/tools-builder -tools /Applications/Arduino 2.app/Contents/Java/hardware/tools/avr -built-in-libraries /Applications/Arduino 2.app/Contents/Java/libraries -libraries /Users/ambleskuse/Documents/Arduino/libraries -fqbn=arduino:avr:uno -vid-pid=0X2A03_0X0043 -ide-version=10801 -build-path /var/folders/mk/hk2wx6k90_74xx2l2v62nnmc0000gn/T/arduino_build_666323 -warnings=null -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/Applications/Arduino 2.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/Applications/Arduino 2.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/Applications/Arduino 2.app/Contents/Java/hardware/tools/avr -verbose /Users/ambleskuse/Documents/Arduino/sketch_feb24d/sketch_feb24d.ino
/Applications/Arduino 2.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino 2.app/Contents/Java/hardware -tools /Applications/Arduino 2.app/Contents/Java/tools-builder -tools /Applications/Arduino 2.app/Contents/Java/hardware/tools/avr -built-in-libraries /Applications/Arduino 2.app/Contents/Java/libraries -libraries /Users/ambleskuse/Documents/Arduino/libraries -fqbn=arduino:avr:uno -vid-pid=0X2A03_0X0043 -ide-version=10801 -build-path /var/folders/mk/hk2wx6k90_74xx2l2v62nnmc0000gn/T/arduino_build_666323 -warnings=null -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/Applications/Arduino 2.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/Applications/Arduino 2.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/Applications/Arduino 2.app/Contents/Java/hardware/tools/avr -verbose /Users/ambleskuse/Documents/Arduino/sketch_feb24d/sketch_feb24d.ino
Using board 'uno' from platform in folder: /Applications/Arduino 2.app/Contents/Java/hardware/arduino/avr
Using core 'arduino' from platform in folder: /Applications/Arduino 2.app/Contents/Java/hardware/arduino/avr
Detecting libraries used...
"/Applications/Arduino 2.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino 2.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino 2.app/Contents/Java/hardware/arduino/avr/variants/standard" "/var/folders/mk/hk2wx6k90_74xx2l2v62nnmc0000gn/T/arduino_build_666323/sketch/sketch_feb24d.ino.cpp" -o "/dev/null"
Generating function prototypes...
"/Applications/Arduino 2.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino 2.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino 2.app/Contents/Java/hardware/arduino/avr/variants/standard" "/var/folders/mk/hk2wx6k90_74xx2l2v62nnmc0000gn/T/arduino_build_666323/sketch/sketch_feb24d.ino.cpp" -o "/var/folders/mk/hk2wx6k90_74xx2l2v62nnmc0000gn/T/arduino_build_666323/preproc/ctags_target_for_gcc_minus_e.cpp"
"/Applications/Arduino 2.app/Contents/Java/tools-builder/ctags/5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/var/folders/mk/hk2wx6k90_74xx2l2v62nnmc0000gn/T/arduino_build_666323/preproc/ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"/Applications/Arduino 2.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os  -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino 2.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino 2.app/Contents/Java/hardware/arduino/avr/variants/standard" "/var/folders/mk/hk2wx6k90_74xx2l2v62nnmc0000gn/T/arduino_build_666323/sketch/sketch_feb24d.ino.cpp" -o "/var/folders/mk/hk2wx6k90_74xx2l2v62nnmc0000gn/T/arduino_build_666323/sketch/sketch_feb24d.ino.cpp.o"
sketch_feb24d:1: error: expected unqualified-id before '/' token
 */
  ^
sketch_feb24d:1: error: expected constructor, destructor, or type conversion before '/' token
/Users/ambleskuse/Documents/Arduino/sketch_feb24d/sketch_feb24d.ino: In function 'void loop()':
sketch_feb24d:47: error: 'ledFadeToBeat' was not declared in this scope
 ledFadeToBeat(); // Makes the LED Fade Effect Happen
               ^
/Users/ambleskuse/Documents/Arduino/sketch_feb24d/amped_arduino.ino: In function 'void serialOutput()':
amped_arduino:9: error: 'outputType' was not declared in this scope
   switch(outputType){
          ^
amped_arduino:10: error: 'PROCESSING_VISUALIZER' was not declared in this scope
     case PROCESSING_VISUALIZER:
          ^
amped_arduino:13: error: 'SERIAL_PLOTTER' was not declared in this scope
[... cut ...]
amped_arduino:29: error: 'PROCESSING_VISUALIZER' was not declared in this scope
     case PROCESSING_VISUALIZER:    // find it here https://github.com/WorldFamousElectronics/PulseSensor_Amped_Processing_Visualizer
          ^
/Users/ambleskuse/Documents/Arduino/sketch_feb24d/interupt.ino: In function 'void __vector_7()':
interupt:29: error: 'pulsePin' was not declared in this scope
   Signal = analogRead(pulsePin);              // read the Pulse Sensor
                       ^
exit status 1
expected unqualified-id before '/' token

tangtangasia, 2020-12-30:
hello .. I have an error problem like this:

esp32pulsesensorthingspeak:15:13: error: 'pulseSensor' was not declared in this scope
 int myBPM = pulseSensor.getBeatsPerMinute();
             ^
esp32pulsesensorthingspeak:17:1: error: expected unqualified-id before 'if'
 if (myBPM < 100 && myBPM > 50)
 ^
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\asus\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\WiFi
exit status 1
'pulseSensor' was not declared in this scope

and this is the code that I made, I use esp32 microcontroller, and pulse sensor. Where did I go wrong? I am just a beginner, please help me study.

#define USE_ARDUINO_INTERRUPTS false
#include <PulseSensorPlayground.h>
#include <WiFi.h>
#include <ThingSpeak.h>

const char ssid[] = "mydream";
const char password[]= "12122234";
WiFiClient client;

const long CHANNEL = 1273034;
const char *WRITE_API = "A3ZUCRHUSPNDSAR8W";

long prevMillisThingSpeak = 0;
int intervalThingSpeak = 20000;
int myBPM = pulseSensor.getBeatsPerMinute();

if (myBPM < 100 && myBPM > 50)
  if (millis() - prevMillisThingSpeak > intervalThingSpeak){
    ThingSpeak.setField(1,myBPM);

    int x = ThingSpeak.writeFields(CHANNEL, WRITE_API);
    if (x==200){
      Serial.println ("Channel update seccessful.")
    }
    else{
      Serial.println("Problem updating channel. HTTP error code "+String (x));
    }
    prevMillisThingSpeak = millis();
  }
}

biomurph, 2020-12-30:
@tangtangasia 
You are commenting on a repo that is not supported. 
Are you using the PulseSensor Playground Library? If so, please move your issue to that repo.

Is that all the code you have? If so, you are definitely doing it wrong. 
Please start by downloading the PulseSensor Playground library in the Arduino IDE.
Follow the getting started guide here
https://pulsesensor.com/pages/installing-our-playground-for-pulsesensor-arduino

yury-g, 2026-08-16:
This repository has been retired as a read-only historical archive. Its code and discussion remain preserved, but fixes and support no longer happen here.

For current Arduino installation, examples, supported boards, and source, use [PulseSensorPlayground](https://github.com/WorldFamousElectronics/PulseSensorPlayground). If a problem is reproducible with the current Playground release, open a new issue there with the board and core versions, library version, wiring and supply voltage, raw waveform, and a minimal sketch.

Closing this historical issue as not planned in this retired repository.

closed 2026-08-16 by yury-g