WorldFamousElectronics/PulseSensor_Amped_Arduino #30: No signal in ASCII Monitor

alecz1k, 2016-11-13:
Hello there, 

I ordered the pulse sensor from the official website. 
I run windows 10. I downloaded the code Pulse Sensor Amped Arduino from pulse sensor/GitHub.
I did the modification (static boolean serialVisual = true;) but when I run the Serial Monitor at 115200 baud I have nothing showing up. If I change baud, then I have random value showing up. 

In Processing, I have no signal even if the Arduino's port is recognized (I previously changed "static boolean serialVisual = true" back to false). 

What  am I doing wrong? 
Thank you for your help!

screenshots: 
flic.kr/p/NUyedW
flic.kr/p/NUyeeY

biomurph, 2016-12-23:
Hi,
did you upload the code to you're board?

alecz1k, 2016-12-23:
that was it,
thank you for your reply!

closed 2016-12-23 by alecz1k