WorldFamousElectronics/PulseSensor_Amped_Arduino #31: Timer0 instead of Timer2

frugallabs, 2016-11-28:
Hello,
Can someone help me out in implementing the interrupt on Timer0 instead of Timer1 or Timer2.

I have a WiFi module connected to pins 9,10,11 of the Arduino Nano. These pins are fixed on myt PCB and cannot be replaced.

So I am unable to use the WiFi module if I use Timer1 or Timer2 as my WiFi does not work. The WiFi module will send the pulse (BPM) to the cloud.

Can anyone help me out in implementing Timer0 for the code? I do not mind compromising on delay().

biomurph, 2016-11-28:
It's recommended that you don't use Timer0 on the ATmega 328 boards because Timer0 is used for the delay() and delayMicroseconds() and I think the millis() and micros() as well.
Try using a 'software interrupt' instead. Here's a short example:

```
unsigned long lastTime;
unsigned long thisTime;

setup(){
lastTime = micros();
}

loop(){
thisTime = micros();
if(thisTime - lastTime > 2000){
lastTime = thisTime;
//goto the read-sensor-and-get-pulse routine
}
// do the other stuff you want to do
} 
```

Hope that helps!

yury-g, 2026-08-16:
This repository has been retired as a read-only historical archive. Its code and discussion remain preserved, but fixes and support no longer happen here.

For current Arduino installation, examples, supported boards, and source, use [PulseSensorPlayground](https://github.com/WorldFamousElectronics/PulseSensorPlayground). If a problem is reproducible with the current Playground release, open a new issue there with the board and core versions, library version, wiring and supply voltage, raw waveform, and a minimal sketch.

Closing this historical issue as not planned in this retired repository.

closed 2026-08-16 by yury-g