WorldFamousElectronics/PulseSensor_Amped_Arduino #32: Error while trying to run default code from nodemcu Esp8266

dotjavafile, 2016-12-01:
Hello I am trying to run the default code provided in the tutorial in nodemcu v1 esp-12e. 

github.com/WorldFamousElectronics/PulseSensor_Amped_Arduino

It runs in my arduino uno fine but when I try it with the nodemcu I get following error. Please help.

Read more: http://pulsesensor.proboards.com/thread/474/error-trying-default-code-nodemcu#ixzz4Rbo00iGf

`Interrupt:27: error: expected constructor, destructor, or type conversion before '(' token

ISR(TIMER2_COMPA_vect){ // triggered when Timer2 counts to 124

^

C:\Users\Zero\Documents\Arduino\PulseSensorAmped_Arduino_1dot4\Interrupt.ino: In function 'void interruptSetup()':

Interrupt:17: error: 'TCCR2A' was not declared in this scope

TCCR2A = 0x02; // DISABLE PWM ON DIGITAL PINS 3 AND 11, AND GO INTO CTC MODE

^

Interrupt:18: error: 'TCCR2B' was not declared in this scope

TCCR2B = 0x06; // DON'T FORCE COMPARE, 256 PRESCALER 

^

Interrupt:19: error: 'OCR2A' was not declared in this scope

OCR2A = 0X7C; // SET THE TOP OF THE COUNT TO 124 FOR 500Hz SAMPLE RATE

^

Interrupt:20: error: 'TIMSK2' was not declared in this scope

TIMSK2 = 0x02; // ENABLE INTERRUPT ON MATCH BETWEEN TIMER2 AND OCR2A

^

C:\Users\Zero\Documents\Arduino\PulseSensorAmped_Arduino_1dot4\Interrupt.ino: At global scope:

Interrupt:27: error: expected constructor, destructor, or type conversion before '(' token

ISR(TIMER2_COMPA_vect){ // triggered when Timer2 counts to 124

^

exit status 1
expected constructor, destructor, or type conversion before '(' token

Read more: http://pulsesensor.proboards.com/thread/474/error-trying-default-code-nodemcu#ixzz4Rbo3ndv4`

dotjavafile, 2016-12-02:
Please any help?

LeonFedotov, 2017-02-25:
Im also having this problem

hagelric, 2018-02-04:
Please any help?

Same error here!

biomurph, 2018-02-05:
This is an older repo. We are directing everyone to use our new PulseSensor Playground library
https://github.com/WorldFamousElectronics/PulseSensorPlayground

Please follow this tutorial
https://pulsesensor.com/pages/getting-advanced
Not all hardwares will work with our code that uses interrupts. Please read the entire tutorial and try the example code PulseSensor_BPM_Alternative.ino

Also, note that the ESP8266 ADC can only accept up to 1V max. I'm not sure how the nodemcu works this, but be aware of that.

yury-g, 2026-08-16:
This repository has been retired as a read-only historical archive. Its code and discussion remain preserved, but fixes and support no longer happen here.

For current Arduino installation, examples, supported boards, and source, use [PulseSensorPlayground](https://github.com/WorldFamousElectronics/PulseSensorPlayground). If a problem is reproducible with the current Playground release, open a new issue there with the board and core versions, library version, wiring and supply voltage, raw waveform, and a minimal sketch.

Closing this historical issue as not planned in this retired repository.

closed 2026-08-16 by yury-g