WorldFamousElectronics/PulseSensor_Amped_Arduino #33: False BPM-calculation on Arduino Mini Pro

MeisterPetz92, 2016-12-13:
Hi,
first i use the Arduino UNO for a project with an bluetooth-modul to android and so on ;-)
Now i want to transfer it to an Arduino Mini Pro 8MHz,3.3V.

  So i used the Settings:  
      void interruptSetup(){     
        // Initializes Timer2 to throw an interrupt every 2mS.
        TCCR2A = 0x02;     // DISABLE PWM ON DIGITAL PINS 3 AND 11, AND GO INTO CTC MODE
        TCCR2B = 0x05;     // DON'T FORCE COMPARE, 128 PRESCALER 
        OCR2A = 0X7C;      // SET THE TOP OF THE COUNT TO 124 FOR 500Hz SAMPLE RATE
        TIMSK2 = 0x02;     // ENABLE INTERRUPT ON MATCH BETWEEN TIMER2 AND OCR2A
        sei();             // MAKE SURE GLOBAL INTERRUPTS ARE ENABLED 

I get a pulse rate but it`s to high, when i have a pulse rate of 80 BPM, i get with the Arduino Mini Pro a value from 150 - 200 BPM.

Has someone of you similiar problems  or get running the code on the mini pro?
Thanks for help!

biomurph, 2016-12-13:
It looks like you have the right interrupt settings. Can you connect to the Processing visualizer to see the signal and verify that you are getting good data?

MeisterPetz92, 2016-12-13:
Thank you for the fast answer.
That`s life... i connected new and upload the code, try it with Processing and it works.
I have no idea what was the problem, but now it works fine ;-)
BR

closed 2016-12-13 by MeisterPetz92