WorldFamousElectronics/PulseSensor_Amped_Arduino #34: "download zip" on tutorial page links to "download zip" image

kaeverens, 2016-12-22:
on the page https://github.com/WorldFamousElectronics/PulseSensor_Amped_Arduino, there is an instruction:
1. Download ZIP on this page:  Download Zip

the "Download Zip" button links to the image for the button, not a zip file.

biomurph, 2016-12-23:
Hi kaeverens,

I just visited the repo, and the Download Zip link from the "Clone or Download" option does download on my computer lickety split. What kind of computer are you using? Do you have limitations on what you can download? 
If you continue to have trouble, contact the nice folks at github, since it's their site and all.

Joel

kaeverens, 2016-12-23:
I'm using Chrome on Linux, but that doesn't matter in this case.

the problem is that the HTML of the page literally links to an image, and not a zip.

Here is a screenshot explaining visually.

[image]

biomurph, 2016-12-23:
Oh, I see the problem! You are looking at our readme. That is an image of the button in the github page above. Scroll up in the browser window, and you will see the green button "Clone or Download" Click that, and you will see the option to clone the repo on your machine, or download the zip file.

Best,
Joel

kaeverens, 2016-12-24:
I understand that. But the README _doesn't say that_. It implies heavily that the way to download the zip is to click the Download Zip image. It says nothing about scrolling up, etc.

anyway... you get the problem now

yury-g, 2026-08-16:
Closing as resolved. The README installation instructions now explicitly direct users to the repository's **Clone or Download** control and no longer present the screenshot as the download link.

closed 2026-08-16 by yury-g