WorldFamousElectronics/PulseSensor_Amped_Arduino #35: Heartbeat data keeps increasing.

rohinrohin, 2016-12-30:
I am using the exact example code from the github repo on an arduino uno. 
The value just keeps increasing continuously. (seems to be some erratic readings?)

**Circuit:** 
ARDUINO => SENSOR
GND -> -
5v -> +
A0 -> SIG

Laptop is powered through the outlet (connected to charger). 

I have added a clear tape on top and bottom to shield from bad data. This data is collected as soon as the arduino is switched on with the sensor resting on the table, heart side up. 

Any help is appreciated. 

**Serial output :**
```
--------------|-
------------
------------
------------
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
[... 579 lines cut ...]
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
*** Heart-Beat Happened *** BPM: 209  --------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
--------------|-
*** Heart-Beat Happened *** BPM: 227  --------------|-
--------------|-
--------------|-
--------------|-
--------------|-

```

biomurph, 2016-12-30:
Hi 
Please try reprogramming the Arduino with the serialVisual boolean set to false, and also run the Processing Visualizer sketch in Processing so that you can see the signal quality.
If you can, send a screenshot.

I will update the Arduino code so that it will output to the Arduino visualizer soon, and add a comment here when I do so that you can try that as well.

Rohankp12, 2017-01-11:
I am having the exact same issue. Any help will be appreciated.

biomurph, 2017-01-11:
@Rohankp12 
Please try reprogramming the Arduino with the serialVisual boolean set to false, and also run the Processing Visualizer sketch in Processing so that you can see the signal quality.
If you can, send a screenshot.

Rohankp12, 2017-01-11:
I tried running it with serialVisual boolean set to false, it had the same issue

Rohankp12, 2017-01-12:
[image]
Here's a screenshot of Visualizer sketch.

yury-g, 2026-08-16:
This repository has been retired as a read-only historical archive. Its code and discussion remain preserved, but fixes and support no longer happen here.

For current Arduino installation, examples, supported boards, and source, use [PulseSensorPlayground](https://github.com/WorldFamousElectronics/PulseSensorPlayground). If a problem is reproducible with the current Playground release, open a new issue there with the board and core versions, library version, wiring and supply voltage, raw waveform, and a minimal sketch.

Closing this historical issue as not planned in this retired repository.

closed 2026-08-16 by yury-g