WorldFamousElectronics/PulseSensor_Amped_Arduino #36: arduino 101-->"quick and dirt solution"

ciapatlano, 2017-01-01:
The code modifications mentioned in the "Timer _Interrupt_Notes" tab is diplaying error when verifying code. Specifically the line with "getPulse()" instead of "ISR(TIMER2_COMPA_vect)". See attached image
[image]

biomurph, 2017-01-02:
Oh, I see the problem. You need to add the word 'void' in front of getPulse() when you define the function. I have updated the notes in the repo, and verified that it does compile and function.

ciapatlano, 2017-01-03:
thank you very much.
I will test it asap even if, in the mean time, I have found this:
[https://github.com/bneedhamia/PulseSensorBPM](url)
I let you check if it make sense to integrate that for 101.

closed 2017-01-03 by ciapatlano