WorldFamousElectronics/PulseSensor_Amped_Arduino #37: Heart rate sensor is working incorrectly

osmnerl, 2017-01-04:
The sensor shows the BPM values ​​without putting my finger on it but I put my finger on the sensor and the values ​​are disappearing. Please help me this is my project homework and very little time left

biomurph, 2017-01-04:
Hi,
You are not giving enough information to solve your problem.
Please upload the Arduino sketch with the variable 
`serialVisual = false;`
and then run the Processing sketch to see what the raw data looks like. You can take a screenshot and send it here. That will help.
When you put your finger on the Pulse Sensor, at first it may take a few seconds for the signal to settle.

PrakharDutta, 2018-09-17:
this is the raw signal my pulse sensor gives even when my finger isn't touching the sensor.
[image]

biomurph, 2018-09-17:
It looks like the Pulse Sensor signal is idling very high. 
Please send a picture of the Pulse Sensor that you have. 
It looks like you are using an Arduino UNO, is that right?

PrakharDutta, 2018-09-17:
yes, I'm using an arduino uno but i'm not entirely sure as to whether the sensor I have is legitimate
[image]

PrakharDutta, 2018-09-17:
[image]

biomurph, 2018-09-17:
I can tell by the color of the wires and the rough edges of the PCB that you have purchased yourself a clone Pulse Sensor. I cannot troubleshoot hardware problems with hardware that I don't know anything about.
Please support the original makers of Pulse Sensor and buy from a reputable source.
www.pulsesensor.com

closed 2018-09-17 by biomurph