WorldFamousElectronics/PulseSensor_Amped_Arduino #38: The code works well with Arduino Uno but not Arduino Pro Mini

diontan524, 2017-01-08:
Hi guys, I have downloaded the code and use it on my arduino uno and it works perfectly fine. When i said fine, it means the range of heartbeat is around 50 plus to 80 plus, and the heartbeat only be detected when i placed my finger on the sensor, else, the display just paused until I place my finger there again. However, when I transfer the same code to arduino pro mini, the heartbeat just randomly detected and displayed on serial monitor even though I did not place my finger there. However it gave correct reading when i placed my finger, the problem is just that it kept reading pulse even though there are none. It don't happen on arduino uno. So, actually which one is working fine ? I tried it with several arduino Unos and Arduino Pro minis, the Unos work fine, but not the Pro minis. Can anyone explain ?

biomurph, 2017-01-09:
Are you using an Arduino Pro Mini that runs on 5V or 3V?

I don't know what to tell you about how the Uno is showing less noise than the Pro Mini. But there is the potential for noise when you don't have your finger (or other body part) on the sensor. This problem could be cleared up with some code that rejects noise, or you could use another sensor that will tell the arduino when there is a finger (or other body part) on the sensor.

diontan524, 2017-01-09:
I'm using pro mini 3v. How to add code that reject noise ? I'm actually doing a project based on this link. 

https://openhomeautomation.net/bluetooth-heart-rate-sensor-arduino/

I asked there no one actually will reply so I asked here. The code is almost similar, except that he removes some lines that are not needed. However, I tried the codes downloaded here also showing the same problem. Another thing is does anyone know what is this node.js stuff ?

biomurph, 2017-01-09:
Since you are using a pro mini 3V, you will need to change the interrupt settings. Please read this page to see how you need to make the changes.
https://github.com/WorldFamousElectronics/PulseSensor_Amped_Arduino/blob/master/PulseSensorAmped_Arduino_1dot4/Timer_Interrupt_Notes.ino
That might help with the noise you're seeing.
Also, if your computer is plugged in, you will likely see some noise from the power supply.
I have some code here that does a simple job of noise cancellation. It measures the amplitude of the signal, and rejects 'pulses' that are below a threshold. It's based on our (soon to be released) v1.5 of the firmware.

diontan524, 2017-01-09:
The interrupt setting is just changing the line TCCR2B = 0x06; to TCCR2B = 0x05; right ? I did that already. still it gave the same problem ? I just don't understand why it don't happen on Uno but Pro mini 3V or 5V (tested both). I have to use pro mini because i will soon modify the hardware for lower power consumption.

biomurph, 2017-01-09:
Oh, sorry. I forgot to link to the code with noise cancelling.
Here it is
https://github.com/biomurph/Pulse-Sensor-Amped-1.5-Noise

diontan524, 2017-01-09:
well is it just the same that i transfer the code to pro mini, then turn on the visualizer and choose the port accordingly ? The visualizer is not showing value and display 0.

diontan524, 2017-01-09:
just no bpm displayed

biomurph, 2017-01-09:
Check the header on this new code. You may need to change the value of the variable 'outpuType' in order to use the visualizer. It also works with the Arduino Serial Plotter

yboujraf, 2021-06-02:
Dear @biomurph , @diontan524 , @joelSensor , @yurygitman , @yury-g 

Did you fixed the code to support the pulse sensor using an arduino pro mini 3V/8MHz ?

I have the same issue.

I am using port A1 for the Pulse Sensor.

Best Regards,

biomurph, 2021-06-02:
You are posting this issue in a repository that is not supported.
Please use the Pulse Sensor Playground LIbrary
https://github.com/WorldFamousElectronics/PulseSensorPlayground
It will work with Arduino Pro Mini no problem

closed 2021-06-02 by biomurph