WorldFamousElectronics/PulseSensor_Amped_Arduino #41: using pulse sensor with ATtiny85 chip

diontan524, 2017-02-19:
Hi there, I am planning to use the Pulse sensor on attiny85 chip. But looks like the code uploaded online have to be modified due to the fact that attiny85 does not have timer2 and serial communication. With added serialsoftware, I am able to serialprint words to the serial monitor with the Attiny85 chip, so im quite sure my code there has no problem. But for the timer2 to be modified as timer1, I am not sure the codes that I found online is correct or not. I tried to serial print the BPM but not luck. Can anyone help Please ?

diontan524, 2017-02-19:
[timer1 pulse rate.txt](https://github.com/WorldFamousElectronics/PulseSensor_Amped_Arduino/files/785991/timer1.pulse.rate.txt)

joelSensor, 2017-02-21:
Hi diontan524,

That looks great! I will give your code mod a test run on the ATtiny85 that I have, and add notes to our master branch (and credit you) when I get it working. Do you have your version posted on a repo somewhere?

diontan524, 2017-02-22:
Nope i have not post this anywhere else. Thank you guys and hope to hear
from u soon.

diontan524, 2017-03-01:
anyone has any solution yet ?

biomurph, 2017-03-01:
Hi,

I have an ATtiny, but it is on a Digitspark, and in order to use it on my mac I need an external USB hub thing. Please give me some time to get it all together.

In the meantime, you can use the example that is at the bottom of the Timer Interrupt Notes tab. That shows how to set up a software timer to sample the pulse sensor data.

diontan524, 2017-03-01:
I think it can use arduino to program it. That's the way I do it, by setting the arduino as ISP

biomurph, 2017-03-01:
Yes, I think that would work. My only ATtiny is attached to this digispark USB board. It already has the bootloader on it, etc.

diontan524, 2017-03-01:
I see... Is it doable? I found many websites already unfortunately non of
them solve this :'(

biomurph, 2017-03-01:
Is it possible for you to use the software timer?
I do know of someone who is using it. Have not been in touch with them in a while...

diontan524, 2017-03-01:
Is it a library ? I heard the previous pulse sensor code is run by using
timer 1 before it was changed to timer 2. Is it true ?

biomurph, 2017-03-01:
Please check the Timer Interrupt Notes tab in this repo. At the end, it describes how to modify the code to use a software timer. It is not a library.

joelSensor, 2017-03-03:
I have updated the Timer Interrupt Notes page here
https://github.com/WorldFamousElectronics/PulseSensor_Amped_Arduino/tree/master/PulseSensorAmped_Arduino_1.5.0
There are Interrupt setup notes for ATtiny 85 running at 8MHz and also at 16MHz. They reference the Adafruit GEMMA and TRINKET. They are tested and they work.

What speed master clock are you using?

I have an original Digispark, Kickstarter Edition board. It runs at 16.5MHz, which makes the timer interrupt kind of messy.

diontan524, 2017-03-03:
> tree/master/PulseSensorAmped_Arduino_1.5.0

ATtiny by default is 1Mhz, but I set and run it at 8Mhz

yury-g, 2026-08-16:
This repository has been retired as a read-only historical archive. Its code and discussion remain preserved, but fixes and support no longer happen here.

For current Arduino installation, examples, supported boards, and source, use [PulseSensorPlayground](https://github.com/WorldFamousElectronics/PulseSensorPlayground). If a problem is reproducible with the current Playground release, open a new issue there with the board and core versions, library version, wiring and supply voltage, raw waveform, and a minimal sketch.

Closing this historical issue as not planned in this retired repository.

closed 2026-08-16 by yury-g