WorldFamousElectronics/PulseSensor_Amped_Arduino #42: I'm getting weird numbers and outputs from the pulse sensor

quackamadoodleduck, 2017-02-20:
Please help quick!
I'm getting weird numbers and outputs when I run the code with the pulse sensor
0,600,0
583
0,604,576
0,604,574
0,604,569
0,604,562
0,604,556
0,604,547
0,604,547
0,604,543
4,587
0,604,586
0,604,584
0,604,583
0,604,576
0,604,574
0,604,569
0,604,562
0,604,556
0,604,547
0,604,547
0,604,543
0,600,0
0,600,434
0,600,418
0,600,424
0,600,444
0,600,459
0,600,474
0,600,485
0,600,495
0,600,502
0,600,505
0,600,504
0,600,507
0,600,510
0,600,513
0,600,510

quackamadoodleduck, 2017-02-20:
Please help, I can't figure out any way to solve this

joelSensor, 2017-02-22:
Did you make any modification to the Pulse Sensor code?

The Arduino sketch is defaulted to output for the Arduino Serial Plotter.
Take a look at the read-me guide for notes on that. Try running the Arduino Serial Plotter to see if that makes the numbers make sense

quackamadoodleduck, 2017-02-23:
thanks,
but is there any way I could somehow output to the serial monitor?
any help is welcome thank you so much

joelSensor, 2017-02-23:
I'm not sure what you mean by output to the serial monitor.
What do you want to output to the serial monitor?

quackamadoodleduck, 2017-02-24:
the BPM itself

joelSensor, 2017-02-24:
At the top of the main page on the sketch change the `outputType` variable to 
`outputType = PROCESSING_VISUALIZER`
Then In the main loop, comment out the line
`serialOutput()`
by putting `//` in front of it.
Then, in the `AllSerialHandling` tab, comment out the line
`sendDataToSerial('Q',IBI);`

quackamadoodleduck, 2017-02-26:
Thank you so much
:D

yury-g, 2026-08-16:
This repository has been retired as a read-only historical archive. Its code and discussion remain preserved, but fixes and support no longer happen here.

For current Arduino installation, examples, supported boards, and source, use [PulseSensorPlayground](https://github.com/WorldFamousElectronics/PulseSensorPlayground). If a problem is reproducible with the current Playground release, open a new issue there with the board and core versions, library version, wiring and supply voltage, raw waveform, and a minimal sketch.

Closing this historical issue as not planned in this retired repository.

closed 2026-08-16 by yury-g