WorldFamousElectronics/PulseSensor_Amped_Arduino #43: Can u please fix this problem in coding of pulse sensorr

sharansjce, 2017-03-01:
When i am running the code iam getting  no output in processing visualizer it showing a single wave continusely going but giving pulse 0 and bpm 0
And i tried serial plotter too by changing the output type in main but it showing the waves going on even i removed my finger
Plz  anyone help me..

sharansjce, 2017-03-03:
Thank you for the reply.I am using arduino uno.and the interrupt code is not working

sharansjce, 2017-03-03:
Thank you for the reply.I am using arduino uno.and the interrup

sharansjce, 2017-03-03:
[image]
I am getting output what i initialized in the code

sharansjce, 2017-03-03:
[image]
This is the output I am getting in the serial plotter continuously even without my finger in the sensor i guess codes are correct. Do i have to change anything in the IDE to enable interrupt programming. I checked the interrupt code too for the timer2. Plz help me

sharansjce, 2017-03-03:
I think i have a problem with interrupt code.
The codes inside the interrupt setup() is not getting compiled and running. But it showing no errors when compiling.

sharansjce, 2017-03-03:
> Please give me more information.
> What Arduino board are you using?
> Verify that you have uploaded the code to your board.
> Can you share a screen shot of what the Visualizer or Plotter look like?

Thank you for the prompt reply. I tried the code in this website. But not
working for me. Do I have to update  any thing in the arduino ide ?.i am
using arduino  uno.bellow I have attached the plotter visuals.It
continuously giving same without my fimgers in sensor

Thanks inadvance

joelSensor, 2017-03-03:
Hi,
I am pretty sure that the interrupt code is being compiled. If it was not, you would get an error.
Please do a couple of things:
First, verify that you are connecting the Pulse Sensor purple wire to the A0 pin on the arduino Uno.
Also, please send a picture of the Pulse Sensor when it is plugged in.

sharansjce, 2017-03-04:
Thank you for the reply
  Here is my attachment of sensor
Thank you so much for your great help.

sharansjce, 2017-03-04:
Thank you so much .... I got the output.
There was a problem with my sensor only. When i tried with another one i got the result .
Once again thank you for your great and timely help

closed 2017-03-04 by sharansjce

biomurph, 2017-03-04:
I never saw the picture that you send of the Pulse Sensor. 
You say that there was a problem with it? I would like to figure out what the problem was.

sharansjce, 2017-03-05:
Thank you so much. I connected wrongly and also there was a problem  with
my sensor. I found out what the problem was. I got the output.

biomurph, 2017-03-05:
Great!