WorldFamousElectronics/PulseSensor_Amped_Arduino #45: Inconsistent BPM readings

showmethemooney, 2017-03-02:
I've changed the code to output only the bpm with the directions that were posted on a prior post and the readings are very inconsistent. It's clear that the sensor has not been working properly because even with the unaltered code, the light I have hooked up blinks very randomly. Might I have a faulty sensor?
[image]

joelSensor, 2017-03-02:
Hi,

Can you share your code for verification?
Where are you placing the sensor?
It is a good idea to use the Arduino Serial Plotter to verify that you are getting a good signal. Please do that, and send along a screenshot.

showmethemooney, 2017-03-03:
/*  Pulse Sensor Amped 1.5    by Joel Murphy and Yury Gitman   http://www.pulsesensor.com

----------------------  Notes ----------------------  ----------------------
This code:
1) Blinks an LED to User's Live Heartbeat   PIN 13
2) Fades an LED to User's Live HeartBeat    PIN 5
3) Determines BPM
4) Prints All of the Above to Serial

Read Me:
https://github.com/WorldFamousElectronics/PulseSensor_Amped_Arduino/blob/master/README.md
 ----------------------       ----------------------  ----------------------
*/

#define PROCESSING_VISUALIZER 1
#define SERIAL_PLOTTER  2

//  Variables
int pulsePin = 0;                 // Pulse Sensor purple wire connected to analog pin 0
int blinkPin = 13;                // pin to blink led at each beat
int fadePin = 5;                  // pin to do fancy classy fading blink at each beat
int fadeRate = 0;                 // used to fade LED on with PWM on fadePin

// Volatile Variables, used in the interrupt service routine!
volatile int BPM;                   // int that holds raw Analog in 0. updated every 2mS
volatile int Signal;                // holds the incoming raw data
volatile int IBI = 600;             // int that holds the time interval between beats! Must be seeded!
volatile boolean Pulse = false;     // "True" when User's live heartbeat is detected. "False" when not a "live beat".
volatile boolean QS = false;        // becomes true when Arduoino finds a beat.

// SET THE SERIAL OUTPUT TYPE TO YOUR NEEDS
// PROCESSING_VISUALIZER works with Pulse Sensor Processing Visualizer
//      https://github.com/WorldFamousElectronics/PulseSensor_Amped_Processing_Visualizer
// SERIAL_PLOTTER outputs sensor data for viewing with the Arduino Serial Plotter
//      run the Serial Plotter at 115200 baud: Tools/Serial Plotter or Command+L
static int outputType = PROCESSING_VISUALIZER;

void setup(){
  pinMode(blinkPin,OUTPUT);         // pin that will blink to your heartbeat!
  pinMode(fadePin,OUTPUT);          // pin that will fade to your heartbeat!
  Serial.begin(115200);             // we agree to talk fast!
  interruptSetup();                 // sets up to read Pulse Sensor signal every 2mS
   // IF YOU ARE POWERING The Pulse Sensor AT VOLTAGE LESS THAN THE BOARD VOLTAGE,
   // UN-COMMENT THE NEXT LINE AND APPLY THAT VOLTAGE TO THE A-REF PIN
//   analogReference(EXTERNAL);
}

//  Where the Magic Happens
void loop(){

    //serialOutput() ;

  if (QS == true){     // A Heartbeat Was Found
                       // BPM and IBI have been Determined
                       // Quantified Self "QS" true when arduino finds a heartbeat
        fadeRate = 255;         // Makes the LED Fade Effect Happen
                                // Set 'fadeRate' Variable to 255 to fade LED with pulse
        serialOutputWhenBeatHappens();   // A Beat Happened, Output that to serial.
        QS = false;                      // reset the Quantified Self flag for next time
  }

  ledFadeToBeat();                      // Makes the LED Fade Effect Happen
  delay(20);                             //  take a break
}

void ledFadeToBeat(){
    fadeRate -= 15;                         //  set LED fade value
    fadeRate = constrain(fadeRate,0,255);   //  keep LED fade value from going into negative numbers!
    analogWrite(fadePin,fadeRate);          //  fade LED
  }

showmethemooney, 2017-03-03:
Thanks for the quick reply Joel. I've been using the sensor on my finger. I've also tried my ear lobe and get the same results. I have been using the serial plotter and it's consistent with the blinking of the led. Anytime the led blinks the serial plotter outputs the new bpm. As you can see from the screen shot I posted of the serial plotter, the heartbeat readings can be a second apart or several seconds apart. I haven't been able to get anything that could be a realistic heart rate.

joelSensor, 2017-03-03:
Hi,
I don't see a screenshot of the serial plotter. There is a shot of the serial monitor in your original post. 
First, change the outputType to SERIAL_PLOTTER, and un-comment the serialOutput(). Then program your board (what board are you using?) and then run the Arduino Serial Plotter
Tools > Serial Plotter
To see if you are getting a good heartbeat signal.

showmethemooney, 2017-03-03:
[image]

showmethemooney, 2017-03-03:
I'm using an arduino uno.

joelSensor, 2017-03-03:
Hi,

I'm sorry this is taking so long to hammer out.
Please send a picture of the actual Pulse Sensor when it is on.

showmethemooney, 2017-03-03:
[image]

showmethemooney, 2017-03-03:
No problem. Is that what you were asking to see?

joelSensor, 2017-03-03:
Ok, Thanks. I was concerned that it might be a clone, but it looks legit.

So when you put it on your finger, you don't see any signal like in the getting started guide on this repo? It may be due to poor circulation? Can you try on other parts of your body? If you hold against your lower lip, you should get a very good signal, for example. Also, please try with different kinds of pressure on the sensor. A light, but not too light, pressure is ideal.

showmethemooney, 2017-03-03:
You know, I think I've just been holding it too tight. I seem to be getting reasonable readings now. While I have you, I was wondering if you can hook up multiple pulse sensors and measure 3 heart rates simultaneously. I'm wanting to calculate the average of three peoples heart rates in real time. Is that possible?

joelSensor, 2017-03-04:
Hi,

So great to hear that you are getting good readings!
For multipule sensor inputs, please look here
https://github.com/WorldFamousElectronics/PulseSensorAmped_2_Sensors
It is possible to add more than two with this setup. Yury and I have done 3 at a time.

The tutorial is coming soon, but if you can read the code, it's pretty straightforward.

Cheers!

showmethemooney, 2017-03-06:
Thanks for your help. I had one more question. I'm trying to get data through the serial to max msp. would you happen to know why it's not sending data. I can get it to send data with Serial.println and Serial.write in other arduino codes.

biomurph, 2017-03-06:
That sounds like an issue with Max... I'm not a Max officianado, but you should be able to connect Max to your serial port.

biomurph, 2019-02-04:
@giotorno 
This code base has been superceeded by the PulseSensor Playground.
https://github.com/WorldFamousElectronics/PulseSensorPlayground
Please use it instead.

In order to help you with your problem, I need to know what Arduino board you are using.
Also, please send a picture of your Pulse Sensor to verify it is a legit part.

Eefried, 2019-04-20:
> 
> 
> Hi,
> 
> Can you share your code for verification?
> Where are you placing the sensor?
> It is a good idea to use the Arduino Serial Plotter to verify that you are getting a good signal. Please do that, and send along a screenshot.

What could be wrong if the signal is showing a voltage of constant 2.44 V having a steady finger contact?
Putting the finger on the LED rhymically produces a respective signal change.
many thanks

biomurph, 2019-04-20:
@Eefried 
This code base has been superceeded by the PulseSensor Playground.
https://github.com/WorldFamousElectronics/PulseSensorPlayground
Please use it instead.

In order to help you with your problem, I need you to please send a picture of your Pulse Sensor to verify it is a legit part.

Eefried, 2019-04-20:
Hello,
yes this might be a grey export
https://www.ebay.at/itm/183422797924?ul_ref=https%253A%252F%252Frover.ebay.com%252Frover%252F0%252F0%252F0%253Fmpre%253Dhttps%25253A%25252F%25252Fwww.ebay.at%25252Fulk%25252Fitm%25252F183422797924%2526rvr_id%253D0%2526rvr_ts%253D3bc5406116a0ac3cb886b50bfffc0a84&ul_noapp=true
is this illegal?
best
regards

https://github.com/WorldFamousElectronics/PulseSensorPlayground

AndrewTan90, 2019-11-15:
Hello, I am using arduino uno. I had face a problem that from the pulse sensor, it shows BPM reach to 200 + when not touching, and also another things is that the BPM reading reach 150+ when I place the fingertips at pulse sensor while talking.  Any help are appreciated.

biomurph, 2019-11-17:
@Eefried I never replied to you! Sorry!
Looking at the link you sent, the Pulse Sensor that you were using is indeed a knockoff. I can't troubleshoot hardware that I didn't make.

@AndrewTan90 The Pulse Sensor may be reading noise when not touching it. This could be your problem? Also please show a picture of your Pulse Sensor for verification.

AndrewTan90, 2019-11-17:
@biomurph Hi, thank for reply .
[image]
This is the image for the pulse sensor that I had used, i don't know whether it is a clone or not.
I bought it from: https://my.cytron.io/p-pulse-sensor
The below is the code:
`
/*  Pulse Sensor Amped 1.4    by Joel Murphy and Yury Gitman   http://www.pulsesensor.com

----------------------  Notes ----------------------  ---------------------- 
This code:
1) Blinks an LED to User's Live Heartbeat   PIN 13
2) Fades an LED to User's Live HeartBeat
3) Determines BPM
4) Prints All of the Above to Serial

Read Me:
https://github.com/WorldFamousElectronics/PulseSensor_Amped_Arduino/blob/master/README.md   
 ----------------------       ----------------------  ----------------------
*/

//  Variables
int pulsePin = 0;                 // Pulse Sensor purple wire connected to analog pin 0
int blinkPin = 13;                // pin to blink led at each beat
int fadePin = 5;                  // pin to do fancy classy fading blink at each beat
int fadeRate = 0;                 // used to fade LED on with PWM on fadePin

// Volatile Variables, used in the interrupt service routine!
volatile int BPM;                   // int that holds raw Analog in 0. updated every 2mS
volatile int Signal;                // holds the incoming raw data
volatile int IBI = 600;             // int that holds the time interval between beats! Must be seeded! 
volatile boolean Pulse = false;     // "True" when User's live heartbeat is detected. "False" when not a "live beat". 
volatile boolean QS = false;        // becomes true when Arduoino finds a beat.

// Regards Serial OutPut  -- Set This Up to your needs
//static boolean serialVisual = true;   // Set to 'false' by Default.  Re-set to 'true' to see Arduino Serial Monitor ASCII Visual Pulse 
const int OUTPUT_TYPE = PROCESSING_VISUALIZER;

void setup(){
  pinMode(blinkPin,OUTPUT);         // pin that will blink to your heartbeat!
  pinMode(fadePin,OUTPUT);          // pin that will fade to your heartbeat!
  Serial.begin(57600);             // we agree to talk fast!
  interruptSetup();                 // sets up to read Pulse Sensor signal every 2mS 
   // IF YOU ARE POWERING The Pulse Sensor AT VOLTAGE LESS THAN THE BOARD VOLTAGE, 
   // UN-COMMENT THE NEXT LINE AND APPLY THAT VOLTAGE TO THE A-REF PIN
//   analogReference(EXTERNAL);   
}

//  Where the Magic Happens
void loop(){
  
    serialOutput() ;       
    
  if (QS == true){     // A Heartbeat Was Found
                       // BPM and IBI have been Determined
                       // Quantified Self "QS" true when arduino finds a heartbeat
        fadeRate = 255;         // Makes the LED Fade Effect Happen
                                // Set 'fadeRate' Variable to 255 to fade LED with pulse
        serialOutputWhenBeatHappens();   // A Beat Happened, Output that to serial.     
        QS = false;                      // reset the Quantified Self flag for next time    
  }
     
  ledFadeToBeat();                      // Makes the LED Fade Effect Happen 
  delay(20);                             //  take a break
}

void ledFadeToBeat(){
    fadeRate -= 15;                         //  set LED fade value
    fadeRate = constrain(fadeRate,0,255);   //  keep LED fade value from going into negative numbers!
    analogWrite(fadePin,fadeRate);          //  fade LED`

yury-g, 2026-08-16:
This repository has been retired as a read-only historical archive. Its code and discussion remain preserved, but fixes and support no longer happen here.

For current Arduino installation, examples, supported boards, and source, use [PulseSensorPlayground](https://github.com/WorldFamousElectronics/PulseSensorPlayground). If a problem is reproducible with the current Playground release, open a new issue there with the board and core versions, library version, wiring and supply voltage, raw waveform, and a minimal sketch.

Closing this historical issue as not planned in this retired repository.

closed 2026-08-16 by yury-g