WorldFamousElectronics/PulseSensor_Amped_Arduino #46: Feedback from Bluetooth Modul on PulseSensor Arduino Mini Pro

MeisterPetz92, 2017-03-03:
Hi,
I use the Pulsesensor with Arduino Mini Pro 3.3V/8MHz in connection with a Bluetooth module 
HC-10. The Mini works great with the Pulsesensor in serial plotting and with the processing app. 
But when connect the HC-10 parallel to the Pulsesensor over the 3.3V Source of the Mini, clock of the Bluetooth modul brought feedback on the pulsesensor -> picture 
Cause of the clock i get only kryptical values :-(
By the way the same project works fine with Arduino UNO and the interrupt settings are fitted.
Someone have an idea why I have that feedback? 
Thanks a lot and sorry for my bad english.
[image]

joelSensor, 2017-03-03:
Woah! That is some serious noise right there.

It looks like what's happening is that the Bluetooth module you're using is causing spikes in the powersupply that are affecting the Pulse Sensor. You can sort of see the pulse waveform in there, but then there are spikes in the signal as well. I would guess that the Bluetooth module is drawing surges of current to do whatever it's doing. Try this for a possible solution:
Place a large capacitor on the power rails. Like a 100uF electrolytic or higher. Put the capacitor as close to the Bluetooth module as you can. Then maybe put another one close to the Pulse Sensor. You should see a reduction or elimination in the spikes. Try different values of capacitors.
If that doesn't work, you may need to give the Bluetooth module (or the Pulse Sensor) it's own power supply.

JashonRex, 2018-06-09:
> Woah! That is some serious noise right there.
> It looks like what's happening is that the Bluetooth module you're using is causing spikes in the powersupply that are affecting the Pulse Sensor. You [...]
Place a large capacitor on the power rails. Like a 100uF electrolytic or higher. Put the capacitor as close to the Bluetooth module as you can. Then maybe put another one close to the Pulse Sensor. You should see a reduction or elimination in the spikes. Try different values of capacitors.
If that doesn't work, you may need to give the Bluetooth module (or the Pulse Sensor) it's own power supply.
Hello joelSensor, I'm doing the same project as MeisterPetz92, use the same module and i do have the same problem. I tried to fix the problem by doing what you've said and it did work well. I know it's been the long time since the last updated of this topic but now i want to ask you something and hope you can help. Could you tell me the reason why we have to give Bluetooth module it's own power? Where's the noise come from? Does the reason have a name?
Hope you can see this!

biomurph, 2018-06-09:
The problem is with power line noise. When the BT module advertises, or whatever it does on the radio, there is a large current draw which likely causes the Pulse Sensor LED to dim slightly, which is why the signal dips or rises.

closed 2018-06-09 by biomurph