WorldFamousElectronics/PulseSensor_Amped_Arduino #47: 'TCCR2A' was not declared in this scope

Tondy, 2017-03-04:
When trying to compile the code for the first time. I am receiving 
the error below.

Interrupt.ino: In function 'void interruptSetup()':
Interrupt:17: error: 'TCCR2A' was not declared in this scope
Interrupt:18: error: 'TCCR2B' was not declared in this scope
Interrupt:19: error: 'OCR2A' was not declared in this scope
Interrupt:20: error: 'TIMSK2' was not declared in this scope
Interrupt.ino: At global scope:
Interrupt:27: error: expected constructor, destructor, or type conversion before '(' token
'TCCR2A' was not declared in this scope

biomurph, 2017-03-04:
What board are you trying to program?

Tondy, 2017-03-05:
Helllo 

Thank you for the reply. So I changed my board to the right one. It works .

deskool, 2017-03-17:
I'm getting the same error using the Lillypad Arduino USB - ATmega32U4 Board. Any tips on how to get around this?

Thanks,

Tondy, 2017-03-17:
Just make sure you have select the right board in arduino.

biomurph, 2017-03-19:
Please confirm the board you are using, and the code you are using.
That will help us solve your peroblem

SavayaShinkaruk, 2017-11-19:
Interrupt:27: error: expected constructor, destructor, or type conversion before '(' token
 ISR(TIMER2_COMPA_vect){                         // triggered when Timer2 counts to 124
     ^
/Users/savayashinkaruk/Desktop/PulseSensor_Amped_Arduino-master/PulseSensorAmped_Arduino_1.5.0/Interrupt.ino: In function 'void interruptSetup()':
Interrupt:17: error: 'TCCR2A' was not declared in this scope
   TCCR2A = 0x02;     // DISABLE PWM ON DIGITAL PINS 3 AND 11, AND GO INTO CTC MODE
   ^
Interrupt:18: error: 'TCCR2B' was not declared in this scope
   TCCR2B = 0x06;     // DON'T FORCE COMPARE, 256 PRESCALER
   ^
Interrupt:19: error: 'OCR2A' was not declared in this scope
   OCR2A = 0X7C;      // SET THE TOP OF THE COUNT TO 124 FOR 500Hz SAMPLE RATE
   ^
Interrupt:20: error: 'TIMSK2' was not declared in this scope
   TIMSK2 = 0x02;     // ENABLE INTERRUPT ON MATCH BETWEEN TIMER2 AND OCR2A
   ^
Interrupt:21: error: 'sei' was not declared in this scope
   sei();             // MAKE SURE GLOBAL INTERRUPTS ARE ENABLED
       ^
/Users/savayashinkaruk/Desktop/PulseSensor_Amped_Arduino-master/PulseSensorAmped_Arduino_1.5.0/Interrupt.ino: At global scope:
Interrupt:27: error: expected constructor, destructor, or type conversion before '(' token
 ISR(TIMER2_COMPA_vect){                         // triggered when Timer2 counts to 124
    ^
Interrupt:36: error: expected unqualified-id before 'if'
   if(Signal < thresh && N > (IBI/5)*3){       // avoid dichrotic noise by waiting 3/5 of last IBI
   ^
exit status 1
expected constructor, destructor, or type conversion before '(' token

I am getting all these errors in my code. I am using a Feather MO - would that have anything to do with it?

CunningLearner, 2018-04-15:
hey,I am using nodemcU and It doesn't compile.But is working fine with arduino uno and arduino nano.
If I can use nodemcU with pulse sensor?
#same_error

biomurph, 2018-04-16:
@CunningLearner This code base has been superseded by the PulseSensor Playground library.
https://github.com/WorldFamousElectronics/PulseSensorPlayground

We have not been able to use interrupts with the ESP8266. In this case, please work with the example sketch called BPM_Alternative. 

I have not used nodemcU. Please start with Arduino, and then move to the nodemcU. If you get it working, please let us know!

closed 2018-04-16 by biomurph

Abir-mishad, 2020-06-19:
Can I compile That code for AVR chip ( atmega8)?

biomurph, 2020-06-19:
We compile down to the ATmega85 chip.
This issue is closed. Please initiate a new issue and ask your question

tangtangasia, 2020-12-30:
hello .. I have an error problem like this:

esp32pulsesensorthingspeak:15:13: error: 'pulseSensor' was not declared in this scope
int myBPM = pulseSensor.getBeatsPerMinute();
^
esp32pulsesensorthingspeak:17:1: error: expected unqualified-id before 'if'
if (myBPM < 100 && myBPM > 50)
^
Multiple libraries were found for "WiFi.h"
Used: C:\Users\asus\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\WiFi
exit status 1
'pulseSensor' was not declared in this scope

**and this is the code that I made, I use esp32 microcontroller, and pulse sensor. Where did I go wrong? I am just a beginner, please help me study.**

#define USE_ARDUINO_INTERRUPTS false
#include <PulseSensorPlayground.h>
#include <WiFi.h>
#include <ThingSpeak.h>

const char ssid[] = "mydream";
const char password[]= "12122234";
WiFiClient client;

const long CHANNEL = 1273034;
const char *WRITE_API = "A3ZUCRHUSPNDSAR8W";

long prevMillisThingSpeak = 0;
int intervalThingSpeak = 20000;
int myBPM = pulseSensor.getBeatsPerMinute();

if (myBPM < 100 && myBPM > 50)
if (millis() - prevMillisThingSpeak > intervalThingSpeak){
ThingSpeak.setField(1,myBPM);

int x = ThingSpeak.writeFields(CHANNEL, WRITE_API);
if (x==200){
  Serial.println ("Channel update seccessful.")
}
else{
  Serial.println("Problem updating channel. HTTP error code "+String (x));
}
prevMillisThingSpeak = millis();
}
}

biomurph, 2020-12-30:
@tangtangasia 
This is a repost of an issue that you put on another repo.
You need to start with the beginning guide that we have on our website.
Your code is not functional, because you don't do a bunch of things, including using the setup() and loop() functions.
Start with the example sketches and go from there.
I'm closing this issue as it is not germain to the original post.
Please begin again, and before you do, start with the examples