WorldFamousElectronics/PulseSensor_Amped_Arduino #48: Waveform Clipping at 1000

Nikeshbajaj, 2017-03-06:
Hello Dear
   I was stuck at clipped waveform as shown in attached figure, 
[image]

Waveform is keep clipping at 1000. If I change power to 3.3v from 5v, waveform is not having clear QRS region

biomurph, 2017-03-06:
Hello Dear,

The signal that Pulse Sensor provides is an analog signal, and the A-D converter in Arduino is 10bit. So it will read a signal from 0 (0V) to 1023 (5V) only. The signal is clipping because it is very large. You may want to try a different part of your body? Or different pressure of the sensor on your skin?

Also bear in mind that the Pulse Sensor is a PPG (https://en.wikipedia.org/wiki/Photoplethysmogram), and NOT an ECG. You will not see the QRS waveform with the Pulse Sensor.

Nikeshbajaj, 2017-03-08:
Thank You, that explains a lot.

closed 2017-05-26 by yury-g