WorldFamousElectronics/PulseSensor_Amped_Arduino #51: Error compiling on RFduino RFD22102

amtsaasan, 2017-03-21:
Hello all, 

I'm very new to arduino work and am trying to setup the pulse sensor on an RFduino using the example code base without modification. I'm not finding obvious hints to interrupt timer code for this hardware;

16MHz ARM Cortex-M0,  	18mA, 4uA ULP

Guidance would be greatly appreciated!

Error
Arduino: 1.8.1 (Mac OS X), Board: "RFduino"

Compiling sketch...

Interrupt:29: error: expected constructor, destructor, or type conversion before '(' token
 ISR(TIMER2_COMPA_vect){                         // triggered when Timer2 counts to 124
     ^
/Users/sarahhuffman/Documents/Arduino/PulseSensor_Amped_Arduino-master/PulseSensorAmped_Arduino_1.5.0/Interrupt.ino: In function 'void interruptSetup()':
Interrupt:18: error: 'TCCR2A' was not declared in this scope
         TCCR2A = 0x02;     // DISABLE PWM ON DIGITAL PINS 3 AND 11, AND GO INTO CTC MODE
         ^
Interrupt:19: error: 'TCCR2B' was not declared in this scope
         TCCR2B = 0x06;     // DON'T FORCE COMPARE, 256 PRESCALER
         ^
Interrupt:20: error: 'OCR2A' was not declared in this scope
         OCR2A = 0X7C;      // SET THE TOP OF THE COUNT TO 124 FOR 500Hz SAMPLE RATE
         ^
Interrupt:21: error: 'TIMSK2' was not declared in this scope
         TIMSK2 = 0x02;     // ENABLE INTERRUPT ON MATCH BETWEEN TIMER2 AND OCR2A
         ^
Interrupt:22: error: 'sei' was not declared in this scope
         sei();             // MAKE SURE GLOBAL INTERRUPTS ARE ENABLED
             ^
/Users/me/Documents/Arduino/PulseSensor_Amped_Arduino-master/PulseSensorAmped_Arduino_1.5.0/Interrupt.ino: At global scope:
Interrupt:29: error: expected constructor, destructor, or type conversion before '(' token
 ISR(TIMER2_COMPA_vect){                         // triggered when Timer2 counts to 124
    ^
exit status 1
expected constructor, destructor, or type conversion before '(' token

joelSensor, 2017-03-22:
Please read the Interruot Notes tab in the Arduino sketch. At the end of that document, there is sample code for use with a 'software interrupt'. That will work with the RFduino for sure

amtsaasan, 2017-03-22:
Thank you ever so much... It's compiles and uploaded!

Some comments to maybe add to the instruction for a later version. For at least this hardware I also needed to comment out three lines on the interrupt tab in the newly renamed void getPulse() :
1) //  cli();                                      // disable interrupts while we do this
2) //        sei();                               // enable interrupts again
3) //  sei();                                   // enable interrupts when youre done!

Many many many thanks! - M

closed 2017-05-26 by yury-g