WorldFamousElectronics/PulseSensor_Amped_Arduino #52: BPM greater than 200 without touching the sensor

ankitmathur45, 2017-04-20:
Hi guys, 

I have purchased original pulse sensor from sparkfun and uploaded the code as it is on my Arduino Uno board. But the sensor gives me erroneous readings eg. BPM > 200 without even touching the sensor. On touching the reading is fine 50% of the time but erroneous values are frustrating. I have been pulling my hair for past week or so due to this issue. Any help will be really appreciated. Please respond ASAP.

yury-g, 2017-05-26:
It sounds like you need some basic debugging.  
Run this code and see how the Pulse Signal looks like in Arduino's built-in serial Monitor. 
[](https://github.com/WorldFamousElectronics/PulseSensorStarterProject
)

yury-g, 2026-08-16:
This repository has been retired as a read-only historical archive. Its code and discussion remain preserved, but fixes and support no longer happen here.

For current Arduino installation, examples, supported boards, and source, use [PulseSensorPlayground](https://github.com/WorldFamousElectronics/PulseSensorPlayground). If a problem is reproducible with the current Playground release, open a new issue there with the board and core versions, library version, wiring and supply voltage, raw waveform, and a minimal sketch.

Closing this historical issue as not planned in this retired repository.

closed 2026-08-16 by yury-g