WorldFamousElectronics/PulseSensor_Amped_Arduino #54: How can I make work via bluetooth?

frankthedead, 2017-05-01:
Without using the data cable.

biomurph, 2017-05-02:
frankthedead,
Your question is too vague to answer in any way. Please be more specific.

frankthedead, 2017-05-02:
How can I use the port /dev/tty.Bluetooth-Incoming-Port? For example, I want to monitor my pulse in my phone.

biomurph, 2017-05-02:
It's not as easy as it sounds. First you need a bluetooth radio module that the Pulse Sensor can plug into. Look at Sparkfun or Adafruit for one of those.
Then you need code on the computer to monitor the incoming bluetooth signal. This is not a question that we are prepared to answer. You are asking for extensive design services that we don't provide

yury-g, 2017-05-17:
The Arduino 101 makes this much easier then before.  We don't have an example project yet, but there are some great ones around the web.  See 👇
[https://create.arduino.cc/projecthub/masteruan/arduino-101-curie-ios-pulse-sensor-2441cd](https://create.arduino.cc/projecthub/masteruan/arduino-101-curie-ios-pulse-sensor-2441cd)

closed 2017-05-26 by yury-g