WorldFamousElectronics/PulseSensor_Amped_Arduino #55: Java stackoverflow errors on Ubuntu

sharmaavtansh, 2017-05-04:
Exception in thread "Thread-2" java.lang.StackOverflowError
	at java.util.regex.Pattern$CharProperty.match(Pattern.java:3775)
	at java.util.regex.Pattern$Curly.match0(Pattern.java:4250)
	at java.util.regex.Pattern$Curly.match(Pattern.java:4234)
	at java.util.regex.Pattern$Neg.match(Pattern.java:5097)
	at java.util.regex.Pattern$BmpCharProperty.match(Pattern.java:3798)
	at java.util.regex.Pattern$GroupHead.match(Pattern.java:4658)
	at java.util.regex.Pattern$Loop.match(Pattern.java:4785)
	at java.util.regex.Pattern$GroupTail.match(Pattern.java:4717)
	at java.util.regex.Pattern$Curly.match0(Pattern.java:4279)
	at java.util.regex.Pattern$Curly.match(Pattern.java:4234)
	at java.util.regex.Pattern$Neg.match(Pattern.java:5097)
	at java.util.regex.Pattern$BmpCharProperty.match(Pattern.java:3798

biomurph, 2017-05-04:
woah, I have no idea what that is.
Are you running our Processing Visualizer?

sharmaavtansh, 2017-05-04:
Hi,

No. I didn't ran your process visualizer on Ubuntu. The code for Arduino on
Ubuntu 16.04 Lts gives error. However on Windows it works fine.
One more thing can we get strap for finger online?

----
Regards,
Avtansh Sharma
-----

biomurph, 2017-05-04:
Hi,

We sell our kits with the finger strap included as part of the kit. If you did not get the kit parts with your Pulse Sensor, then you likely purchased a clone of our hardware. There are many copycats out there, but only one original. Please support the original makers!

sharmaavtansh, 2017-05-04:
Hi,

Thanks for info. Is there anyway I can verify the product? I will be
requesting this product in bulk orders after my product is completed with
other sensors. For the time being can you provide any source where I can
get a strap to complete my wearable device.

-----

Regards,
Avtansh Sharma

biomurph, 2017-05-05:
Hi,
You can verify the product by sending a picture of what you have. I can tell if it is ours or not  by looking. For bulk order request, please email us directly. You can find our email on our about page.
https://pulsesensor.com/pages/about-us
It would be best for you to find a local supplier for your strap needs.

sharmaavtansh, 2017-05-18:
> Hi,
>
> Thanks for help . Please check attached screenshots.
> I am also giving you the purchase link from where I purchased it.
> http://www.ebay.in/itm/181949135410?aff_source=Sok-Goog
>
>
> ------
> Regards,
> Avtansh Sharma

Hi,

Please find the attached images for your reference

[image: Inline image 1]

[image: Inline image 2]

------
Regards,
Avtansh Sharma

joelSensor, 2017-05-18:
No images came through

joelSensor, 2017-05-18:
The link you sent to the Ebay site does show that they use our assets and images.
They also link to all our content.
I can only know for sure if I can see a picture of the sensor that you have.
Please attach or link to an image

yury-g, 2026-08-16:
This repository has been retired as a read-only historical archive. Its code and discussion remain preserved, but fixes and support no longer happen here.

For current Arduino installation, examples, supported boards, and source, use [PulseSensorPlayground](https://github.com/WorldFamousElectronics/PulseSensorPlayground). If a problem is reproducible with the current Playground release, open a new issue there with the board and core versions, library version, wiring and supply voltage, raw waveform, and a minimal sketch.

Closing this historical issue as not planned in this retired repository.

closed 2026-08-16 by yury-g