WorldFamousElectronics/PulseSensor_Amped_Arduino #56: sensor not working properly

dheimaz, 2017-07-23:
Guys, i got the problem about this project. My sensor not working properly. raw data doesn't same with Startedproject, i was change my finger position, but the graph still not drawn like the project.
[image]

I use arduino nano Clone with CH340 driver

biomurph, 2017-07-25:
It doesn't look like you are having an issue... That looks like it is OK signal from the PPG.
Can you verify that you are using an Original Pulse Sensor and not a clone? Send us a pic of your Pulse Sensor if you can't tell, and also tell us where you bought it.

dheimaz, 2017-07-26:
@biomurph i think my sensor is clone, because i bought it from online store on my country not from official store of pulse sensor, but i't looks like same with the original. 

if the sensor OK why my project read my BPM more than 150++, i allready double check to my finger and my friend finger, but the result is same.

biomurph, 2017-07-26:
You may indeed have an Original Pulse Sensor. We distribute to Spark Fun, who then ships 'kits' of hardware to stores all over the world. Please send a photo of the one you have.
I think I see the problem!
Looking at your screenshot, you are targeting the Arduino Nano.
Please look into the 'Timer Interrupt Notes' tab for info on how to adjust the interrupt settings.

You can also use the beta version of our Arduino Library, found here
https://github.com/WorldFamousElectronics/PulseSensorPlayground
Follow the ReadMe to get set up with that.

yury-g, 2026-08-16:
This repository has been retired as a read-only historical archive. Its code and discussion remain preserved, but fixes and support no longer happen here.

For current Arduino installation, examples, supported boards, and source, use [PulseSensorPlayground](https://github.com/WorldFamousElectronics/PulseSensorPlayground). If a problem is reproducible with the current Playground release, open a new issue there with the board and core versions, library version, wiring and supply voltage, raw waveform, and a minimal sketch.

Closing this historical issue as not planned in this retired repository.

closed 2026-08-16 by yury-g