WorldFamousElectronics/PulseSensor_Amped_Arduino #57: Running this code with AD8232

beingmachine, 2017-08-20:
Hello,
I have [https://www.sparkfun.com/products/12650](url) AD8232 board, from which i can use to plot heart rate PQRST data, however they don't provide code for calculating BPM from the peak like the pulse sensor code do.
I want to use Pulse sensor given code to calculate BPM from AD8232, I am not much into Arduino, and need help in that.

what metrics do i have to change in the code to work with the Ad8232.

I am  getting this output at my Serial Plotter in Arduino..

[image]

Please help
thank you..

biomurph, 2017-08-21:
The Pulse Sensor code relies on knowing what the sample rate is, and assumes that it is 500Hz (sample taken every 2mS). It looks like the signal you are getting idles around V/2, or 512. You can simply look for the R wave going up by setting a threshold > 520 and then time between R waves to then derive BPM. The threshold level will likely need to be adjusted.

beingmachine, 2017-08-22:
I increased the value of Threshold to 550 and i am still unbale to get the BPM out properly, Here's what i am see i Seral Plotter
[image]
BPM goes upto 150, when i am idle - It should be around 70-80 at this stage..

Can you tell what values and where should be change in the code.?

Thank you

beingmachine, 2017-08-22:
Hey, I have just used this code `/*  PulseSensor™ Starter Project   http://www.pulsesensor.com
 *   
This an Arduino project. It's Best Way to Get Started with your PulseSensor™ & Arduino. 
-------------------------------------------------------------
1) This shows a live human Heartbeat Pulse. 
2) Live visualization in Arduino's Cool "Serial Plotter".
3) Blink an LED on each Heartbeat.
4) This is the direct Pulse Sensor's Signal.  
5) A great first-step in troubleshooting your circuit and connections. 
6) "Human-readable" code that is newbie friendly." 
*/

//  Variables
int PulseSensorPurplePin = 0;        // Pulse Sensor PURPLE WIRE connected to ANALOG PIN 0
int LED13 = 13;   //  The on-board Arduion LED

int Signal;                // holds the incoming raw data. Signal value can range from 0-1024
int Threshold = 550;            // Determine which Signal to "count as a beat", and which to ingore. 

// The SetUp Function:
void setup() {
  pinMode(LED13,OUTPUT);         // pin that will blink to your heartbeat!
   Serial.begin(9600);         // Set's up Serial Communication at certain speed. 
   
}

// The Main Loop Function
void loop() {

  Signal = analogRead(PulseSensorPurplePin);  // Read the PulseSensor's value. 
                                              // Assign this value to the "Signal" variable.

   Serial.println(Signal);                    // Send the Signal value to Serial Plotter.

   
   if(Signal > Threshold){                          // If the signal is above "550", then "turn-on" Arduino's on-Board LED.  
     digitalWrite(LED13,HIGH);          
   } else {
     digitalWrite(LED13,LOW);                //  Else, the sigal must be below "550", so "turn-off" this LED.
   }

delay(10);
   
   
}
`

^from PulseSensorStarterProject folder.
and it works fine and LED on pin 13 blinks according to heart beat

beingmachine, 2017-08-22:
How can i get BPM from this code.? that too working on Timer2

beingmachine, 2017-08-22:
There something wrong with IBI values.?
[image]

biomurph, 2017-08-22:
What Arduino hardware are you using?
What code of ours are you using?

Our device is a PPG, and you are using an ECG. The signals are quite different.

beingmachine, 2017-08-22:
I am using Atmega2560V at 8Mhz running on 3.3V.
I am using [https://github.com/WorldFamousElectronics/PulseSensor_Amped_Arduino](url) this code

can this code be edited to run with ECG.?

biomurph, 2017-08-22:
It may be able to be edited, but that's not something that we can do for you. Take a look at the last example on this page
https://github.com/WorldFamousElectronics/PulseSensor_Amped_Arduino/blob/master/PulseSensorAmped_Arduino_1.5.0/Timer_Interrupt_Notes.ino
it explains how to use a software timer to get more accurate sample rate for determining BPM.

dev-bhargav, 2018-03-27:
Hey @beingmachine 
Which Arduino code did you ran to get this clear ECG waveform.
Please also share your circuit diagram or image of hardware.
[image]

biomurph, 2018-03-27:
@dev-bhargav 
This code repo has been superseded by the Pulse Sensor Playground Library
https://github.com/WorldFamousElectronics/PulseSensorPlayground
Also, the Pulse Sensor is not an ECG. It is a PPG. If you are using an ECG, then you need to go to the forum that supports your hardware.

closed 2018-03-27 by biomurph

yoginugroho, 2018-08-07:
Hello guys, you cannot just set the treshold to get the peak, for the best BPM detection, the signal must be preprocessing by various filter, you can read about it on pan tompkin algoritm in IEEE

biomurph, 2018-08-07:
This issue is closed and this repo is superseded by the Pulse Sensor Playground library
https://github.com/WorldFamousElectronics/PulseSensorPlayground
Please make your comments there.