WorldFamousElectronics/PulseSensor_Amped_Arduino #58: Pulse sensor not working correctly with arduini Pro mini

shadowwalkersteam, 2017-08-23:
Pulse sensor is not working correctly with arduino Pro Mini 5v version. It is showing values even when there is no finger touched. LED13 blinks or sometimes Turn on and remains constant when there is no finger attached to pulse sensor. Please help me how i can configure it according to Pro Mini 5v version? What changes should i do in timer interrupt?

beingmachine, 2017-08-23:
> It is showing values even when there is no finger touched

Tried removing pulse sensor at this time.? if the LED still blinks then it might be problem in the Arduino..

why don't you try with different Arduino's like UNO and Mega.? Moreover i would suggest you to use isolated power source OR battery power source - That means connect your Arduino to battery powered computer like laptop, and have a test..

shadowwalkersteam, 2017-08-23:
I'm connecting it with laptop. Even my device is using battery so i also checked it by removing USB and run it on DC battery supply

shadowwalkersteam, 2017-08-23:
No the LED didn't blink when i removed the pulse sensor..

shadowwalkersteam, 2017-08-23:
I'm making a wearable pulse monitor that's why i can't use UNO i have limited and that is the reason i'm using Pro mini

shadowwalkersteam, 2017-08-23:
Should i edit the timer interrupt or should i switch to timer2? As the the pulse sensor code is compatible with arduino UNO by default

biomurph, 2017-08-23:
The default code should work with the ProMini 5V. It's the same processor as the UNO.
If you're seeing blinks when there is no finger on the pulse sensor, then that is just noise.
Try a couple of things.
In the Interrupt.ino file, change the variable called 'thresh' to a higher number on line 9 and line 95.

Also, you can try our new PulseSensor library 
https://github.com/WorldFamousElectronics/PulseSensorPlayground
and see if that gives you better results

shadowwalkersteam, 2017-08-23:
Ok i'll try it.. 1 more thing as there is nothing connected on pin 5 of arduino pro mini. What will be the effect of fade?? As arduino pro mini has LED on pin 13 and pin 5 is free nothing is connected there

biomurph, 2017-08-23:
If nothing is connected there then you will not see anything happen. pin 13 LED should still blink with your heartbeat

arghyakusumdas7, 2017-09-09:
I am also having some issues with my Arduino Pro Mini. I am not getting any output on the serial monitor.

The starter-project worked for me. But this PulseSensor_Amped_Arduino is not working on Arduino Pro mini. However, the compilation and upload was successful.

yury-g, 2026-08-16:
This repository has been retired as a read-only historical archive. Its code and discussion remain preserved, but fixes and support no longer happen here.

For current Arduino installation, examples, supported boards, and source, use [PulseSensorPlayground](https://github.com/WorldFamousElectronics/PulseSensorPlayground). If a problem is reproducible with the current Playground release, open a new issue there with the board and core versions, library version, wiring and supply voltage, raw waveform, and a minimal sketch.

Closing this historical issue as not planned in this retired repository.

closed 2026-08-16 by yury-g