WorldFamousElectronics/PulseSensor_Amped_Arduino #59: Calculate the average of pulse

shadowwalkersteam, 2017-08-31:
How i can calculate the average of pulse sensor? I have tried this code but it didn't worked and even the values didn't showed up in serial monitor and in processing. 
`  
 for(int i=0; i<64; i++)
  { // Average over 16 measurements
     Signal += analogRead(pulsePin);
  }
  Signal= Signal/64;
`

shadowwalkersteam, 2017-08-31:
I did this in interrupt.ino where it is reading from A0

joelSensor, 2017-08-31:
Interesting.
Well, you are taking 64 readings of the ADC. It takes Arduino (ATmega) about 100uS to do an Analog read. Your loop takes ~6.4mS to complete. There's your problem right there.
You need to reduce the number of samples you are averaging. Our Arduino Code interrupts to sample every 2mS. You need to limit the amount of samples you take to >20

shadowwalkersteam, 2017-08-31:
Ok i'll try that... I'm using arduino pro mini and i'm using timer1 as it mentioned that is used 9 and 11 pin and my 9th and 11th pin are not free will it cause a problem or should i use timer2 for pro mini 5v version

shadowwalkersteam, 2017-08-31:
which is more stable? and what is the threshold value so that i could minimize the noise

biomurph, 2017-08-31:
The only thing that the Timer selection does to those pins is not allow you to use PWM.
Timer 2 breaks PWM on pins 3 and 11
Timer 1 breaks PWM on pins 9 and 10
You can use those pins for other digital things.

As for the threshold, you will have to play around with the default value of the thresh varialbe in the interrupts

shadowwalkersteam, 2017-09-01:
Okay.. please suggest me which timer is should i use for pro mini 5v 16mhz ?

biomurph, 2017-09-01:
You could use either one.
Depends on if you want to use PWM on any of the pins. Simple.

yury-g, 2026-08-16:
This repository has been retired as a read-only historical archive. Its code and discussion remain preserved, but fixes and support no longer happen here.

For current Arduino installation, examples, supported boards, and source, use [PulseSensorPlayground](https://github.com/WorldFamousElectronics/PulseSensorPlayground). If a problem is reproducible with the current Playground release, open a new issue there with the board and core versions, library version, wiring and supply voltage, raw waveform, and a minimal sketch.

Closing this historical issue as not planned in this retired repository.

closed 2026-08-16 by yury-g