WorldFamousElectronics/PulseSensor_Amped_Arduino #60: Pulse sensor showing BPM 150+

shadowwalkersteam, 2017-09-01:
Pulse signal is smooth and looks good but the BPM is giving 150+ sometimes it also show values when there is no finger.. My wiring is correct and i'm supplying 5v DC to arduino. I'm using Arduino pro mini 5v

biomurph, 2017-09-01:
If you see values when there is no finger, then that is noise.
please share a link to your code for review. It may be something simple that is causing the large BPM numbers

shadowwalkersteam, 2017-09-01:
`//  Variables
int pulsePin = A0;                 // Pulse Sensor purple wire connected to analog pin 0
//int blinkPin = 13;                // pin to blink led at each beat

// Volatile Variables, used in the interrupt service routine!
volatile int BPM;                   // int that holds raw Analog in 0. updated every 2mS
volatile int Signal;                // holds the incoming raw data
volatile int IBI = 600;             // int that holds the time interval between beats! Must be seeded!
volatile boolean Pulse = false;     // "True" when User's live heartbeat is detected. "False" when not a "live beat".
volatile boolean QS = false;        // becomes true when Arduoino finds a beat.

void setup(){
//  pinMode(blinkPin,OUTPUT);         // pin that will blink to your heartbeat!
  Serial.begin(115200);             // we agree to talk fast!
  interruptSetup();                 // sets up to read Pulse Sensor signal every 2mS
}

void loop(){
  Serial.print("Your Beat   ");
  Serial.println(BPM);
  delay(1000);
  
}`

shadowwalkersteam, 2017-09-01:
This is the code for Interrupt.ino
`volatile int rate[10];                    // array to hold last ten IBI values
volatile unsigned long sampleCounter = 0;          // used to determine pulse timing
volatile unsigned long lastBeatTime = 0;           // used to find IBI
volatile int P =512;                      // used to find peak in pulse wave, seeded
volatile int T = 512;                     // used to find trough in pulse wave, seeded
volatile int thresh = 512;                // used to find instant moment of heart beat, seeded
volatile int amp = 0;                   // used to hold amplitude of pulse waveform, seeded
volatile boolean firstBeat = true;        // used to seed rate array so we startup with reasonable BPM
volatile boolean secondBeat = false;      // used to seed rate array so we startup with reasonable BPM

void interruptSetup(){  // CHECK OUT THE Timer_Interrupt_Notes TAB FOR MORE ON INTERRUPTS 
  // Initializes Timer2 to throw an interrupt every 2mS.
  TCCR2A = 0x02;     // DISABLE PWM ON DIGITAL PINS 3 AND 11, AND GO INTO CTC MODE
  TCCR2B = 0x06;     // DON'T FORCE COMPARE, 256 PRESCALER
  OCR2A = 0X7C;      // SET THE TOP OF THE COUNT TO 124 FOR 500Hz SAMPLE RATE
  TIMSK2 = 0x02;     // ENABLE INTERRUPT ON MATCH BETWEEN TIMER2 AND OCR2A
  sei();             // MAKE SURE GLOBAL INTERRUPTS ARE ENABLED
}

// THIS IS THE TIMER 2 INTERRUPT SERVICE ROUTINE.
// Timer 2 makes sure that we take a reading every 2 miliseconds
ISR(TIMER2_COMPA_vect){                         // triggered when Timer2 counts to 124
  cli();                                      // disable interrupts while we do this
  Signal = analogRead(pulsePin);              // read the Pulse Sensor
  sampleCounter += 2;                         // keep track of the time in mS with this variable
  int N = sampleCounter - lastBeatTime;       // monitor the time since the last beat to avoid noise

    //  find the peak and trough of the pulse wave
  if(Signal < thresh && N > (IBI/5)*3){       // avoid dichrotic noise by waiting 3/5 of last IBI
    if (Signal < T){                        // T is the trough
      T = Signal;                         // keep track of lowest point in pulse wave
    }
  }

  if(Signal > thresh && Signal > P){          // thresh condition helps avoid noise
    P = Signal;                             // P is the peak
  }                                        // keep track of highest point in pulse wave

  //  NOW IT'S TIME TO LOOK FOR THE HEART BEAT
  // signal surges up in value every time there is a pulse
  if (N > 250){                                   // avoid high frequency noise
    if ( (Signal > thresh) && (Pulse == false) && (N > (IBI/5)*3) ){
      Pulse = true;                               // set the Pulse flag when we think there is a pulse
//      digitalWrite(blinkPin,HIGH);                // turn on pin 13 LED
      IBI = sampleCounter - lastBeatTime;         // measure time between beats in mS
      lastBeatTime = sampleCounter;               // keep track of time for next pulse

      if(secondBeat){                        // if this is the second beat, if secondBeat == TRUE
        secondBeat = false;                  // clear secondBeat flag
        for(int i=0; i<=9; i++){             // seed the running total to get a realisitic BPM at startup
          rate[i] = IBI;
        }
      }

      if(firstBeat){                         // if it's the first time we found a beat, if firstBeat == TRUE
        firstBeat = false;                   // clear firstBeat flag
        secondBeat = true;                   // set the second beat flag
        sei();                               // enable interrupts again
        return;                              // IBI value is unreliable so discard it
      }

      // keep a running total of the last 10 IBI values
      word runningTotal = 0;                  // clear the runningTotal variable

      for(int i=0; i<=8; i++){                // shift data in the rate array
        rate[i] = rate[i+1];                  // and drop the oldest IBI value
        runningTotal += rate[i];              // add up the 9 oldest IBI values
      }

      rate[9] = IBI;                          // add the latest IBI to the rate array
      runningTotal += rate[9];                // add the latest IBI to runningTotal
      runningTotal /= 10;                     // average the last 10 IBI values
      BPM = 60000/runningTotal;               // how many beats can fit into a minute? that's BPM!
      QS = true;                              // set Quantified Self flag
      // QS FLAG IS NOT CLEARED INSIDE THIS ISR
    }
  }

  if (Signal < thresh && Pulse == true){   // when the values are going down, the beat is over
//    digitalWrite(blinkPin,LOW);            // turn off pin 13 LED
    Pulse = false;                         // reset the Pulse flag so we can do it again
    amp = P - T;                           // get amplitude of the pulse wave
    thresh = amp/2 + T;                    // set thresh at 50% of the amplitude
    P = thresh;                            // reset these for next time
    T = thresh;
  }

  if (N > 2500){                           // if 2.5 seconds go by without a beat
    thresh = 512;                          // set thresh default
    P = 512;                               // set P default
    T = 512;                               // set T default
    lastBeatTime = sampleCounter;          // bring the lastBeatTime up to date
    firstBeat = true;                      // set these to avoid noise
    secondBeat = false;                    // when we get the heartbeat back
  }

  sei();                                   // enable interrupts when youre done!
}// end isr`

shadowwalkersteam, 2017-09-01:
When i place finger it becomes stable and shows correct value for sometime but then after few moments it gives HIGH BPM again (120+ or 150+ too)

biomurph, 2017-09-01:
In the loop, you have a delay of 100mS. Take that out.
Do your serial write on the condition that QS is true.

The variable QS is set to true in the interrupt when there is a beat.
You have to reset it in the loop. Try this

if (QS == true){     // A Heartbeat Was Found
                       // BPM and IBI have been Determined
                       // Quantified Self "QS" true when arduino finds a heartbeat
        Serial.print("Your Beat ");
        Serial.println(BPM);
        QS = false;                      // reset the Quantified Self flag for next time
  }

And take out the delay.

shadowwalkersteam, 2017-09-05:
Is this sensor is compatible with wrist? I want to use pulse sensor on my wrist

biomurph, 2017-09-05:
For that you will have to experiment. We know people have read pulse from the wrist, however you will have to experiment with it yourself.

ZaroonRashid, 2018-01-20:
Hi shadowwalkersteam and biomurph ... i have same problem .. and i apply instructions which biomurph told ... but the problem is same that after sometimes it is giving values above 135 ... what should i do ?

biomurph, 2018-01-22:
@ZaroonRashid I need more information to help you out. Do you have a link to your code?
What Arduino board are you using?
Please share a picture of the Pulse Sensor you are using. There are a number of clones out there.

ZaroonRashid, 2018-01-22:
[image]
[image]

ZaroonRashid, 2018-01-22:
Sir ye Pulse sensor he ... 
And I am using this code for interrupt Class

volatile int rate[10];                    // array to hold last ten IBI values
volatile unsigned long sampleCounter = 0;          // used to determine pulse timing
volatile unsigned long lastBeatTime = 0;           // used to find IBI
volatile int P =512;                      // used to find peak in pulse wave, seeded
volatile int T = 512;                     // used to find trough in pulse wave, seeded
volatile int thresh = 525;                // used to find instant moment of heart beat, seeded
volatile int amp = 100;                   // used to hold amplitude of pulse waveform, seeded
volatile boolean firstBeat = true;        // used to seed rate array so we startup with reasonable BPM
volatile boolean secondBeat = false;      // used to seed rate array so we startup with reasonable BPM

void interruptSetup(){     
  // Initializes Timer2 to throw an interrupt every 2mS.
  TCCR2A = 0x02;     // DISABLE PWM ON DIGITAL PINS 3 AND 11, AND GO INTO CTC MODE
  TCCR2B = 0x06;     // DON'T FORCE COMPARE, 256 PRESCALER 
  OCR2A = 0X7C;      // SET THE TOP OF THE COUNT TO 124 FOR 500Hz SAMPLE RATE
  TIMSK2 = 0x02;     // ENABLE INTERRUPT ON MATCH BETWEEN TIMER2 AND OCR2A
  sei();             // MAKE SURE GLOBAL INTERRUPTS ARE ENABLED      
} 

// THIS IS THE TIMER 2 INTERRUPT SERVICE ROUTINE. 
// Timer 2 makes sure that we take a reading every 2 miliseconds
ISR(TIMER2_COMPA_vect){                         // triggered when Timer2 counts to 124
  cli();                                      // disable interrupts while we do this
  Signal = analogRead(pulsePin);              // read the Pulse Sensor 
  sampleCounter += 2;                         // keep track of the time in mS with this variable
  int N = sampleCounter - lastBeatTime;       // monitor the time since the last beat to avoid noise

    //  find the peak and trough of the pulse wave
  if(Signal < thresh && N > (IBI/5)*3){       // avoid dichrotic noise by waiting 3/5 of last IBI
    if (Signal < T){                        // T is the trough
      T = Signal;                         // keep track of lowest point in pulse wave 
    }
  }

  if(Signal > thresh && Signal > P){          // thresh condition helps avoid noise
    P = Signal;                             // P is the peak
  }                                        // keep track of highest point in pulse wave

  //  NOW IT'S TIME TO LOOK FOR THE HEART BEAT
  // signal surges up in value every time there is a pulse
  if (N > 250){                                   // avoid high frequency noise
    if ( (Signal > thresh) && (Pulse == false) && (N > (IBI/5)*3) ){        
      Pulse = true;                               // set the Pulse flag when we think there is a pulse
//      digitalWrite(blinkPin,HIGH);                // turn on pin 13 LED
      IBI = sampleCounter - lastBeatTime;         // measure time between beats in mS
      lastBeatTime = sampleCounter;               // keep track of time for next pulse

      if(secondBeat){                        // if this is the second beat, if secondBeat == TRUE
        secondBeat = false;                  // clear secondBeat flag
        for(int i=0; i<=9; i++){             // seed the running total to get a realisitic BPM at startup
          rate[i] = IBI;                      
        }
      }

      if(firstBeat){                         // if it's the first time we found a beat, if firstBeat == TRUE
        firstBeat = false;                   // clear firstBeat flag
        secondBeat = true;                   // set the second beat flag
        sei();                               // enable interrupts again
        return;                              // IBI value is unreliable so discard it
      }   

      // keep a running total of the last 10 IBI values
      word runningTotal = 0;                  // clear the runningTotal variable    

      for(int i=0; i<=8; i++){                // shift data in the rate array
        rate[i] = rate[i+1];                  // and drop the oldest IBI value 
        runningTotal += rate[i];              // add up the 9 oldest IBI values
[... 201 lines cut ...]
                                // Set 'fadeRate' Variable to 255 to fade LED with pulse
        serialOutputWhenBeatHappens();   // A Beat Happened, Output that to serial.     
        QS = false;                      // reset the Quantified Self flag for next time    
       } 
      else { 

      digitalWrite(blinkPin,LOW);            // There is not beat, turn off pin 13 LED
      }
     
   ledFadeToBeat();                      // Makes the LED Fade Effect Happen 
  if (Genotronex.available()){
BluetoothData=Genotronex.read();
   if(BluetoothData=='1'){   // if number 1 pressed ....
   digitalWrite(ledpin,1);
   Genotronex.println("LED  On D12 ON ! ");
   }
  if (BluetoothData=='0'){// if number 0 pressed ....
  digitalWrite(ledpin,0);
   Genotronex.println("LED  On D12 Off ! ");
  }
}
  delay(50);                             //  take a break
}

//=================================================

biomurph, 2018-01-22:
The device you show is not a legit Pulse Sensor.
I have no way of troubleshooting your problem because I cannot know what hardware that the ripoff company used.

Please support the original makers and purchase genuine Pulse Sensor.

ZaroonRashid, 2018-01-22:
well ... it seems all clear ... i have no genuine sensor ... Thanks

Dushyant1996, 2019-01-11:
@biomurph
I'm using the same sensor, and have a similar problem. Can you tell me where can i get the original sensor for the project?

biomurph, 2019-01-11:
You can get the Original and Best Pulse Sensor at
[Pulse Sensor](https://pulsesensor.com)
[Adafruit](https://www.adafruit.com/product/1093)
[SparkFun](https://www.sparkfun.com/products/11574)
[Amazon](https://www.amazon.com/PulseSensor-com-Original-Pulse-Sensor-project/dp/B01CPP4QM0/ref=sr_1_3?ie=UTF8&qid=1547220066&sr=8-3&keywords=pulse+sensor)

rgranasinghe, 2019-02-03:
Arduino Pulse Sensor Same Problem
i download the pulse sensor playground library and use the example code  "Getting_BPM_to_Monitor". without any changes i try to run it and when it running at some point it give heart beta value and increase it up to 200 BPM and sometimes even without place it on a finger it give the output. need advice to recover this problem.  the code i use is:

/*  Getting_BPM_to_Monitor prints the BPM to the Serial Monitor, using the least lines of code and PulseSensor Library.
 *  Tutorial Webpage: https://pulsesensor.com/pages/getting-advanced
 *
--------Use This Sketch To------------------------------------------
1) Displays user's live and changing BPM, Beats Per Minute, in Arduino's native Serial Monitor.
2) Print: "♥  A HeartBeat Happened !" when a beat is detected, live.
2) Learn about using a PulseSensor Library "Object".
4) Blinks LED on PIN 13 with user's Heartbeat.
--------------------------------------------------------------------*/

#define USE_ARDUINO_INTERRUPTS true    // Set-up low-level interrupts for most acurate BPM math.
#include <PulseSensorPlayground.h>     // Includes the PulseSensorPlayground Library.   

//  Variables
const int PulseWire = 0;       // PulseSensor PURPLE WIRE connected to ANALOG PIN 0
const int LED13 = 13;          // The on-board Arduino LED, close to PIN 13.
int Threshold = 550;           // Determine which Signal to "count as a beat" and which to ignore.
                               // Use the "Gettting Started Project" to fine-tune Threshold Value beyond default setting.
                               // Otherwise leave the default "550" value. 
                               
                               
PulseSensorPlayground pulseSensor;  // Creates an instance of the PulseSensorPlayground object called "pulseSensor"

void setup() {   

  Serial.begin(9600);          // For Serial Monitor

  // Configure the PulseSensor object, by assigning our variables to it. 
  pulseSensor.analogInput(PulseWire);   
  pulseSensor.blinkOnPulse(LED13);       //auto-magically blink Arduino's LED with heartbeat.
  pulseSensor.setThreshold(Threshold);   

  // Double-check the "pulseSensor" object was created and "began" seeing a signal. 
   if (pulseSensor.begin()) {
    Serial.println("We created a pulseSensor Object !");  //This prints one time at Arduino power-up,  or on Arduino reset.  
  }
}

void loop() {

 int myBPM = pulseSensor.getBeatsPerMinute();  // Calls function on our pulseSensor object that returns BPM as an "int".
                                               // "myBPM" hold this BPM value now. 

if (pulseSensor.sawStartOfBeat()) {            // Constantly test to see if "a beat happened". 
 Serial.println("♥  A HeartBeat Happened ! "); // If test is "true", print a message "a heartbeat happened".
 Serial.print("BPM: ");                        // Print phrase "BPM: " 
 Serial.println(myBPM);                        // Print the value inside of myBPM. 
} 

  
  delay(20);                    // considered best practice in a simple sketch.

}

biomurph, 2019-02-04:
@rgranasinghe 
This code base is superceeded by the PulseSensor Playground.
You are posting this issue in the wrong place.
https://github.com/WorldFamousElectronics/PulseSensorPlayground

I need you to tell me what hardware you are using and also send me a picture of your Pulse Sensor so that I can verify it is a legit part.

suchethanreddy, 2019-03-14:
@biomurph 
Can u please help me  on this!

suchethanreddy, 2019-03-14:
@biomurph
I have recently brought my pulse sensor
To get started with my project can you 
please send me the total progamme of the 
Pulse sensor!!

biomurph, 2019-03-15:
@suchethanreddy 
This code base is superceeded by the PulseSensor Playground.
You are posting this issue in the wrong place.
https://github.com/WorldFamousElectronics/PulseSensorPlayground

I need you to tell me what hardware you are using and also send me a picture of your Pulse Sensor so that I can verify it is a legit part.

FerrariF70, 2019-05-16:
Hi I have same problem but my problem show is low pulse 30, 42 and if I dont put finger on sensor, sensor read pulse. I use original pulsesensora and use Flora [https://www.adafruit.com/product/659](url). 
`#include <Adafruit_CircuitPlayground.h>
volatile int rate[10];                    // array to hold last ten IBI values
volatile unsigned long sampleCounter = 0;          // used to determine pulse timing
volatile unsigned long lastBeatTime = 0;           // used to find IBI
volatile int P = 512;                     // used to find peak in pulse wave, seeded
volatile int T = 512;                     // used to find trough in pulse wave, seeded
volatile int thresh = 525;                // used to find instant moment of heart beat, seeded
volatile int amp = 100;                   // used to hold amplitude of pulse waveform, seeded
volatile boolean firstBeat = true;        // used to seed rate array so we startup with reasonable BPM
volatile boolean secondBeat = false;      // used to seed rate array so we startup with reasonable BPM

/*
  >> Pulse Sensor Amped 1.2 <<
  This code is for Pulse Sensor Amped by Joel Murphy and Yury Gitman
    www.pulsesensor.com
    >>> Pulse Sensor purple wire goes to Analog Pin 0 <<<
  Pulse Sensor sample aquisition and processing happens in the background via Timer 2 interrupt. 2mS sample rate.
  PWM on pins 3 and 11 will not work when using this code, because we are using Timer 2!
  The following variables are automatically updated:
  Signal :    int that holds the analog signal data straight from the sensor. updated every 2mS.
  IBI  :      int that holds the time interval between beats. 2mS resolution.
  BPM  :      int that holds the heart rate value, derived every beat, from averaging previous 10 IBI values.
  QS  :       boolean that is made true whenever Pulse is found and BPM is updated. User must reset.
  Pulse :     boolean that is true when a heartbeat is sensed then false in time with pin13 LED going out.

  This code is designed with output serial data to Processing sketch "PulseSensorAmped_Processing-xx"
  The Processing sketch is a simple data visualizer.
  All the work to find the heartbeat and determine the heartrate happens in the code below.
  Pin 13 LED will blink with heartbeat.
  If you want to use pin 13 for something else, adjust the interrupt handler
  It will also fade an LED on pin fadePin with every beat. Put an LED and series resistor from fadePin to GND.
  Check here for detailed code walkthrough:
  http://pulsesensor.myshopify.com/pages/pulse-sensor-amped-arduino-v1dot1

  Code Version 1.2 by Joel Murphy & Yury Gitman  Spring 2013
  This update fixes the firstBeat and secondBeat flag usage so that realistic BPM is reported.

*/

//  VARIABLES
int pulsePin = 6;                 // Pulse Sensor purple wire connected to analog pin 0
int blinkPin = 7;                // pin to blink led at each beat
int fadePin = 5;                  // pin to do fancy classy fading blink at each beat
int fadeRate = 0;                 // used to fade LED on with PWM on fadePin

// these variables are volatile because they are used during the interrupt service routine!
volatile int BPM;                   // used to hold the pulse rate
volatile int Signal;                // holds the incoming raw data
volatile int IBI = 600;             // holds the time between beats, must be seeded!
volatile boolean Pulse = false;     // true when pulse wave is high, false when it's low
volatile boolean QS = false;        // becomes true when Arduoino finds a beat.

void setup() {
  pinMode(9, OUTPUT);
  digitalWrite(9, HIGH);
  pinMode(blinkPin, OUTPUT);        // pin that will blink to your heartbeat!
  pinMode(fadePin, OUTPUT);         // pin that will fade to your heartbeat!
  Serial.begin(115200);             // we agree to talk fast!
  interruptSetup();                 // sets up to read Pulse Sensor signal every 2mS
  // UN-COMMENT THE NEXT LINE IF YOU ARE POWERING The Arduino at 5V and the Pulse Sensor AT 3V
  // AND APPLY THE LOWER VOLTAGE TO THE A-REF PIN
  //analogReference(EXTERNAL);
}

void loop() {
  sendDataToProcessing('S', Signal);     // send Processing the raw Pulse Sensor data
  if (QS == true) {                      // Quantified Self flag is true when arduino finds a heartbeat
    fadeRate = 255;                  // Set 'fadeRate' Variable to 255 to fade LED with pulse
    sendDataToProcessing('B', BPM);  // send heart rate with a 'B' prefix
[... 88 lines cut ...]
      // QS FLAG IS NOT CLEARED INSIDE THIS ISR
    }
  }

  if (Signal < thresh && Pulse == true) {  // when the values are going down, the beat is over
    digitalWrite(blinkPin, LOW);           // turn off pin 13 LED
    Pulse = false;                         // reset the Pulse flag so we can do it again
    amp = P - T;                           // get amplitude of the pulse wave
    thresh = amp / 2 + T;                  // set thresh at 50% of the amplitude
    P = thresh;                            // reset these for next time
    T = thresh;
  }

  if (N > 2500) {                          // if 2.5 seconds go by without a beat
    thresh = 512;                          // set thresh default
    P = 512;                               // set P default
    T = 512;                               // set T default
    lastBeatTime = sampleCounter;          // bring the lastBeatTime up to date
    firstBeat = true;                      // set these to avoid noise
    secondBeat = false;                    // when we get the heartbeat back
  }

  sei();                                   // enable interrupts when youre done!
}// end isr`
please help me

biomurph, 2019-05-17:
This code has been deprecated. Please don't use it.
Use the PulseSensor Playground library instead.

https://github.com/WorldFamousElectronics/PulseSensorPlayground

closed 2019-05-17 by biomurph

FerrariF70, 2019-05-22:
> 
> 
> This code has been deprecated. Please don't use it.
> Use the PulseSensor Playground library instead.
> 
> https://github.com/WorldFamousElectronics/PulseSensorPlayground

I solved this problem with analog pin 11 thats flora pin 12

[image]

`#define USE_ARDUINO_INTERRUPTS true   
#include <PulseSensorPlayground.h>      

const int PulseWire = 11; //this ping #12 on flora    
const int LED13 = 13;         
int Threshold = 550;          
                               
PulseSensorPlayground pulseSensor; 

void setup() {   
  Serial.begin(9600);          
  pulseSensor.analogInput(PulseWire);   
  pulseSensor.blinkOnPulse(LED13);       
  pulseSensor.setThreshold(Threshold);    
   if (pulseSensor.begin()) {
    Serial.println("We created a pulseSensor Object !");   
  }
}
void loop() {
 int myBPM = pulseSensor.getBeatsPerMinute(); 
if (pulseSensor.sawStartOfBeat()) {           
 Serial.println("♥  A HeartBeat Happened ! "); 
 Serial.print("BPM: ");                       
 Serial.println(myBPM);                        
}
  delay(20);                   
}
`
that's work cool!!!

S20180020238, 2019-11-03:
> > This code has been deprecated. Please don't use it.
> > Use the PulseSensor Playground library instead.
> > https://github.com/WorldFamousElectronics/PulseSensorPlayground
> 
> I solved this problem with analog pin 11 thats flora pin 12
> 
> [image]
> 
> `#define USE_ARDUINO_INTERRUPTS true
> #include <PulseSensorPlayground.h>
> 
> const int PulseWire = 11; //this pin #12 on flora
> const int LED13 = 13;
> int Threshold = 550;
> 
> PulseSensorPlayground pulseSensor;
> 
> void setup() {
> Serial.begin(9600);
> pulseSensor.analogInput(PulseWire);
> pulseSensor.blinkOnPulse(LED13);
> pulseSensor.setThreshold(Threshold);
> if (pulseSensor.begin()) {
> Serial.println("We created a pulseSensor Object !");
> }
> }
> void loop() {
> int myBPM = pulseSensor.getBeatsPerMinute();
> if (pulseSensor.sawStartOfBeat()) {
> Serial.println("♥ A HeartBeat Happened ! ");
> Serial.print("BPM: ");
> Serial.println(myBPM);
> }
> delay(20);
> }
> `
> that's work cool!!!

bro, what pulse sensor you used to set up this code, and can you send the circuit diagram.
thanks..

FerrariF70, 2019-11-03:
> bro, what pulse sensor you used to set up this code, and can you send the circuit diagram.
> thanks..

Hi what you use controller Arduino or Flora?

S20180020238, 2019-11-03:
Arduino bro

FerrariF70, 2019-11-03:
> Arduino bro
you need this code and look [this page](https://pulsesensor.com/pages/code-and-guide) you need set wire red to 5v black ground and violet wire to one of analog port for example A0. If you set A0 in code you need set variable int PulseSensorPurplePin = 0, if A1 PulseSensorPurplePin = 1

//  Variables
int PulseSensorPurplePin = 0;        // Pulse Sensor PURPLE WIRE connected to ANALOG PIN 0
int LED13 = 13;   //  The on-board Arduion LED

int Signal;                // holds the incoming raw data. Signal value can range from 0-1024
int Threshold = 550;            // Determine which Signal to "count as a beat", and which to ingore.

// The SetUp Function:
void setup() {
  pinMode(LED13,OUTPUT);         // pin that will blink to your heartbeat!
   Serial.begin(9600);         // Set's up Serial Communication at certain speed.

}

// The Main Loop Function
void loop() {
  Signal = analogRead(PulseSensorPurplePin);  // Read the PulseSensor's value.
                                              // Assign this value to the "Signal" variable.
  Serial.println(Signal);                    // Send the Signal value to Serial Plotter.
   if(Signal > Threshold){                          // If the signal is above "550", then "turn-on" Arduino's on-Board LED.
     digitalWrite(LED13,HIGH);
   } else {
     digitalWrite(LED13,LOW);                //  Else, the sigal must be below "550", so "turn-off" this LED.
   }
delay(10);
}

for this code same you need change only variable PulseWire = 0 or PulseWire = 1 ...

#define USE_ARDUINO_INTERRUPTS true
#include <PulseSensorPlayground.h>

const int PulseWire = 0
const int LED13 = 13;
int Threshold = 550;

PulseSensorPlayground pulseSensor;

void setup() {
Serial.begin(9600);
pulseSensor.analogInput(PulseWire);
pulseSensor.blinkOnPulse(LED13);
pulseSensor.setThreshold(Threshold);
if (pulseSensor.begin()) {
Serial.println("We created a pulseSensor Object !");
}
}
void loop() {
int myBPM = pulseSensor.getBeatsPerMinute();
if (pulseSensor.sawStartOfBeat()) {
Serial.println("♥ A HeartBeat Happened ! ");
Serial.print("BPM: ");
Serial.println(myBPM);
}
delay(20);
}

S20180020238, 2019-11-03:
Thanks a lot bro.
Bro I am doing a project will you help me in that. If you are interested I
will tell it.

FerrariF70, 2019-11-03:
ok well than can I help,I will help.

S20180020238, 2019-11-03:
Thank you bro.
Actually my project is basically health monitoring system. I should data
like body temperature, heart rate, glucose levels and sleeping time of a
man and send that data to an app. I have to make that app. Can you suggest
ideas.

FerrariF70, 2019-11-03:
you have all that sensors?

S20180020238, 2019-11-04:
I have temperature sensor and heart rate sensor. I didn't have sensors for
sleeping and glucose sensor.

rohan30497, 2019-11-20:
Hi guys... I am also doing the same project on "patient health monitoring system", but am having some problem in the coding section. Like I have seperate codes for my temperature sensor and pulse sensor, but facing difficulty in combing both the codes into one? so please can anyone help me?

FerrariF70, 2019-11-20:
Hello rohan, as you understand if you have problems? and what is your problem?

rohan30497, 2019-11-21:
`#include <LiquidCrystal.h>

const int rs = 12, en = 11, d4 = 5, d5 = 4, d6 = 3, d7 = 2;
LiquidCrystal lcd(12, 11, 5, 4, 3, 2);

int pulsePin = A0;                 // Pulse Sensor purple wire connected to analog pin A0
int blinkPin = 13;                // pin to blink led at each beat

// Volatile Variables, used in the interrupt service routine!
volatile int BPM;                   // int that holds raw Analog in 0. updated every 2mS
volatile int Signal;                // holds the incoming raw data
volatile int IBI = 600;             // int that holds the time interval between beats! Must be seeded! 
volatile boolean Pulse = false;     // "True" when User's live heartbeat is detected. "False" when not a "live beat". 
volatile boolean QS = false;        // becomes true when Arduoino finds a beat.

static boolean serialVisual = true;   // Set to 'false' by Default.  Re-set to 'true' to see Arduino Serial Monitor ASCII Visual Pulse 

volatile int rate[10];                      // array to hold last ten IBI values
volatile unsigned long sampleCounter = 0;          // used to determine pulse timing
volatile unsigned long lastBeatTime = 0;           // used to find IBI
volatile int P = 512;                      // used to find peak in pulse wave, seeded
volatile int T = 512;                     // used to find trough in pulse wave, seeded
volatile int thresh = 525;                // used to find instant moment of heart beat, seeded
volatile int amp = 100;                   // used to hold amplitude of pulse waveform, seeded
volatile boolean firstBeat = true;        // used to seed rate array so we startup with reasonable BPM
volatile boolean secondBeat = false;      // used to seed rate array so we startup with reasonable BPM

void setup()
{
  pinMode(blinkPin,OUTPUT);         // pin that will blink to your heartbeat!
  Serial.begin(115200);             // we agree to talk fast!
  interruptSetup();                 // sets up to read Pulse Sensor signal every 2mS 
                                    // IF YOU ARE POWERING The Pulse Sensor AT VOLTAGE LESS THAN THE BOARD VOLTAGE, 
                                    // UN-COMMENT THE NEXT LINE AND APPLY THAT VOLTAGE TO THE A-REF PIN
                                    //   analogReference(EXTERNAL);   
 lcd.begin(16, 2);
 lcd.clear();
}

//  Where the Magic Happens
void loop()
{
   serialOutput();  
   
  if (QS == true) // A Heartbeat Was Found
    {     
      // BPM and IBI have been Determined
      // Quantified Self "QS" true when arduino finds a heartbeat
      serialOutputWhenBeatHappens(); // A Beat Happened, Output that to serial.     
      QS = false; // reset the Quantified Self flag for next time    
    }
     
  delay(20); //  take a break
}

void interruptSetup()
{     
  // Initializes Timer2 to throw an interrupt every 2mS.
  TCCR2A = 0x02;     // DISABLE PWM ON DIGITAL PINS 3 AND 11, AND GO INTO CTC MODE
  TCCR2B = 0x06;     // DON'T FORCE COMPARE, 256 PRESCALER 
  OCR2A = 0X7C;      // SET THE TOP OF THE COUNT TO 124 FOR 500Hz SAMPLE RATE
  TIMSK2 = 0x02;     // ENABLE INTERRUPT ON MATCH BETWEEN TIMER2 AND OCR2A
  sei();             // MAKE SURE GLOBAL INTERRUPTS ARE ENABLED      
} 

void serialOutput()
{   // Decide How To Output Serial. 
 if (serialVisual == true)
  {  
     arduinoSerialMonitorVisual('-', Signal);   // goes to function that makes Serial Monitor Visualizer
[... 139 lines cut ...]
float tempf;  //variable to store temperature in Fahreinheit 
float vout;  //temporary variable to hold sensor reading
void setup()
{
pinMode(sensor,INPUT); // Configuring pin A1 as input
Serial.begin(9600);
lcd.begin(16,2);  
  delay(500);
}
void loop() 
{
vout=analogRead(sensor);
vout=(vout*500)/1023;
tempc=vout; // Storing value in Degree Celsius
tempf=(vout*1.8)+32; // Converting to Fahrenheit 
lcd.setCursor(0,0);
lcd.print("in DegreeC= ");
lcd.print(tempc);
lcd.setCursor(0,1);
lcd.print("in Fahrenheit=");
lcd.print(tempf);
delay(1000); //Delay of 1 second for ease of viewing in serial monitor
}`

So how to combine both the codes, so that I can simultaneously get both values of temp. and pulse on the lcd display?

FerrariF70, 2019-11-21:
This code for PulseSensor has been deprecated. Please don't use it.
Use the PulseSensor Playground library instead.
[https://github.com/WorldFamousElectronics/PulseSensorPlayground](https://github.com/WorldFamousElectronics/PulseSensorPlayground)

I never used display for arduino. But what is problem use both code for Pulse sensor and Temperature? you have two sensor ok and you read signals from analogs pin, simple read data from pulse sensor pin and temperature pin and display to screen
Example this pulse sensor code it's work:
`
#define USE_ARDUINO_INTERRUPTS true    // Set-up low-level interrupts for most acurate BPM math.
#include <PulseSensorPlayground.h>     // Includes the PulseSensorPlayground Library.   

//  Variables
const int PulseWire = 0;       // PulseSensor PURPLE WIRE connected to ANALOG PIN 0
const int LED13 = 13;          // The on-board Arduino LED, close to PIN 13.
int Threshold = 550;           // Determine which Signal to "count as a beat" and which to ignore.
                               // Use the "Gettting Started Project" to fine-tune Threshold Value beyond default setting.
                               // Otherwise leave the default "550" value. 
                               
PulseSensorPlayground pulseSensor;  // Creates an instance of the PulseSensorPlayground object called "pulseSensor"

void setup() {   

  Serial.begin(9600);          // For Serial Monitor

  // Configure the PulseSensor object, by assigning our variables to it. 
  pulseSensor.analogInput(PulseWire);   
  pulseSensor.blinkOnPulse(LED13);       //auto-magically blink Arduino's LED with heartbeat.
  pulseSensor.setThreshold(Threshold);   

  // Double-check the "pulseSensor" object was created and "began" seeing a signal. 
   if (pulseSensor.begin()) {
    Serial.println("We created a pulseSensor Object !");  //This prints one time at Arduino power-up,  or on Arduino reset.  
  }
}

void loop() {

 int myBPM = pulseSensor.getBeatsPerMinute();  // Calls function on our pulseSensor object that returns BPM as an "int".
                                               // "myBPM" hold this BPM value now. 

if (pulseSensor.sawStartOfBeat()) {            // Constantly test to see if "a beat happened". 
 Serial.println("♥  A HeartBeat Happened ! "); // If test is "true", print a message "a heartbeat happened".
 Serial.print("BPM: ");                        // Print phrase "BPM: " 
 Serial.println(myBPM);                        // Print the value inside of myBPM. 
}

  delay(20);                    // considered best practice in a simple sketch.

}
`
```
#include<LiquidCrystal.h>
 LiquidCrystal lcd(12, 11, 5, 4, 3, 2); 
const int sensor=A1; analog pin A1 to variable 'sensor' 
float tempc; 
float tempf; 
 float vout; 
void setup() 
{ 
pinMode(sensor,INPUT); 
 Serial.begin(9600); 
lcd.begin(16,2); 
delay(500);
 }
 void loop() 
{ 
vout=analogRead(sensor); 
vout=(vout*500)/1023; 
tempc=vout; 
Celsius tempf=(vout*1.8)+32; 
 lcd.setCursor(0,0); lcd.print("in DegreeC= ");
 lcd.print(tempc); lcd.setCursor(0,1); 
lcd.print("in Fahrenheit="); 
lcd.print(tempf); delay(1000); 
 }
```
and insert your code for temperature in this code

rohan30497, 2019-11-21:
`#define USE_ARDUINO_INTERRUPTS true // Set-up low-level interrupts for most acurate BPM math.
#include <PulseSensorPlayground.h> // Includes the PulseSensorPlayground Library.
#include<LiquidCrystal.h>
// Variables
const int PulseWire = 0; // PulseSensor PURPLE WIRE connected to ANALOG PIN 0
const int LED13 = 13; // The on-board Arduino LED, close to PIN 13.
int Threshold = 550; // Determine which Signal to "count as a beat" and which to ignore.
// Use the "Gettting Started Project" to fine-tune Threshold Value beyond default setting.
// Otherwise leave the default "550" value.

LiquidCrystal lcd(12, 11, 5, 4, 3, 2); 
const int sensor=A1; //analog pin A1 to variable 'sensor' 
float tempc; 
float tempf; 
 float vout; 

PulseSensorPlayground pulseSensor; // Creates an instance of the PulseSensorPlayground object called "pulseSensor"

void setup() {

Serial.begin(9600); // For Serial Monitor

// Configure the PulseSensor object, by assigning our variables to it.
pulseSensor.analogInput(PulseWire);
pulseSensor.blinkOnPulse(LED13); //auto-magically blink Arduino's LED with heartbeat.
pulseSensor.setThreshold(Threshold);

// Double-check the "pulseSensor" object was created and "began" seeing a signal.
if (pulseSensor.begin()) {
Serial.println("We created a pulseSensor Object !"); //This prints one time at Arduino power-up, or on Arduino reset.
}

pinMode(sensor,INPUT); 
 Serial.begin(9600); 
lcd.begin(16,2); 
delay(500);
 }

void loop() {
  
int myBPM = pulseSensor.getBeatsPerMinute(); // Calls function on our pulseSensor object that returns BPM as an "int".
// "myBPM" hold this BPM value now.
vout=analogRead(sensor); 
vout=(vout*500)/1023; 
tempc=vout; 
Celsius tempf=(vout*1.8)+32; 
lcd.setCursor(0,0); lcd.print("in DegreeC= ");
 lcd.print(tempc); lcd.setCursor(0,1); 
lcd.print("in Fahrenheit="); 
lcd.print(tempf); delay(1000); 

if (pulseSensor.sawStartOfBeat()) { // Constantly test to see if "a beat happened".
Serial.println("♥ A HeartBeat Happened ! "); // If test is "true", print a message "a heartbeat happened".
Serial.print("BPM: "); // Print phrase "BPM: "
Serial.println(myBPM); // Print the value inside of myBPM.
}

delay(20); // considered best practice in a simple sketch.

}
`

Is this code okay if I want to display both temperature and pulse rate values at the same time in my lcd display?

rohan30497, 2019-11-21:
For my project, I need to display the values of temperature and pulse sensor at the same time in both serial monitor as well as in lcd screen. So can you help me with it?

FerrariF70, 2019-11-22:
have you tried run this code? 

`void loop() {

int myBPM = pulseSensor.getBeatsPerMinute(); // Calls function on our pulseSensor object that returns BPM as an "int".
// "myBPM" hold this BPM value now.
vout=analogRead(sensor);
vout=(vout500)/1023;
tempc=vout;
Celsius tempf=(vout1.8)+32;
lcd.setCursor(0,0); lcd.print("in DegreeC= ");
lcd.print(tempc); lcd.setCursor(0,1);
lcd.print("in Fahrenheit=");
lcd.print(tempf); delay(1000);

if (pulseSensor.sawStartOfBeat()) { // Constantly test to see if "a beat happened".
lcd.print(myBPM); // Print the value inside of myBPM.
}`

gsb-19, 2020-03-26:
i have a doubt i am a beginner,
why do we need an interrupt in this case...
and when will we know at which time i will be needing an interrupt..
can anyone please help,
thank you.

FerrariF70, 2020-03-26:
> i have a doubt i am a beginner,
> why do we need an interrupt in this case...
> and when will we know at which time i will be needing an interrupt..
> can anyone please help,
> thank you.

interruptions are needed so that the sensor can count how many heartbeats

gsb-19, 2020-03-26:
that is only being done by "analogRead" right again why need an separate interrupt?
im sorry it might sound dumb but im new so only im asking,
thank you

gsb-19, 2020-03-28:
so??

biomurph, 2020-03-28:
We use interrupts in order to make sure that the sample rate is accurate. If you don't have an accurate sample rate, you can't really do effective digital signal processing. On each interrupt, a sample is taken, and the algorithm is run on that sample.

FerrariF70, 2020-03-29:
> that is only being done by "analogRead" right again why need an separate interrupt?
> im sorry it might sound dumb but im new so only im asking,
> thank you

that would completely understand please read this guide [Processing](https://pulsesensor.com/pages/processing-visualization)

jil7102000, 2021-04-19:
> Thank you bro. Actually my project is basically health monitoring system. I should data like body temperature, heart rate, glucose levels and sleeping time of a man and send that data to an app. I have to make that app. Can you suggest ideas.
> […](#)

Can u send the code if u have completed the pRojEct and have got proper output.   I am facing issues. I m nt getting stable readings of temperature as well as the BPM

jil7102000, 2021-04-19:
> have you tried run this code?
>
> `void loop() {
>
> int myBPM = pulseSensor.getBeatsPerMinute(); // Calls function on our pulseSensor object that returns BPM as an "int".
> // "myBPM" hold this BPM value [...]

What are the connections for this code.
Can u help me out