WorldFamousElectronics/PulseSensor_Amped_Arduino #61: Pulse sensor Sampling rate

leghealer2, 2017-09-07:
Looks like the sampling rate on timer 2 is 2ms. But when I retrieve the data on to the serial monitor and check the sampling rate of pulse data it comes around 20ms. Is the sampling rate of interrupt different from the rate at which pulse sensor detects the data?

Please help me understand the concept.

Thanks

biomurph, 2017-09-07:
That is correct. The Arduino samples at a highly accurate 2mS based on the timer interrupt. This makes our IBI and BPM calculations accurate. If you look in the loop function, there is a 20mS delay there, which is used so that the serial port does not get clogged. 
Since we are determining IBI and BPM on the hardware, we don't need to send all that data to the computer.

If you need to get all the data, and want to modify the code, I would suggest NOT putting any Serial.print function in the interrupt routine. That will likely break. A better route would be to use a boolean variable, and set it true in the interrupt, and then have an if statement in the loop that checks it and sends the latest sample when it's true. Also would want to reset the boolean variable to false. 

Does that make sense?

leghealer2, 2017-09-07:
That Makes sense.I removed the 20ms delay and its sampling at 2ms. Amazing!
Thank you for the reply.

closed 2017-09-09 by leghealer2

xueting123456, 2017-10-26:
Hi, hello. I saw your information on GitHub. Do you have the code to calculate the peak time difference through 2 pulse sensors?