WorldFamousElectronics/PulseSensor_Amped_Arduino #62: Advanced program not working as expected

Yerion, 2017-09-11:
Hello, 

I have been trying to get the advance code working, but i'm not getting the output i expect to see when i compare it to the examples. But the basic code does work more or less like expected. Any idea what could be wrong? I haven't changed the code and i'm using a arduino uno.

Results of the serial monitor, running the basic code and the advanced code.
[image]
[image]

I bought this sensor from a dutch webshop. So i'm not a 100% sure it is the original sensor. 
Here is a photo of the sensor:
[image]
[image]

biomurph, 2017-09-11:
Unfortunately, you have purchased a knockoff Pulse Sensor.
Sorry, but we can't help you troubleshoot it, since we have no idea of the parts or quality.
Please support the original makers and purchase from www.pulsesensor.com
You can also find our product at www.adafruit.com and www.sparkfun.com
It sounds like you are in the EU. There are many brick and mortar stores that receive stock from SparkFun, and may have Pulse Sensors. To find SparkFun distributors, go to the bottom of the SparkFun page and there you will find a link.

Yerion, 2017-09-11:
Aah, that is a shame. I notified the seller and will buy a real one. Thanks for the quick reply.

closed 2017-09-11 by Yerion