WorldFamousElectronics/PulseSensor_Amped_Arduino #63: my pulse isnt stable 200 bpm more

shinkaairu, 2017-11-23:
ask, why pulse isnt stable and sometime up to 200bpm 
here i just using adruino uno, ethernetshield, and pulse sensor.
[image]

and this is my code

#define PROCESSING_VISUALIZER 1
#define SERIAL_PLOTTER  2

//  Variables
const int numPulseSensors = 2;
const int threshold = 530;
int pulsePin[numPulseSensors];                 // Pulse Sensor purple wire connected to analog pin 0
int blinkPin[numPulseSensors];                // pin to blink led at each beat
int fadePin[numPulseSensors];                  // pin to do fancy classy fading blink at each beat
int fadeRate[numPulseSensors];                 // used to fade LED on with PWM on fadePin

// Volatile Variables, used in the interrupt service routine!
volatile int BPM[numPulseSensors];                   // int that holds raw Analog in 0. updated every 2mS
volatile int Signal[numPulseSensors];                // holds the incoming raw data
volatile int IBI[numPulseSensors];             // int that holds the time interval between beats! Must be seeded!
volatile boolean Pulse[numPulseSensors];     // "True" when User's live heartbeat is detected. "False" when not a "live beat".
volatile boolean QS[numPulseSensors];        // becomes true when Arduoino finds a beat.

// Regards Serial OutPut  -- Set This Up to your needs
static boolean serialVisual = false;   // Set to 'false' by Default.  Re-set to 'true' to see Arduino Serial Monitor ASCII Visual Pulse

volatile int rate[numPulseSensors][10];                    // array to hold last ten IBI values
volatile unsigned long sampleCounter = 0;          // used to determine pulse timing
volatile unsigned long lastBeatTime[numPulseSensors];           // used to find IBI
volatile int P[numPulseSensors];                      // used to find peak in pulse wave, seeded
volatile int T[numPulseSensors];                     // used to find trough in pulse wave, seeded
volatile int thresh[numPulseSensors];                // used to find instant moment of heart beat, seeded
volatile int amp[numPulseSensors];                   // used to hold amplitude of pulse waveform, seeded
volatile boolean firstBeat[numPulseSensors];        // used to seed rate array so we startup with reasonable BPM
volatile boolean secondBeat[numPulseSensors];      // used to seed rate array so we startup with reasonable BPM

// SET THE SERIAL OUTPUT TYPE TO YOUR NEEDS
// PROCESSING_VISUALIZER works with Pulse Sensor Processing Visualizer
//      https://github.com/WorldFamousElectronics/PulseSensor_Amped_Processing_Visualizer
// SERIAL_PLOTTER outputs sensor data for viewing with the Arduino Serial Plotter
//      run the Arduino Serial Plotter at 115200 baud: Tools/Serial Plotter or Command+L

 static int outputType = PROCESSING_VISUALIZER;
// static int outputType = SERIAL_PLOTTER;

void setup() {

  setStuph();                       // initialize variables and pins

  Serial.begin(115200);             // we agree to talk fast!

  interruptSetup();                 // sets up to read Pulse Sensor signal every 2mS
}

void loop() {

  serialOutput() ;

  for (int i = 0; i < numPulseSensors; i++) {
    if (QS[i] == true) {
      fadeRate[i] = 0;         // Makes the LED Fade Effect Happen
      // Set 'fadeRate' Variable to 255 to fade LED with pulse
      serialOutputWhenBeatHappens(i);   // A Beat Happened, Output that to serial.
      QS[i] = false;
    }
  }

  ledFadeToBeat();                      // Makes the LED Fade Effect Happen
  delay(20);                            //  take a break
[... 187 lines cut ...]
    default:
      break;
  }

}

void serialOutputWhenBeatHappens(int p){
	char c;
  switch(outputType){
      case PROCESSING_VISUALIZER:    
	      c = 'A'+ p;                   // send first sensor BPM with prefix 'A'
        sendDataToSerial(c,BPM[p]);   // send successive sensor BPM with prefix 'B', 'C', etc
      	c = 'M'+ p;                   // send first sensor IBI with prefix 'M'
        sendDataToSerial(c,IBI[p]);   // send successive sensor IBI with prefix 'N', 'O', etc 
      break;
    default:
      break;
  }
}

//  Sends Data to Pulse Sensor Processing App, Native Mac App, or Third-party Serial Readers.
void sendDataToSerial(char symbol, int data ){
    Serial.print(symbol);
    Serial.println(data);
  }

biomurph, 2017-11-23:
Are you using two pulsesensors?
Please send pictures of your pulsesensors.
Also, what are you doing with the eternet shield?

shinkaairu, 2017-11-24:
i just use one pulse sensor
i'm using ethernet shield to send pulse sensor data to mysql.

[image]

biomurph, 2017-11-24:
Hi,
It is clear from the photographs that the hardware you are using a 'clone' or 'ripoff' of the original Pulse Sensor found here www.pulsesensor.com
I cannot help you with your problem, because I did not design the hardware you have and I have no idea what is on that PCB. Please support the original makers and purchase only legitimate products.

shinkaairu, 2017-11-24:
really > sooorry i dont know if this item not original :((
and thanks for your suggestion.

fernandeshenrique15, 2018-04-26:
Hi, I did the test and it worked normally, here it is 80BPM, I did some exercises and it went up to 115BPM as expected. @yury-g

yury-g, 2018-04-27:
Normal PulseSensor behavior reported by user. thank you @fernandeshenrique15 !

closed 2018-04-27 by yury-g