WorldFamousElectronics/PulseSensor_Amped_Arduino #65: pls help

ShadiShady, 2018-02-26:
When i leave pulse sensor it gives me BPM 
How can make BPM=0 when i leave sensor?

biomurph, 2018-02-26:
There is a setting called Threshold in the code at the top of the sketch.
Try increasing this value (range is 0 - 1023) to see if that helps.

yury-g, 2026-08-16:
This repository has been retired as a read-only historical archive. Its code and discussion remain preserved, but fixes and support no longer happen here.

For current Arduino installation, examples, supported boards, and source, use [PulseSensorPlayground](https://github.com/WorldFamousElectronics/PulseSensorPlayground). If a problem is reproducible with the current Playground release, open a new issue there with the board and core versions, library version, wiring and supply voltage, raw waveform, and a minimal sketch.

Closing this historical issue as not planned in this retired repository.

closed 2026-08-16 by yury-g