WorldFamousElectronics/PulseSensor_Amped_Arduino #66: URGENT Pulse Sensor not registering pulse w Adafruit Flora

lottieholland, 2018-03-03:
This is for a project due next Friday (6 days) so quick help would be very welcome! but even if I don't get it done in time I'd still love some help.

I have created a garment using the Adafruit Flora and conductive thread for the wiring.  The idea is that, when the heart beats, 4 NeoPixels will pulse with white light. I have tried lighting the NeoPixels in isolation, and they work fine. When I try to incorporate the Pulse Sensor into the code, it does not pick up a pulse when I place it on my finger. The LEDs very occasionally flash when my finger is nowhere near the sensor, or when I apply pressure to the cable or the Flora board itself. They also flash several times on startup, as I understand should be expected because the initial readings from the sensor fluctuate a bit.

Before I began wiring(sewing) everything together, I tried running code with the Pulse Sensor crocodile-clipped to the Flora, with the on-board NeoPixel as an output. This appeared to run fine and the light flashed at a steady rate. However, now that everything is sewn down, it doesn't seem to be responding properly anymore. I have tried running it both powered by my laptop via micro USB and powered by an external battery, neither made any difference, although I'm sticking to USB until I can fix it.

My setup looks like this:
[image]
[image]

And here is a diagram for clarity:
[image]

I had to solder the Pulse Sensor to a JST sewable converter, but again I tested this with crocodile clips after soldering and it worked fine, so I don't think that's the issue. The thread from the JST convertor runs - to ground, + to 3.3v and signal to pin 11 (#12).

Here is the code I'm using. I copied parts from various tutorials as I could not find code which deals specifically with the Flora. I'm quite new to Arduino so I don't entirely understand what's happening in the code, but it looks okay from what I can tell.
[pulsesensor.ino]
`
#include <Adafruit_NeoPixel.h>    // Importing the NeoPixel library

// Behavior setting variables ---- can change the pin numbers according to wiring setup
// Remember Pulse Width Modulation is disabled on pins 9 & 10 because Timer1 is in use 
int pulsePin = 11;                // Pulse Sensor purple signal wire connected to pulsePin
//note that pin #12 is the same as (analog) 11
int blinkPin = 7;                 // Digital 7 to blink red LED at each beat
int fadePin  = 6;                 // pin to do neopixel effects at each beat(8 is onboard pixel)
int fadeRate = 0;                 // used to fade LED on with PWM on fadePin

// these variables are volatile because they are used during the interrupt service routine
volatile int BPM;                   // used to hold the pulse rate
volatile int Signal;                // holds the incoming raw data
volatile int IBI = 600;             // holds the time between beats, the Inter-Beat Interval
volatile boolean Pulse = false;     // true when pulse wave is high, false when it's low
volatile boolean QS = false;        // becomes true when Arduoino finds a beat.

// Set up use of NeoPixels
const int NUMPIXELS = 4;            // Put the number of NeoPixels you are using here
const int BRIGHTNESS = 50;          // Set brightness of NeoPixels here
Adafruit_NeoPixel strip = Adafruit_NeoPixel(NUMPIXELS, fadePin, NEO_GRB + NEO_KHZ800);//declaring the "strip"

void setup(){
  pinMode(pulsePin, INPUT);         // Set pulsePin(the signal wire from Pulse Sensor) as input
  pinMode(blinkPin,OUTPUT);         // Set NeoPixel signal pin to an output
   Serial.begin(115200);           // Serial output data for debugging or external use
  strip.begin();
  strip.setBrightness(BRIGHTNESS);  // uses above brightness variable
  
  for (int x=0; x < NUMPIXELS; x++) {  // Initialize all pixels to 'off'
     strip.setPixelColor(x, strip.Color(0, 0, 0));
  }
  strip.show();                     // Ensure the pixels are off 
  delay(1000);                      // Wait a second
  interruptSetup();                 // sets up to read Pulse Sensor signal every 2mS 
}

void loop(){
  sendDataSerial('S', Signal);      // send Processing the raw Pulse Sensor data
  if (QS == true){                    // Quantified Self flag is true when arduino finds a heartbeat
     fadeRate = 255;                  // Set 'fadeRate' Variable to 255 to fade LED with pulse
     sendDataSerial('B',BPM);       // send heart rate with a 'B' prefix
     sendDataSerial('Q',IBI);       // send time between beats with a 'Q' prefix
     QS = false;                      // reset the Quantified Self flag for next time    
  }
  ledFadeToBeat();                    // Routine that fades color intensity to the beat
  delay(20);                          //  take a break
}

void ledFadeToBeat() {
    fadeRate -= 15;                         // Set LED fade value
    fadeRate = constrain(fadeRate,0,255);   // Keep LED fade value from going into negative numbers
[... 118 lines cut ...]
  if (N > 2500){                             // if 2.5 seconds go by without a beat
      thresh = 512;                          // set thresh default
      P = 512;                               // set P default
      T = 512;                               // set T default
      lastBeatTime = sampleCounter;          // bring the lastBeatTime up to date        
      firstBeat = true;                      // set these to avoid noise
      secondBeat = true;                     // when we get the heartbeat back
     }
  
  sei();                                     // enable interrupts when youre done!
}

// end isr
`

I couldn't get anything to show in the Serial Plotter window - it just showed up blank. I did get these values in the Serial Monitor though - it looks like the sensor had failed to detect a pulse since there are no BPM or IBI values.
[image]
These values also (sometimes) drop to constant 0s when I put my finger on the sensor. 

I'm using a plastic cover provided stuck to the pulse sensor, and I'm pretty sure it's a genuine.
[image]

Any help? Please bear in mind I really don't "get" Arduino code and so if you could explain or give pointers for how I can integrate new bits of code or make them compatible with the Flora still, that would be great.

Please please help me!!!!

lottieholland, 2018-03-03:
Woah I don't know why my code has displayed a bit stangely but it's still readable so whatever

yury-g, 2018-03-03:
A very fast and easy way to diagnose what kind of signal you’re getting is
to use this code
https://youtu.be/RbB8NSRa5X4

https://github.com/WorldFamousElectronics/PulseSensorPlayground?files=1

lottieholland, 2018-03-03:
Thanks for your response. I've uploaded the example code and opened the serial plotter, but the values are all over the place.

First, when nothing was touching the sensor:
[image]
[image]

It was going a bit mad. Next, when I first placed my finger on the sensor:
[image]
[image]

And finally, once I had left my finger on the sensor for several minutes to lets things "settle down":
[image]

Whilst my finger was on there, the red LED would flash on and off for a few seconds, then stay off, then do more sporadic flashing. If I could get it to flash consistently I guess the problem would be solved. 
If I shake the wire around without touching the sensor I actually get a pretty rhythmic flash from the LED.  This is the opposite of what I want- since it's a wearable, I need to be able to move the wires around a bit without getting false readings.

Any idea why it's behaving like this?

biomurph, 2018-03-03:
I'm not sure why this is happening, but there is a thing about the neopixel library in that it messes with interrupts whenever it does the 'strip.show()' command. This could be causing havoc with the Pulse Sensor code because Pulse Sensor that you're using uses interrupts.

The code in this library has been superseded by our new Pulse Sensor library
https://github.com/WorldFamousElectronics/PulseSensorPlayground
Please load that into your arduino (you can use the library manager tool)
There will be an example that is called 'PulseSensor_BPM_Alternative' which does not use interrupts.
Try building on that.

Regarding the noise that you see. I don't really find any problems with the wired setup that you're using.

lottieholland, 2018-03-03:
Thanks, I'll give that a shot. I'm not entirely sure if that's the problem, or at least if its the sole problem, since when I ran the basic Getting Started code above, the on board LED was still acting up. As well as reacting to shakes in the wire and stopping flashing when my finger was on there, it would also stay red for extended periods even when my finger wasn't on there. And it's a standard high/low LED rather than a NeoPixel.
Anyways, I'm gonna give what you suggested a bash. Trial and error!

lottieholland, 2018-03-05:
Thanks a bunch, I'm making progress now. There just seems to be one more (major) issue.

I re-wrote the code based on the "PulseSensor_BPM_Alternative" example.

When I trial this code using the onboard NeoPixel (pin 8) as an output, it performs correctly. The NeoPixel blinks as expected and the Serial Plotter looks like this:
[image]

Seems like good data to me. The only issue is, when I change the code so that the output is to pin 6 (where I have connected my 4 NeoPixels), something gets jumbled. I no longer get any pixel blinks, and it also seems to have interfered with pin 11 (pulse sensor) input. Because now when I look at the Serial Plotter, I get this:
[image]

Uh-oh. Not much going on there, not even noise.

I'm guessing that possibly the conductive thread traces could be touching and interfering with each other? But I have placed layers of card and fabric between the two, and even tried physically holding them apart. So then could it be the code? I can't see why it would be okay with pin 8 but not pin 6???

New code below:
`
#define USE_ARDUINO_INTERRUPTS false
#include <PulseSensorPlayground.h>
#include <Adafruit_NeoPixel.h>

const int OUTPUT_TYPE = SERIAL_PLOTTER;

const int PIN_INPUT = 11;
const int PIN_BLINK = 7;    // Pin 7 is the on-board LED
const int NEO_PIXELS = 8;     // Pin 8 is onboard NeoPixel
const int NUMPIXELS = 1; // how many in the strip
const int BRIGHTNESS = 30;
const int THRESHOLD = 560;   // Adjust this number to avoid noise when idle
int SIGNAL;
int fadeRate = 0; //used later in code to fade the NeoPixles

byte samplesUntilReport;
const byte SAMPLES_PER_SERIAL_SAMPLE = 10;

PulseSensorPlayground pulseSensor;
Adafruit_NeoPixel strip = Adafruit_NeoPixel(NUMPIXELS, NEO_PIXELS, NEO_GRB + NEO_KHZ800);//setting up the strip

void setup() {
  pinMode (PIN_INPUT, INPUT);
  pinMode (NEO_PIXELS, OUTPUT);
  Serial.begin(115200); //keep this set to 115200 to use the Serial Plotter effectively
  strip.begin();
  strip.setBrightness(BRIGHTNESS);

  for (int x=0; x < NUMPIXELS; x++){ //initialize all pixels to be off
    strip.setPixelColor(x, strip.Color(0, 0, 0));
  }
  strip.show();// just showing the pixels now to make sure they're off
  
  // Configure the PulseSensor manager.
  pulseSensor.analogInput(PIN_INPUT);
  pulseSensor.blinkOnPulse(PIN_BLINK);
  pulseSensor.setSerial(Serial);
  pulseSensor.setOutputType(OUTPUT_TYPE);
  pulseSensor.setThreshold(THRESHOLD);

  // Skip the first SAMPLES_PER_SERIAL_SAMPLE in the loop().
  samplesUntilReport = SAMPLES_PER_SERIAL_SAMPLE;

  // Now that everything is ready, start reading the PulseSensor signal.
  if (!pulseSensor.begin()) {//if the pulse sensor fails to begin, flash in error
    for(;;) {
      // Flash the led to show things didn't work.
      digitalWrite(PIN_BLINK, LOW);
      delay(50);
      digitalWrite(PIN_BLINK, HIGH);
      delay(50);
    }
  }
}

void fade(){
  fadeRate -= 15;
  fadeRate = constrain(fadeRate, 0, 255);
  setStrip(fadeRate);
  }

void setStrip(int r){
  //int  g = 0;
  //int  b = 0;
  for (int x=0; x < NUMPIXELS; x++) {
    strip.setPixelColor(x, strip.Color(r, r, r));
  }
  strip.show();
}

void loop() {

  if (pulseSensor.sawNewSample()) {
    /*
       Every so often, send the latest Sample.
       We don't print every sample, because our baud rate
       won't support that much I/O.
    */
    if (--samplesUntilReport == (byte) 0) { //this is to do with the timing of the samples being taken
      samplesUntilReport = SAMPLES_PER_SERIAL_SAMPLE; 
      pulseSensor.outputSample();//once the time has passed it outputs the sample
      /*
         At about the beginning of every heartbeat,
         report the heart rate and inter-beat-interval.
      */
      if (pulseSensor.sawStartOfBeat()) {
        pulseSensor.outputBeat();
      }
    }

  SIGNAL = analogRead(PIN_INPUT);
  if(SIGNAL > THRESHOLD){                          // If there is a beat
    digitalWrite(PIN_BLINK,HIGH);
    fadeRate = 255;
  } else {
    digitalWrite(PIN_BLINK,LOW);
    //fadeRate = 0
  }

  fade();
  
//end of void loop
}

}
`
As always, help very much appreciated!!!! Please? :'(

j3nsykes, 2018-03-05:
What version of the Flora are you using?
Try changing this line
Adafruit_NeoPixel strip = Adafruit_NeoPixel(NUMPIXELS, NEO_PIXELS, NEO_GRB + NEO_KHZ800);//setting up the strip

to this

Adafruit_NeoPixel strip = Adafruit_NeoPixel(NUMPIXELS, NEO_PIXELS, NEO_RGB + NEO_KHZ400);//setting up the strip

lottieholland, 2018-03-05:
Hi, I'm using V3. I tried this, but it doesn't make a difference. What is really perplexing me is that the code basically works until I change my NEO_PIXELS variable to pin 6. I can put any other number in there and it's all fine, but as soon as I pop 6 into that int, I get a flatline on the pulse sensor signal. I thought that maybe it's a problem with the stitching to pin 6, but I can run seperate code which lights up the pin 6 neopixels just fine, so I don't know if that's it. Just the combination of pins 6 and 11 seems bad. Do you think it would be worth stitching the NeoPixels to a different pin?

biomurph, 2018-03-05:
You are definitely getting good heartbeat, etc when you run the neopixel on pin 8.
Maybe I'm confused, but I don't see pin 8 or pin 11 on this flora v3
https://www.adafruit.com/product/659
have not played with a flora in a while.

Is it possible to just switch to the other pin? Clearly there is something going on with the flora that is cause this ruckus...

lottieholland, 2018-03-06:
Hi. Pin 8 is actually the on board NeoPixel, and pin 11 is the same as #12 - addressing it as 11 refers to the analogue pin that it doubles as.

I've actually had a bit of help troubleshooting now and deduced that the issue was a faulty connection to pin 6 - looks like there is a short in the thread. I'm going to re-sew this connection and report back, but it's looking like that might well solve the issue :D

biomurph, 2018-03-06:
OK! Good luck sewing!

lottieholland, 2018-03-10:
Yup, it was just a dodgy connection. Sewing it anew has fixed the problem. Code working smoothly now. 
[image]
Here's the finished dress. Thanks all!!

closed 2018-03-10 by lottieholland

engr-khalil, 2019-12-02:
@lottieholland can you send me your updated code

lottieholland, 2019-12-03:
Hi  @engr-khalil . I've found the file which I think I used in the end, but I had a few versions saved, so I hope this is the right one. I would run the code to double check it works, but I've since dismantled the dress and circuitry so I don't have a way to test it.

Here's the code I found, there are only a couple minor changes:

```
#define USE_ARDUINO_INTERRUPTS false
#include <PulseSensorPlayground.h>
#include <Adafruit_NeoPixel.h>

const int OUTPUT_TYPE = SERIAL_PLOTTER;

const int PIN_INPUT = 11;
const int PIN_BLINK = 7;    // Pin 7 is the on-board LED
const int NEO_PIXELS = 10;     // Pin 10 is where my NeoPixels are connected
const int NUMPIXELS = 4; // how many in the strip (in this case, 4)
const int BRIGHTNESS = 30;
const int THRESHOLD = 560;   // Adjust this number to avoid noise when idle
int SIGNAL;
int fadeRate = 0; //used later in code to fade the NeoPixles

byte samplesUntilReport;
const byte SAMPLES_PER_SERIAL_SAMPLE = 10;

PulseSensorPlayground pulseSensor;
Adafruit_NeoPixel strip = Adafruit_NeoPixel(NUMPIXELS, NEO_PIXELS, NEO_GRB + NEO_KHZ400);//setting up the strip

void setup() {
  pinMode (PIN_INPUT, INPUT);
  pinMode (NEO_PIXELS, OUTPUT);
  Serial.begin(115200); //keep this set to 115200 to use the Serial Plotter effectively
  strip.begin();
  strip.setBrightness(BRIGHTNESS);

  for (int x=0; x < NUMPIXELS; x++){ //initialize all pixels to be off
    strip.setPixelColor(x, strip.Color(0, 0, 0));
  }
  strip.show();// just showing the pixels now to make sure they're off
  
  // Configure the PulseSensor manager.
  pulseSensor.analogInput(PIN_INPUT);
  //pulseSensor.blinkOnPulse(PIN_BLINK);
  pulseSensor.setSerial(Serial);
  pulseSensor.setOutputType(OUTPUT_TYPE);
  pulseSensor.setThreshold(THRESHOLD);

  // Skip the first SAMPLES_PER_SERIAL_SAMPLE in the loop().
  samplesUntilReport = SAMPLES_PER_SERIAL_SAMPLE;
  digitalWrite(PIN_BLINK,LOW);

  // Now that everything is ready, start reading the PulseSensor signal.
  if (!pulseSensor.begin()) {//if the pulse sensor fails to begin, flash in error
    for(;;) {
      // Flash the led to show things didn't work.
      //digitalWrite(PIN_BLINK, LOW);
      delay(50);
      //digitalWrite(PIN_BLINK, HIGH);
      delay(50);
    }
  }
}

void fade(){
  fadeRate -= 15;
  fadeRate = constrain(fadeRate, 0, 255);
  setStrip(fadeRate);
  }

void setStrip(int r){	
  for (int x=0; x < NUMPIXELS; x++) {	
    strip.setPixelColor(x, strip.Color(r, r, r));	
  }	
  strip.show();	
}

void loop() {

  if (pulseSensor.sawNewSample()) {
    /*
       Every so often, send the latest Sample.
       We don't print every sample, because our baud rate
       won't support that much I/O.
    */
    if (--samplesUntilReport == (byte) 0) { //this is to do with the timing of the samples being taken
      samplesUntilReport = SAMPLES_PER_SERIAL_SAMPLE; 
      pulseSensor.outputSample();//once the time has passed it outputs the sample
      /*
         At about the beginning of every heartbeat,
         report the heart rate and inter-beat-interval.
      */
      if (pulseSensor.sawStartOfBeat()) {
        pulseSensor.outputBeat();
      }
    }

  SIGNAL = analogRead(PIN_INPUT);
  if(SIGNAL > THRESHOLD){ // If there is a beat
   // digitalWrite(PIN_BLINK,HIGH);
    fadeRate = 255;
  } else {
  //  digitalWrite(PIN_BLINK,LOW);
    //fadeRate = 0
  }

  fade();
  
//end of void loop
}

}

```

Hope this helps you!