WorldFamousElectronics/PulseSensor_Amped_Arduino #68: Interfacing pulse senor with display shows abnormal reading

bhavindave9, 2018-03-13:
My main aim is to count Beats per minute form pulse sensor and displaying it on a display.
I am using the code from the example section of the pulsesensor playground. and clubbing it with my display.

I am using the following to build my project-
Pulse sensor
Arduino Uno
Nokia 5110 LCD display

Problem -
As I am using the code from examples as well as I have created a custom code to display the bpm along with the graphics.

1) Both the code are running correctly without any error individually i.e [ 1) detecting the heartbeat and calculating bpm 2) custom display to display an integer for about a second ].
But when I club both the code together code together to display bmp on display for a second and refreshing it,  the program starts to malfunction and shows garbage value when there is no figure on sensor.

My assumption and doubt- 
it is not advised to use delay for more than 20ms. but if I print the calculated bpm on display with the 20ms display  it cannot be visible to the human eye and if I increase delay t0 200ms the pulse sensor will skip a beat and  malfunction )

so what is the possible solution for it?

I tied to use millis() but it seems both timers cannot work together and throw a compilation error for uno.

please help me 

I really appreciate it.

thanks 
[lcd.zip](https://github.com/WorldFamousElectronics/PulseSensor_Amped_Arduino/files/1808245/lcd.zip)

biomurph, 2018-03-14:
IT doesn't look like the LCD library is messing too much with the Pulse Sensor code, but there must be something happening. It looks like the LCD library is using direct port control of the Arduino. 
You might want to try to use the BPM_Alternative example. In that case I would suggest the following changes to what you already have.
First, don't use any delays in your loop. 
Second, add a variable lastBPM that you can compare myBPM with

```
if (pulseSensor.sawStartOfBeat()) 
{
myBPM = pulseSensor.getBeatsPerMinute();      
if(myBPM != lastBPM){     
lastBPM = myBPM;
myGLCD.setFont(BigNumbers);
myGLCD.printNumI(myBPM, 11, 20);
myGLCD.update();
}
}
```

I don't know what therm is or what you're trying to do with that. 

If you only update the LCD when the BPM changes, you won't be hitting it so hard.

Also, your threshold is set pretty high
Also, you are declaring Serial.begin twice...

bhavindave9, 2018-03-14:
Okay, thanks for the suggestions. I will try to update the code and let you know. Your suggestions are always appreciated

yury-g, 2026-08-16:
This repository has been retired as a read-only historical archive. Its code and discussion remain preserved, but fixes and support no longer happen here.

For current Arduino installation, examples, supported boards, and source, use [PulseSensorPlayground](https://github.com/WorldFamousElectronics/PulseSensorPlayground). If a problem is reproducible with the current Playground release, open a new issue there with the board and core versions, library version, wiring and supply voltage, raw waveform, and a minimal sketch.

Closing this historical issue as not planned in this retired repository.

closed 2026-08-16 by yury-g