WorldFamousElectronics/PulseSensor_Amped_Arduino #69: Getting BPM value in Serial MOnitor

ayush162883, 2018-03-21:
Is there any way to print the BPM values in Serial Monitor??

bhavindave9, 2018-03-21:
Just download and install pulseSensor playground library in Arduino Ide and oper example section under file then select Getting_BPM_to_Monitor program to display bpm values in serial monitor

biomurph, 2018-03-22:
Please use the Pulse Sensor Playground library. This code is deprecated.
https://github.com/WorldFamousElectronics/PulseSensorPlayground

In the doc below, you can find useful commands to access data in the library
getBeatsPerMinute() is a function you can call that returns the most recent BPM value.

https://github.com/WorldFamousElectronics/PulseSensorPlayground/blob/master/resources/PulseSensor%20Playground%20Tools.md

closed 2018-03-22 by biomurph