WorldFamousElectronics/PulseSensor_Amped_Arduino #70: noise signal overpowers heartbeat signal

ghanding, 2018-04-14:
Hello!

When I read the signal from my pulse sensor, the waves always appear at the same frequency. Therefore, the bpm always shows up the same: for different people, before and after exercise, and with and without a finger often times. 

I think the sensor is registering a noise signal that is larger than the actual heartbeat signal. For some reason, the noise signal appears like a regular looking waveform that completely overpowers the heartbeat. I don't know what could be causing this other than a defective sensor. I ordered the real product, not a knockoff.

Thanks!

biomurph, 2018-04-16:
@ghanding Please provide a couple of things for us to troubleshoot.
Can you send a screen shot of the signal that you're seeing?
What Arduino board are you using? What OS?
Also, please send a pic of the Pulse Sensor you are using.

yury-g, 2026-08-16:
This repository has been retired as a read-only historical archive. Its code and discussion remain preserved, but fixes and support no longer happen here.

For current Arduino installation, examples, supported boards, and source, use [PulseSensorPlayground](https://github.com/WorldFamousElectronics/PulseSensorPlayground). If a problem is reproducible with the current Playground release, open a new issue there with the board and core versions, library version, wiring and supply voltage, raw waveform, and a minimal sketch.

Closing this historical issue as not planned in this retired repository.

closed 2026-08-16 by yury-g